platformio/platform-espressif32 #782: gpio_hold_en not working with versions >= 4.0.0 on plateform espressif32

dynek, 2022-04-28:
Hey guys,

Got a working piece of code to hold gpio state using `gpio_hold_en` when going into deep sleep mode on 3.5.0 (espressif32) but since I upgraded to 4.0.0 and 4.1.0 it doesn't work anymore.

Not sure if it's related to plateform io's extension or Espressif's SDK but there's obviously a regression.

Behavior works as expected with 3.5.0 but with >= 4.0.0 as soon as the ESP wakes up, GPIOs seem to be reset.

Thank you

Jason2866, 2022-04-29:
It is IDF / Arduino framework related not related to Platformio

dynek, 2022-05-04:
Is there any difference "espressif-wise" between 4.0.0 and 4.1.0 then ?
Because I was having i2c issues with 3.5.0 and 4.0.0 that are resolved in 4.1.0 but I can't seem to grasp the difference between 4.0.0 and 4.1.0 that would explain such a thing.

Jason2866, 2022-05-04:
There are many fixes from IDF 4.4.0 to IDF 4.4.1. Later Arduino Core use more actual IDF 4.4.1 version.
So all fixes in IDF 4.4.x are in newer Arduino builds. Yes, there are changes in IDF 4.4.1 regarding GPIOs which may causes regressions. If you want to know more details you have to follow espressif IDF and Arduino issues

enfo, 2022-05-06:
I got the same issue, has anyone found a fix for this other than downgrading?

dynek, 2022-05-08:
Haven't found any fix, haven't checked with Espressif directly (yet) and can't downgrade because it breaks i2c, at least when there are multiple i2c devices on the same bus, even when using semaphore.

Jason2866, 2022-05-08:
This is the wrong github for issues with Arduino espressif code.
Needs to be opened [there](https://github.com/espressif/arduino-esp32)

dynek, 2022-05-08:
You're totally right @Jason2866, closing this.

closed 2022-05-08 by dynek

dynek, 2022-05-09:
@Jason2866 have you tried with v4.2.0 released 10 days ago ? Issue persists ?

enfo, 2022-05-09:
> @enfo have you tried with v4.2.0 released 10 days ago ? Issue persists ?

Yes, I have and it still persists

dynek, 2022-06-01:
I'm really sorry for not being able to find out the info by myself but I'd like to raise an issue on arduino-esp32's github repo but I'm unable to find valuable information for them to investigate.
What is version "~3.10006.0" in `platform.json` ? It doesn't match any release in arduino-esp32's repo, currently at 2.0.3 🤔