platformio/platform-espressif32 #783: LittleFs --> Upload Filesystem Image errors on espressif 4.2.0

zekageri, 2022-04-29:
[image]

### platformio.ini

```C
[env:arduino-esp32]

platform = espressif32
;platform = espressif32 @ 4.1.0
;platform    = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

board       = esp32dev  ;esp-wrover-kit
framework   = arduino

;platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
;platform_packages =
   ;framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master

board_build.f_cpu   = 240000000L

upload_port     = COM8
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder

board_build.flash_size = 16MB
board_build.flash_mode = dio
;board_build.partitions = ./hsh_NewPartition.csv ;10 mb flash 3mb app 2x
board_build.partitions = ./hsh_Partition.csv   ;11 468 800 byte flash 2 621 440 byte app 2x
board_build.f_flash    = 80000000L

build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -mfix-esp32-psram-cache-strategy=memw
                -DCORE_DEBUG_LEVEL=0
                -D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
                -D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
              
extra_scripts = mklittlefs/replace_fs.py
```

### ESP32 Chip: ESP-WROVER-E ( 16mb flash 8mb PSRAM )

**Extra scripts to pack LittleFs instead of SPIFFS :** 

```py
Import("env")
print("Replace MKSPIFFSTOOL with mklittlefs.exe")
env.Replace (MKSPIFFSTOOL = "./mklittlefs/mklittlefs.exe")
```

### Fs upload debug:

```C
Replace MKSPIFFSTOOL with mklittlefs.exe
Building FS image from 'data' directory to .pio\build\arduino-esp32\spiffs.bin
/admin/bundles/zipped/admin.html
/assets/adminLogo.png
SPIFFS_write error(-10010): unknown
error adding file!
Error for adding content from zipped!
/assets/fonts/icomoon.eot
/assets/fonts/icomoon.svg
/assets/fonts/icomoon.ttf
/assets/fonts/icomoon.woff
/assets/fonts/minified/zipped/hsh_icons.min.css
SPIFFS_write error(-10010): unknown
/assets/hsh.webmanifest
error adding file!
Error for adding content from zipped!
/assets/hshAdmin.webmanifest
/assets/hshLogo.png
/assets/iconObj.json
/assets/icons/android-chrome-192x192.png
SPIFFS_write error(-10010): unknown
/assets/languages/DE.json
error adding file!
Error for adding content from icons!
/assets/languages/EN.json
/assets/languages/HU.json
/assets/languages/RO.json
/assets/moduleImages/CM4.svg
/assets/moduleImages/CR16.svg
/assets/moduleImages/LS10.svg
/assets/moduleImages/M8.svg
/assets/moduleImages/R16.svg
/assets/moduleImages/R32.svg
/assets/moduleImages/THERM_BLACK.svg
SPIFFS_write error(-10010): unknown
/assets/tftLogo.jpg
error adding file!
Error for adding content from moduleImages!
/assets/vendors/zipped/chart-3.4.0.minGZ.js
SPIFFS_write error(-10010): unknown
/login/bundles/zipped/login.html
error adding file!
/main/bundles/zipped/index.html
Error for adding content from zipped!
SPIFFS_write error(-10010): unknown
error adding file!
Error for adding content from zipped!
/main/bundles/zipped/mainScript.min.js
SPIFFS_write error(-10010): unknown
/notFound/notFound.html
error adding file!
Error for adding content from zipped!
```

### Results in a boot loop with the following serial messages:

```C
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
ets Jul 29 2019 12:21:46
```

valeros, 2022-04-29:
Hi @zekageri ! Could you try replacing your custom extra script with the `board_build.filesystem` option? Something like this: 
```ini
[env:arduino-esp32]
platform = espressif32
board       = esp32dev
framework   = arduino
board_build.filesystem = littlefs

board_build.f_cpu   = 240000000L
upload_port     = COM8
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder
board_build.flash_size = 16MB
board_build.flash_mode = dio
board_build.partitions = ./hsh_Partition.csv   ;11 468 800 byte flash 2 621 440 byte app 2x
board_build.f_flash    = 80000000L

build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -mfix-esp32-psram-cache-strategy=memw
                -DCORE_DEBUG_LEVEL=0
                -D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
                -D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
```

zekageri, 2022-04-29:
Oh absolutely. I did not know about that!

zekageri, 2022-04-29:
Should i delete cache and other folders?

zekageri, 2022-04-29:
**I erased the flash and i got this serial message now**

```C
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

zekageri, 2022-04-29:
Maybe my partition.csv is not quite right?

```csv
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x280000,
app1,     app,  ota_1,   0x290000,0x280000,
spiffs,   data, spiffs,  0x510000,0xAF0000,
```

valeros, 2022-04-29:
So at least there are no errors with creating a filesystem image now. Just to clarify, it worked before? Any chance you could try your project in Arduino IDE? Just to verify this behavior has nothing to do with the framework itself.

zekageri, 2022-04-29:
Nah, my project contains a bunch of cpp files and built totally around PIO. 
It takes up 

```C
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.5% (used 57368 bytes from 327680 bytes)
Flash: [=====     ]  46.6% (used 1221401 bytes from 2621440 bytes)
```
I can not port this easily to Arduino IDE unfortunately.

But this is my first attempt with the official PIO 4.2.0

zekageri, 2022-04-29:
I deleted cache, packages and platforms folder, recompiled my sketch, uploaded the flash again, and the sketch again.  deleted old 3.3.1 sha3784198 in order to leave just the 4.2.0. Did not help. Boot loop again.

zekageri, 2022-04-29:
Tired with 40mhz flash and qio config. did not help

valeros, 2022-04-29:
Did it work before? Maybe you could provide a minimal project to reproduce the issue?

zekageri, 2022-04-29:
Arduino IDE latest 2.0.3 RC-1 almost empty sketch works.

```C
void setup() {
  Serial.begin(115200);
  vTaskDelay(1000);
  Serial.println("Hello from ARDUINO IDE!");
}

void loop() { }
```

```C
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
Hello from ARDUINO IDE!
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

zekageri, 2022-04-29:
> Did it work before? Maybe you could provide a minimal project to reproduce the issue?

Yes it did work before. I used the following pio ini

```
platform    = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
```

valeros, 2022-04-29:
The `board_build.filesystem = littlefs` option was introduced in v4.0 and not backward compatible.

zekageri, 2022-04-29:
I will test my partition.csv in Arduino IDE too.

BlueAndi, 2022-04-29:
Just as a note, an hour ago I tried it successfully in my project with
```
platform = espressif32 @ ~4.2.0
framework = arduino
board_build.filesystem = littlefs
...
```

zekageri, 2022-04-29:
Can it be that my flash is died? But why does it work with Arduino ide....

BlueAndi, 2022-04-29:
Can you try 40 Mhz for the flash?

Jason2866, 2022-04-30:
Do **NOT** use this 
```
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```
anymore. It is totally outdated and will generate issues.

Delete folder `.platformio` and just do `platform = espressif32`

zekageri, 2022-05-02:
Okay, i will try with 40mhz with the following pio ini:

```C
[env:arduino-esp32]
platform = espressif32
board       = esp32dev ; or wrover??
framework   = arduino
board_build.filesystem = littlefs

board_build.f_cpu   = 240000000L
upload_port     = COM8
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder, colorize, log2file

board_build.flash_size = 16MB
board_build.flash_mode = qio ;dio
board_build.partitions = ./hsh_Partition.csv   ;11 468 800 byte flash 2 621 440 byte app 2x
board_build.f_flash    = 40000000L

build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -mfix-esp32-psram-cache-strategy=memw
                -DCORE_DEBUG_LEVEL=0
                -D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
                -D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
```

zekageri, 2022-05-02:
I did delete .platformio folder, build and uploda with 40mhz flash and using `platform = espressif32`

When the chip tries to boot up it says the flash mode is `DIO` even when my ini file contains `QIO`. Strange.

```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13480
ho 0 tail 12 room 4
load:0x40080400,len:3604
entry 0x400805f0
ets Jul 29 2019 12:21:46
```

zekageri, 2022-05-02:
```C
void setup() {
    Serial.begin(115200,SERIAL_8N1,DEBUG_RX,DEBUG_TX);

    //esp_log_set_vprintf(sysLog);    // Does not seem to work for crash logs
    Serial.println("Hi from PIO 4.2.0!");  // Doesnt even gets to this line!!

    ledAmpWriteQueue = xQueueCreate( 10, sizeof( struct ledsAmperData ) );
    thermCalibWriteQueue = xQueueCreate( 2, sizeof( struct ledsAmperData ) );

    hsh_fileSystem.init(-1,5,FILE_SYS_STACK_SIZE);
    while( !hsh_fileSystem.ready() ){vTaskDelay(50);}

    hsh_modbus.init(-1,3,MODBUS_SYS_STACK_SIZE);
    
    hshDisplay.init(-1,4,DISPLAY_SYS_STACK_SIZE);
    hsh_networkSystem.init(-1,3,NETWORK_SYS_STACK_SIZE);
    hsh_Server.init(-1,3,SERVER_SYS_STACK_SIZE);
    hsh_Performance.init(-1,2,PERFORMANCE_SYS_STACK_SIZE);
    hsh_GeoSystem.init(-1,2,GEO_SYS_STACK_SIZE);
    hsh_timeSystem.init(-1,2,TIME_SYS_STACK_SIZE);
    
    zones.init(-1,3,ZONES_SYS_STACK_SIZE);
}
```

zekageri, 2022-05-02:
Tried with this setup since i have a wrover-e ( 16mb )

```C
[env:esp-wrover-kit]
platform = espressif32
board       = esp-wrover-kit ;esp32dev ; or wrover??
framework   = arduino
board_build.filesystem = littlefs
```

Result is the same boot loop.

zekageri, 2022-05-02:
Okay so it turns out that something is wrong with my partition file. I commented out my custom partition ( which was working fine until this point ) and flashed with the default partition table, and it is turning on. Here is my partition.csv

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x280000,
app1,     app,  ota_1,   0x290000,0x280000,
spiffs,   data, spiffs,  0x510000,0xAF0000,
```

Iam gonna make a different one.

zekageri, 2022-05-02:
With the default 16mb.csv i got this serial message:

```C
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

My flash died?

zekageri, 2022-05-02:
### I got a new module. Erased flash, uploaded sketch, uploaded flash with this pio INI

```
;[env:arduino-esp32]
[env:esp-wrover-kit]
platform = espressif32
board       = esp-wrover-kit ;esp32dev ; or wrover??
framework   = arduino
board_build.filesystem = littlefs

board_build.f_cpu   = 240000000L
upload_port     = COM8
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder, colorize, log2file

board_build.flash_size = 16MB
board_build.flash_mode = dio
board_build.partitions = default_16MB.csv   ;./hsh_NewPartition.csv
board_build.f_flash    = 80000000L

build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -mfix-esp32-psram-cache-strategy=memw
                -DCORE_DEBUG_LEVEL=0
                -D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
                -D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
```

Both esp is a ESP32-Wrover-E ( 16mb flash 8 mb psram )
**Got boot loop on both.** 

### With default_4mb.csv both chip is working,

zekageri, 2022-05-02:
Creating new issue for partition table problem.

closed 2022-05-02 by zekageri