platformio/platform-espressif32 #786: I2C fails after upgrading to 4.2.0

ivancmz, 2022-05-01:
Just upgraded a working project from 3.2.1 to 4.2.0 and the fw wont even start it shows this backtrace:

`ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12776
load:0x40080400,len:3032
entry 0x400805e4
[     3][E][Wire.cpp:319] beginTransmission(): could not acquire lock
[     3][E][esp32-hal-i2c.c:142] i2cWrite(): could not acquire lock

assert failed: xQueueGenericSend queue.c:820 (pxQueue)

Backtrace:0x40083ce5:0x3ffe39000x4008d061:0x3ffe3920 0x40092651:0x3ffe3940 0x4008daba:0x3ffe3a70 0x400ff2e5:0x3ffe3ab0 0x400ff499:0x3ffe3ad0 0x400e8119:0x3ffe3af0 0x400e79fe:0x3ffe3b20 0x400e7e20:0x3ffe3b40 0x400e349b:0x3ffe3bb0 0x400e360e:0x3ffe3bd0 0x400ede39:0x3ffe3bf0 0x4011e20b:0x3ffe3c10 0x400833fe:0x3ffe3c40 0x400791f2:0x3ffe3c90  |<-CORRUPTED
  #0  0x40083ce5:0x3ffe39000 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402`

I removed the I2C initialization and calls and the firmware starts working again, so it's confirmed that the issue is on I2C.

ivancmz, 2022-05-01:
I tryed disabling the locks by adding -DCONFIG_DISABLE_HAL_LOCKS to platformio.ini but now it won't compile:

```
Compiling .pio\build\esp32release\FrameworkArduino\esp32-hal-ledc.c.o
C:/Users/imarquez/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c-slave.c: In function 'i2cSlaveDeinit':
C:/Users/imarquez/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c-slave.c:363:12: error: 'i2c_slave_struct_t' {aka 'struct i2c_slave_struct_t'} has no member named 'lock'
     if(!i2c->lock){
            ^~
C:/Users/imarquez/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c-slave.c: In function 'i2cSlaveWrite':
C:/Users/imarquez/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c-slave.c:380:12: error: 'i2c_slave_struct_t' {aka 'struct i2c_slave_struct_t'} has no member named 'lock'
     if(!i2c->lock){
            ^~
Compiling .pio\build\esp32release\FrameworkArduino\esp32-hal-matrix.c.o
*** [.pio\build\esp32release\FrameworkArduino\esp32-hal-i2c-slave.c.o] Error 1
```

Jason2866, 2022-05-01:
Wrong github for this issue. The problem you encounter is from the Arduino framework NOT from Platformio setup. The correct place would be here [https://github.com/espressif/arduino-esp32](https://github.com/espressif/arduino-esp32)

ivancmz, 2022-05-01:
> Wrong github for this issue. The problem you encounter is from the Arduino framework NOT from Platformio setup. The correct place would be here https://github.com/espressif/arduino-esp32

Thanks Jason2866, this is very confusing, ¿what is this github for then?, ¿How are you certain (or how can I be certain) that it's an Arduino framework issue and not PlatformIO so I won't make the same mistake twice?

I'm opening the issue where you told me and I will close this one. 
Here's the [link](https://github.com/espressif/arduino-esp32/issues/6674) if anyone is interested.

Jason2866, 2022-05-11:
@wgaylord 
```
platform = espressif32 @ 4.2.0
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/2.0.3/esp32-2.0.3.zip
```

wgaylord, 2022-05-11:
That fixed it for sure. Now I can compile my program which didn't even use I2C in the first place.

valeros, 2022-05-16:
Hi guys  ! The latest core v2.0.3 is now available in the dev branch of this platform. Please retest your projects using upstream version of the platform. For example:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
board = esp32dev
monitor_speed = 115200
```

ivancmz, 2022-05-17:
Hey everyone, it turns out in my case the problem was I2C being initialized on a ctor that was probably running before the libraries. You can see the discussion [here](https://github.com/espressif/arduino-esp32/issues/6674). I will give Arduino 2.0.3 a try, my goal for upgrading is to be able to use the certificate bundle, and I was waiting for this.

closed 2022-05-17 by ivancmz