platformio/platform-espressif32 #787: No quotes included in deefine for custom board_build.partitions

fionn-r, 2022-05-02:
### Configuration

**Operating system**: Ubuntu 20.04

**PlatformIO Version** (`platformio --version`):
```
❯ platformio --version
PlatformIO Core, version 6.0.0a2
```
`platform = espressif32@4.2.0`
### Description of problem

Board is esp-wrover-kit

When I set a custom board_build.partitions to a path (e.g. `board_build.partitions = extras/partition_table`), the generated call to g++ doesn't add quotes around `extras/partition_table` and causes a `warning: ISO C++11 requires whitespace after the macro name` for every object compiled.

#### Steps to Reproduce

1. Create a project with a custom board_puild.partitions
2. Build the firmware and check the output logs.

### Actual Results
A lot of `<command-line>: warning: ISO C++11 requires whitespace after the macro name` is printed. Also, building in debug mode you can see that `-DARDUINO_PARTITION_firmware-shared-libs/system/partition_table` is a parameter to g++.

### Expected Results
`-DARDUINO_PARTITION_firmware-shared-libs/system/partition_table` is defined instead as `-DARDUINO_PARTITION_\"firmware-shared-libs/system/partition_table\"`

zekageri, 2022-05-02:
I can confirm this

valeros, 2022-05-04:
Resolved in https://github.com/espressif/arduino-esp32/pull/6681

closed 2022-05-04 by valeros