platformio/platform-espressif32 #788: Boot loop if platform == espressif32 if using default_16MB.csv partition table

zekageri, 2022-05-02:
I upgraded to `Espressif 4.2.0` using PIO.
My chip is ESP32_WROVER_E ( 16 mb flash and 8mb external ram )

My sketch worked fine with custom partition table before. I used this platform and package:

```
platform    = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
```

### Now with this INI:

```
[env:esp-wrover-kit]
platform = espressif32
board       = esp-wrover-kit
framework   = arduino
board_build.filesystem = littlefs

board_build.f_cpu   = 240000000L
upload_port     = COM8
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder, colorize, log2file

board_build.flash_size = 16MB
board_build.flash_mode = dio
board_build.partitions = default_16MB.csv
board_build.f_flash    = 80000000L

build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -mfix-esp32-psram-cache-strategy=memw
                -DCORE_DEBUG_LEVEL=0
                -D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
                -D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY
```

**Now my ESP bootloops with the following message:**

```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12804
ho 0 tail 12 room 4
load:0x40080400,len:3032
entry 0x400805e4
ets Jul 29 2019 12:21:46
```

If i comment out the `board_build.partitions = default_16MB.csv` so the compiler use the default csv file, my board is working fine.

I have tried with different flash_mode, flash frequency and board. Tried to erase the flash, upload empty data folder. Deleted .platformio folder. Nothing helps. If i use default_16MB.csv ( or maybe if it is bigger than 4mb?? ) my esp boot loops.

Tested with two ESP32_WROVER-E boards. Both with external ram and 16mb flash size. The upload monitor confirms that these chips has 16mb flash. ( and i know that for sure because i used the exact same sketch and partition before )

**Uploading output:**

```C
Processing esp-wrover-kit (platform: espressif32; board: esp-wrover-kit; framework: arduino)--------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 (4.2.0) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.20002.0 (2.0.2)
 - tool-esptoolpy 1.30300.0 (3.3.0)
 - tool-mkfatfs 2.0.1
 - tool-mklittlefs 1.203.210628 (2.3)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.19.2
|-- <FS> 2.0.0
|-- <LittleFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <HTTPClient> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <WiFiClientSecure> 2.0.0    
|   |   |-- <WiFi> 2.0.0
|-- <ESPmDNS> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <Ethernet> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <WiFi> 2.0.0
|-- <AsyncTCPSock> 0.0.1
|   |-- <WiFiClientSecure> 2.0.0    
|   |   |-- <WiFi> 2.0.0
|-- <ESP Async WebServer> 1.2.3     
|   |-- <AsyncTCPSock> 0.0.1        
|   |   |-- <WiFiClientSecure> 2.0.0
|   |   |   |-- <WiFi> 2.0.0        
|   |-- <FS> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <ArduinoJson> 6.19.2        
|-- <Update> 2.0.0
|-- <JPEGDecoder> 1.8.0
|   |-- <LittleFS> 2.0.0
|   |   |-- <FS> 2.0.0 
|   |-- <FS> 2.0.0     
|   |-- <SD> 2.0.0     
|   |   |-- <FS> 2.0.0 
|   |   |-- <SPI> 2.0.0
|-- <SPI> 2.0.0        
|-- <TFT_eSPI> 2.3.4
|   |-- <SPIFFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <ErriezDS3231> 2.0.0
|   |-- <Wire> 2.0.0
|-- <Wire> 2.0.0
Building in release mode
Building FS image from 'data' directory to .pio\build\esp-wrover-kit\littlefs.bin
Looking for upload port...
Use manually specified: COM8
Uploading .pio\build\esp-wrover-kit\littlefs.bin
esptool.py v3.3
Serial port COM8
Connecting.....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None        
Crystal is 40MHz
MAC: e0:e2:e6:06:fd:08
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00c90000 to 0x00ffffff...
Compressed 3604480 bytes to 3646...
Writing at 0x00c90000... (100 %)
Wrote 3604480 bytes (3646 compressed) at 0x00c90000 in 13.9 seconds (effective 2074.7 kbit/s)...
Hash of data verified.
```

Thieuzor, 2022-05-03:
I have exactly the same issue. This was also the case in 4.1.0.
When setting the platform fixed at 3.5.0, everything works well:

`[env:esp32dev]
platform = espressif32@3.5.0
board = esp32dev
framework = arduino

board_build.partitions = 16MB.csv
board_build.f_flash = 80000000L
board_build.flash_mode = qio
build_flags =
    -DBOARD_HAS_PSRAM
    -DCORE_DEBUG_LEVEL=5

; Serial Monitor options
monitor_speed = 115200
upload_port = COM7
upload_speed = 921600
`

valeros, 2022-05-03:
First of all, let's verify that this behavior has nothing to do with the framework itself? Any chance you could try Arduino IDE with your custom partition table?

zekageri, 2022-05-03:
Don't have to use custom partition table, just the default_16mb.csv.
On arduinoIDE, there is no option for 16mb.

[image]

zekageri, 2022-05-03:
> I have exactly the same issue. This was also the case in 4.1.0. When setting the platform fixed at 3.5.0, everything works well:
>
> ```
> [env:esp32dev]
> platform = espressif32@3.5.0
> board = esp32dev [...]

But you can not use `board_build.filesystem = littlefs` with 3.5.0 :/

valeros, 2022-05-03:
> On arduinoIDE, there is no option for 16mb SPIFFS or LITTLEFS

Could you please simply back up the original `default_16mb.csv` and replace that default partition file with your custom one?

zekageri, 2022-05-03:
I could do that, but none of them works. I tested with the following csv files:

- default_8MB.csv
- default_16MB.csv
- large_spiffs_16MB.csv
- my_customPartition.csv

**My custom csv looks like this**

```C
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x280000,
app1,     app,  ota_1,   0x290000,0x280000,
spiffs,   data, spiffs,  0x510000,0xAF0000,
```

None of them works. But if i comment out entierly the `board_build.partitions` option so pio will use the default_4mb.csv both of my chips boots up fine. Just there is not enough space for my stuffs. But everything works with the same setup.

valeros, 2022-05-03:
> But if i comment out entierly the board_build.partitions option so pio will use the default_4mb.csv

I was hoping you could try your custom partition in Arduino IDE.

zekageri, 2022-05-03:
Oh sorry. i will try that.

Thieuzor, 2022-05-03:
I've build the sketch in Arduino IDE 1.8.16 and it works fine.
[image]

Tried the same in the new IDE 2.0.0 RC6 and also there it's working.
[image]

(I'm not using LittleFS right now)

zekageri, 2022-05-03:
Nah, iam struggeling with this stupid partitioning in Arduino IDE. I replaced my custom csv with the default csv, put new sizes into the boards.txt and my flash is still 1.5mb. god...

zekageri, 2022-05-03:
oh fine, it works.

**Sketch i tried with on both PIO and Arduino IDE:**

```C
#include "FS.h"
#include <LittleFS.h>

long lastInfoCheckMS = 0;
boolean hasPSRAM = false;
boolean fsMounted = false;
void setup() {
  Serial.begin(115200);
  vTaskDelay(1000);
  Serial.println("Hello from ARDUINO IDE!");
  checkPS_RAM();
  littleFsBegin();
}

void loop() {
  printInfos();
}

void printInfos() {
  if ( millis() - lastInfoCheckMS >= 10000 ) {
    lastInfoCheckMS = millis();
    Serial.println("********* INFOS *********");
    if( fsMounted ){
      Serial.printf("FS - USED BYTES: %d\n", LittleFS.usedBytes());
      Serial.printf("FS - TOTAL BYTES: %d\n", LittleFS.totalBytes());
    }
    Serial.printf("ESP - HEAP SIZE: %d\n", ESP.getHeapSize());
    if ( hasPSRAM ) {
      Serial.printf("ESP - PS RAM SIZE: %d\n", ESP.getPsramSize());
      Serial.printf("ESP - FREE PS RAM SIZE: %d\n", ESP.getFreePsram());
    }
  }
}

void littleFsBegin() {
  if (!LittleFS.begin()) {
    Serial.println("FS - Failed to mount the filesystem");
    fsMounted = false;
  } else {
    Serial.println("FS - Filesystem mounted");
    fsMounted = true;
  }
}

void checkPS_RAM() {
  char * psRam = (char *) ps_malloc (500 * sizeof (char));
  if (psRam == NULL) {
    Serial.println ("PS_RAM: ERROR");
    hasPSRAM = false;
  } else {
    Serial.println ("PS_RAM: OK");
    free(psRam);
    hasPSRAM = true;
  }
}
```

**Serial output on Arduino IDE:** 

```C
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
Hello from ARDUINO IDE!
PS_RAM: OK
FS - Filesystem mounted
********* INFOS *********
FS - USED BYTES: 1617920
FS - TOTAL BYTES: 11468800
ESP - HEAP SIZE: 364592
ESP - PS RAM SIZE: 4192139
ESP - FREE PS RAM SIZE: 4192139
********* INFOS *********
FS - USED BYTES: 1617920
FS - TOTAL BYTES: 11468800
ESP - HEAP SIZE: 364592
ESP - PS RAM SIZE: 4192139
ESP - FREE PS RAM SIZE: 4192139
```

**Partition table used:**

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x280000,
app1,     app,  ota_1,   0x290000,0x280000,
spiffs,   data, spiffs,  0x510000,0xAF0000,
```

valeros, 2022-05-03:
@zekageri could you please compile your sketch in Arduino IDE in verbose mode and upload the log to https://pastebin.com ?

zekageri, 2022-05-03:
Yes, in a minute.

zekageri, 2022-05-03:
Verbose sketch upload:

https://pastebin.com/49WRRLwD

valeros, 2022-05-03:
I see you're using 2.0.3-RC in Arduino IDE, while PlatformIO uses the latest stable 2.0.2. 

Anyway, could you please also upload to https://pastebin.com/ two verbose logs from PlatformIO?

zekageri, 2022-05-03:
PIO Compile

https://pastebin.com/vSD1m1n3

zekageri, 2022-05-03:
Ah sorry it is in the Project tasks. But PIO verbose build does not fit into my terminal. It is tooo long.

valeros, 2022-05-03:
Please upload your app in verbose mode as well. To redirect the output, just open the IDE terminal and run the following command:
```bash
pio run -e esp-wrover-kit -t upload --verbose > upload-log.txt
```

zekageri, 2022-05-03:
okay, thanks

zekageri, 2022-05-03:
How do i know if it is uploaded if i can't see the output

zekageri, 2022-05-03:
oh nice
[image]

valeros, 2022-05-03:
Try to attach it to your comment on Github

zekageri, 2022-05-03:
https://drive.google.com/drive/folders/1FdsFFuFN_0Q7Qb00IG0dGz6FMi1nHqpX?usp=sharing

valeros, 2022-05-03:
OK, thanks. First of all, I see that you're still using different configurations. In PlatformIO you use 4MB and the `QIO` flash mode, while in Arduino it's 16MB and `DIO`. Please adjust your PlatformIO project so it's fully consistent with the Arduino project.

zekageri, 2022-05-04:
What do you mean by 4mb? This is my pio thingy 

[image]

zekageri, 2022-05-04:
Can it be the problem? No matter what board i define, and what flash_size i set up, when compile, the HARDWARE will always be 4mb flash.

```
;[env:esp-wrover-kit]
[env:arduino-esp32]
platform = espressif32

board       = esp32dev ;esp-wrover-kit
framework   = arduino
board_build.filesystem = littlefs

board_build.f_cpu   = 240000000L
upload_port     = COM8
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder

board_build.flash_size = 16MB
board_build.flash_mode = qio
board_build.partitions = large_spiffs_16MB.csv ;./hsh_Partition.csv
board_build.f_flash    = 80000000L

build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -mfix-esp32-psram-cache-strategy=memw
                -DCORE_DEBUG_LEVEL=5
                ;-D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
                ;-D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY

;extra_scripts = mklittlefs/replace_fs.py
```

**COMPILE:**

```C
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)PACKAGES:
 - framework-arduinoespressif32 3.20002.0 (2.0.2)
 - tool-esptoolpy 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
```

zekageri, 2022-05-04:
Result of **(esp32dev board)** `pio run -e arduino-esp32 -t upload --verbose > upload-log.txt` is:
[upload-log.txt](https://github.com/platformio/platform-espressif32/files/8618203/upload-log.txt)

With the following PIO INI:

```C
;[env:esp-wrover-kit]
[env:arduino-esp32]
platform = espressif32

board       = esp32dev ;esp-wrover-kit
framework   = arduino
board_build.filesystem = littlefs

board_build.f_cpu   = 240000000L
upload_port     = COM8
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder

board_build.flash_size = 16MB
board_build.flash_mode = dio
board_build.partitions = large_spiffs_16MB.csv ;./hsh_Partition.csv
board_build.f_flash    = 80000000L

build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -mfix-esp32-psram-cache-strategy=memw
                -DCORE_DEBUG_LEVEL=5
                ;-D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
                ;-D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY

;extra_scripts = mklittlefs/replace_fs.py
```
 
Result of **(esp-wrover-kit board)**  `pio run -e esp-wrover-kit -t upload --verbose > upload-log.txt` is:
[upload-log.txt](https://github.com/platformio/platform-espressif32/files/8618215/upload-log.txt)

With the same PIO INI which is `16MB` configuration and `dio` mode. With `large_spiffs_16mb.csv`

zekageri, 2022-05-04:
I can see that on PIO the HARDWARE is always this: `HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash` but on my ini file it is obvious that i configured 16mb and picked large flash table

valeros, 2022-05-04:
> No matter what board i define, and what flash_size i set up, when compile, the HARDWARE will always be 4mb flash.

Actually, it's an error in your project configuration,  as it should be `board_upload.flash_size = 16MB` instead of `board_build.flash_size = 16MB`

zekageri, 2022-05-04:
### With this configuration:

```
[env:esp-wrover-kit]
;[env:arduino-esp32]
platform = espressif32

board       = esp-wrover-kit ;esp32dev
framework   = arduino
board_build.filesystem = littlefs

board_build.f_cpu   = 240000000L
upload_port     = COM8
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder

board_upload.flash_size = 16MB
board_build.flash_mode = dio
board_build.partitions = large_spiffs_16MB.csv ;./hsh_Partition.csv
board_build.f_flash    = 80000000L

build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -mfix-esp32-psram-cache-strategy=memw
                -DCORE_DEBUG_LEVEL=5
                ;-D CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY
                ;-D CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY

;extra_scripts = mklittlefs/replace_fs.py
```

The hardware is still the same:
[upload-log.txt](https://github.com/platformio/platform-espressif32/files/8618926/upload-log.txt)

zekageri, 2022-05-04:
But the boot loop is gone.

valeros, 2022-05-04:
So, does it work as expected now?

zekageri, 2022-05-04:
Yes, it looks like the problem was that it should be `board_upload.flash_size = 16MB` instead of `board_build.flash_size = 16MB`.
Thank you @valeros, i really appreciate your help. I wonder why did it work before. I have always used this option.
Sorry for my stupidness.

Closing this.

closed 2022-05-04 by zekageri

garageeks, 2022-06-13:
I had a particular project which failed with this bootloop when switched to newest 2.0.3 framework. I've been doing cleaning, restarting, whatever and still had bootloop. Switched to 2.0.2 and it started working.
But the other project with 2.0.3 were fine, I realized only in this one I was using 16MB flash store some pictures in SPIFFS.
So I googled and found this issue.
With 2.0.2 and below, just this line was enough: `board_build.partitions = large_spiffs_16MB.csv`

With 2.0.3 using these lines the bootloop is gone, so the fix is indeed confirmed!
```
board_upload.flash_size = 16MB
board_build.flash_mode = dio
board_build.partitions = large_spiffs_16MB.csv
board_build.f_flash    = 80000000L
```