platformio/platform-espressif32 #790: Wire Library and RTC

goparlenvythelingum, 2022-05-03:
What kind of issue is this?

Using Wire library to control the transfer of data and time using RTClib DS3231 RTC. Experiment using the RX (16) and TX (17) lines. The code I have written works in the Arduino IDE 2.0.0.Uploads. No issues using the Wire and RTC  library. The serial monitor displays the correct time, date, and temperature. Whilst within the Platformio IDE, there are no errors when compiling. It uploads, runs but gets stuck in the while loop to check for the RTC, it reports Can't find the RTC. WOrks absolutely fine in Arduino envirement. Using latest and identical  Libraries for both IDE's
void setup() {
  Serial.begin(115200);
    Wire.begin(SDA1,SCL1);
    Wire.beginTransmission(DS3231_I2C_ADDRESS);
  // start the connexion to the RTC
  if (! rtc.begin()) {
    Serial.println("Couldn't find RTC");
    while (1); // can't go further
 }

Serial.println("Couldn't find RTC6");

  if (rtc.lostPower()) {
   rtc.adjust(DateTime(2020, 10, 22, 13, 37, 22));
    Serial.println("Adjusted time!");
   
  }
 
 rtc.adjust(DateTime(2022, 04, 28, 13, 58, 26));
 Serial.println(" find RTC-Done");
        Wire.endTransmission(); 
    //}
    
    Serial.println("-- Default Test --");
    //delayTime = 1000;
    
    Serial.println();

  Serial.println("RTC date is set");
}

ivankravets, 2022-05-03:
Could you try to set [library_archive = no](https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-archive)?

goparlenvythelingum, 2022-05-03:
Done in Platformio.ini

[env:esp32doit-devkit-v1]

library_archive = no

platform = espressif32

board = esp32doit-devkit-v1

framework = arduino

monitor_speed = 115200

lib_deps =

  adafruit/RTClib @ ^1.12.0

 

Compiled and uploaded and same results. Nothing. Not working

 

 

 

Best Wishes

Goparlen Nanda Vythelingum 

 

Please do not forward this email out of my distribution list, E-mail tracking program in place.

Standup for: Respect, inclusivity, equality and diversity at all times.

No to descrimination

No room for racism.

IMPORTANT NOTICE / DISCLAMER: 

Please note Conscious and unconscious racial bias content is not tolerated nor any other forms of hateful or harmful content and will be reported.

This e-mail (and its attachment(s) if any) is intended for the named addressee(s) only and may be legally privileged or prohibited from disclosure and unauthorised use and under no circunstances can the email be forwarded to any persons via any electronic means without the consent of the owner. It contains information which may be confidential and which may also be privileged.  If you have received this transmission in error please contact the sender

 

This email has been checked for viruses by AVG.
https://www.avg.com

goparlenvythelingum, 2022-05-03:
Done, 
[env:esp32doit-devkit-v1]
lib_archive = no
platform = espressif32
board = esp32doit-devkit-v1

does not work

goparlenvythelingum, 2022-05-05:
any updates on this one?

closed 2022-08-11 by stale[bot]