platformio/platform-espressif32 #793: Debugging does not work in platform-espressif32 release 4.2.0

NRollo, 2022-05-04:
After updating to platform-espressif32 release 4.2.0 the debugger has stopped working. Downgrading to release 3.5.0 makes everything work again. This has also been verified with the espidf-blink project.

The debug console does not show any errors but always end at the same line (see below), no code break at 'app_main()' is performed.
**Debug console:**
```
.
.
.
Info : esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Hardware assisted breakpoint 1 at 0x400d0ac7: file src/blink.c, line 28.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to debug_init_break = thb app_main
PlatformIO: More configuration options → [Redirecting...](https://bit.ly/pio-debug)
```

**Debugger:**
Olimex ARM-USB-OCD-h

**PC:**
Device name	DESKTOP-ACPDJ0C
Processor	11th Gen Intel(R) Core(TM) i7-11700 @ 2.50GHz   2.50 GHz
Installed RAM	16,0 GB (15,7 GB usable)
Device ID	AEB7CB17-0E52-484C-9569-240367E763DD
Product ID	00330-54022-74773-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

**PIO system info**
PlatformIO Core 5.2.5
Python 3.9.8-final.0
System Type windows_amd64
Platform Windows-10
File System Encoding utf-8
Locale Encoding cp1252
PlatformIO Core Directory C:\Users\niels.platformio
PlatformIO Core Executable C:\Users\niels.platformio\penv\Scripts\platformio.exe
Python Executable C:\Users\niels.platformio\penv\Scripts\python.exe
Global Libraries 0
Development Platforms 2
Tools & Toolchains 20

**platform.ini:**
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
framework = espidf
board = nodemcu-32s
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
debug_tool = olimex-arm-usb-ocd-h
debug_speed = 4000
build_flags =
	; https://docs.espressif.com/projects/esp-idf/en/latest/get-started/get-started-wrover-kit.html#rgb-led
	-D CONFIG_BLINK_GPIO=2
```

ashelman-ec, 2022-05-05:
I have the same issue when trying to use "PIO Debug".

If using "PIO Debug (without uploading)" then the debugger will halt at app_main as expected.

**Debugger:**
Tried both `esp-prog` and `jlink`

**PIO system info:**
```
PlatformIO Core             5.2.5
Python                      3.9.10-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp936
PlatformIO Core Directory   C:\Users\n\.platformio
PlatformIO Core Executable  C:\Users\n\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\n\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       1
Tools & Toolchains          14
```

**platform.ini**
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
monitor_speed = 115200
monitor_port = COM15
upload_port = COM15
debug_tool = esp-prog
```

yka8247, 2022-05-06:
same issue here. please look into this.
I've been using v3.5.0 for a while, then I updated directly into 4.2.0 which gave me this issue.
I have tested out other versions, and following versions had no problems with debugging and compiling.
- 3.5.0
- 4.0.0
- 4.1.0
Compile and Monitoring works on v4.2.0, but debug doesn't respond at all
-

NRollo, 2022-05-06:
@valeros - Tried it out, seems to work with both espidf-blink and my own much bigger project (multiple tasks).  :o)

NRollo, 2022-05-06:
I think I accidently closed this issue, not sure I should have done that?

ashelman-ec, 2022-05-07:
> Hi guys, I've just pushed a fix, could you please try to launch a new debug session using the upstream version of the platform?
> 
> ```
> [env:esp32dev]
> platform = https://github.com/platformio/platform-espressif32.git
> framework = espidf
> board = esp32dev
> monitor_speed = 115200
> ```

@valeros Yes, the fix also works for me (for both blink & other project). Thanks; that was fast work!

ashelman-ec, 2022-05-08:
Update:

Actually, the fix does not seem to work completely. Though debugging starts, and one can step through execution, there is some problem with symbol lookup.

When using `platform = https://github.com/platformio/platform-espressif32.git`, hovering over a symbol produces error message "No symbol "..." in current context." and does not display its definition or value:
[image]

When using `platform = espressif32@3.5.0` hovering over symbols gives the expected behavior:
[image]

closed 2026-10-20 by valeros