platformio/platform-espressif32 #796: ESP32-S2 (with Arduino Framework): collect2.exe: error

lloydrichards, 2022-05-10:
Ive been working on the Arduino IDE for the past could months since the ESP32-S2 was not really being supported on PlatformIO with Arduino framework (due to the Arduino 2.0 migration).  I recently checked back and saw that there is now support for Arduino framework and so have come back to try again but so far have not had any luck.

platofrmio.ini
```
[env:esp32-s2-saola-1]
platform = espressif32
framework = arduino
board = esp32-s2-saola-1
monitor_speed = 115200
lib_deps = 
	OneButton
	ArduinoJson@6
	256dpi/MQTT @ ^2.5.0
	googlecloudplatform/Google Cloud IoT Core JWT @ ^1.1.11
	adafruit/Adafruit Unified Sensor @ ^1.1.5
	adafruit/Adafruit SHT31 Library @ ^2.1.0
	adafruit/Adafruit LC709203F @ ^1.3.0
	sparkfun/SparkFun Ambient Light Sensor Arduino Library @ ^1.0.3
	adafruit/Adafruit NeoPixel @ ^1.10.4
	ESP Async WebServer
	ayushsharma82/AsyncElegantOTA @ ^2.2.6
	SPI
```

`pio build`
```
...
Archiving .pio\build\esp32-s2-saola-1\libFrameworkArduino.a
Indexing .pio\build\esp32-s2-saola-1\libFrameworkArduino.a
Linking .pio\build\esp32-s2-saola-1\firmware.elf
c:/users/lr/.platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: cannot open map file C:/Users/lr/Documents/GitHub/mimir³/2.Firmware/.pio/build/esp32-s2-saola-1/firmware.map: No such file or directory        
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s2-saola-1\firmware.elf] Error 1
======================================================== [FAILED] Took 35.96 seconds ========================================================
The terminal process "C:\Users\lr\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```

From what I can tell this is an issue with the toolchain-xtensa-esp32s2 and I've seen threads from 2020 and 2021 referring to adding the platform-package to the .ini but have not had any luck with this.  Since the official docs say that Arduino framework should work, I'm assuming this is either a bug or something I'm doing wrong.  Any help would be appreciated as I so much prefer VSCode to Arduino IDE

lloydrichards, 2022-05-10:
have rerun as it with the verbose tag and got the following: https://pastebin.com/c9g67rGc

lloydrichards, 2022-05-10:
tries changing the ³ to a 3 and got the slightly different error:
```
...
Indexing .pio\build\esp32-s2-saola-1\libFrameworkArduino.a
Linking .pio\build\esp32-s2-saola-1\firmware.elf
c:/users/lr/.platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: .pio/build/esp32-s2-saola-1/lib565/libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x8): undefined reference to `mbedtls_md5_starts'
c:/users/lr/.platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: .pio/build/esp32-s2-saola-1/lib565/libESP Async WebServer.a(WebAuthentication.cpp.o): in function `getMD5(unsigned char*, unsigned short, char*)':
C:\Users\lr\Documents\GitHub\mimir3\2.Firmware/.pio/libdeps/esp32-s2-saola-1/ESP Async WebServer/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s2-saola-1\firmware.elf] Error 1
```

lloydrichards, 2022-05-10:
and changing the . in 2.Firmware to a _ got the same slightly different error:
```
...
Indexing .pio\build\esp32-s2-saola-1\libFrameworkArduino.a
Linking .pio\build\esp32-s2-saola-1\firmware.elf
c:/users/lr/.platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: .pio/build/esp32-s2-saola-1/lib851/libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x8): undefined reference to `mbedtls_md5_starts'
c:/users/lr/.platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: .pio/build/esp32-s2-saola-1/lib851/libESP Async WebServer.a(WebAuthentication.cpp.o): in function `getMD5(unsigned char*, unsigned short, char*)':
C:\Users\lr\Documents\GitHub\mimir3\2_Firmware/.pio/libdeps/esp32-s2-saola-1/ESP Async WebServer/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s2-saola-1\firmware.elf] Error 1
```

lloydrichards, 2022-05-10:
https://pastebin.com/zWTfLMJ8

valeros, 2022-05-10:
> have rerun as it with the verbose tag and got the following

Seems like a problem with the Arduino build script, I'll take a look while updating to v2.0.3

> tries changing the ³ to a 3 and got the slightly different error

The last error regarding `mbedtls_md5_starts` is actually a problem with the `ESPAsyncWebServer` library caused by the migration from Arduino v1 to v2. I guess it's fixed in the master branch, so you'd need to update your `ESP Async WebServer` dependency to point to the latest library revision:
```ini
lib_deps = 
	OneButton
	ArduinoJson@6
	256dpi/MQTT @ ^2.5.0
	googlecloudplatform/Google Cloud IoT Core JWT @ ^1.1.11
	adafruit/Adafruit Unified Sensor @ ^1.1.5
	adafruit/Adafruit SHT31 Library @ ^2.1.0
	adafruit/Adafruit LC709203F @ ^1.3.0
	sparkfun/SparkFun Ambient Light Sensor Arduino Library @ ^1.0.3
	adafruit/Adafruit NeoPixel @ ^1.10.4
	; ESP Async WebServer
        https://github.com/me-no-dev/ESPAsyncWebServer 
	ayushsharma82/AsyncElegantOTA @ ^2.2.6
	SPI
```

lloydrichards, 2022-05-10:
thank you so much! indeed changing the ESPAsyncWebServer worked for building out the image with the changes to ³ to a 3.  I can make this cha get for now with my directory, but am wondering if this is an expected behaviour or something that'll be patched out in 2.0.3?  For now I'm very happy to be able to go back to VSCode for my cpp development 👍

closed 2022-08-11 by stale[bot]

vlkozl, 2024-02-09:
Have the similar issue with Adafruit QT Py ESP32-S2, cannot compile "I2C_Scanner" from "Adafruit TestBed".