platformio/platform-espressif32 #800: Incompatibility with arduino-esp32 v2 ?

guestisp, 2022-05-12:
Guys this is strange. i'm developing an esp32 application and i'm struggling from days with some strangeness.
I've found out that with pio 3.4.0 and arduino-esp32 1.0.6, all works properly.

If i upgrade to arduino 2.0.3 and pio 4.3.0, the captive portal breaks (not sure if i'ts a capitve portal issue, a dns issue or dhcp or whatever, what I know is that android is unable to detect the captive porta and when it's open is veeeeeeeeeeeeery slow).
Same code on arduino 1.0.6 works perfectly (the only change is the platform version in platformio.ini

This is what i've found so far . Non sure if it's a pio issue or an espressif issue or both.

These are the combinations tested and the result.

Platform IO 4.2.0 + arduino-esp32 2.0.3: BROKEN
Platform IO 4.2.0 + arduino-esp32 2.0.2: BROKEN
Platform IO 4.2.0 + arduino-esp32 2.0.1: BROKEN
Platform IO 4.2.0 + arduino-esp32 2.0.0: BROKEN
Platform IO 4.2.0 + arduino-esp32 1.0.6: PERFECT

Platform IO 4.1.0 + arduino-esp32 2.0.3: BROKEN
Platform IO 4.1.0 + arduino-esp32 2.0.1: BROKEN
Platform IO 4.1.0 + arduino-esp32 2.0.0: PERFECT

Platform IO 4.0.0 + arduino-esp32 2.0.0: BROKEN

Platform IO 3.2.0 + arduino-esp32 1.0.6: PERFECT

So, the culprit seems to be a combination between PIO 4.2.0
and arduino-esp32 v2.x or PIO 4.1.0 and arduino-esp32 >= 2.0.1

what happened in these versions ? 
I'm unsure if stay on v1.6 with pio latest versione or using 
arduino v2.0.0 and pio 4.1.0

I think there was some changes on ESP-IDF...

any idea ? code-side i don't have errors or issues, it simply doesn't work. it compiles, run, adn so on, but the captive portal doesn't open (or it's open after 1 or 2 minutes).

guestisp, 2022-05-12:
The device used for building could affect this in some way ? I've found that building the project from my laptop, pio 4.1.0 & arduino 2.0.0 works, building the same from my desktop at work, doesn't work.

The only realiable one, working every time, is with arduino-esp32 1.0.6 (and pio 3.5.0, the latest one with arduino 1.0.6 as default)

valeros, 2022-05-16:
Hi @guestisp ! Apparently, the problem has something to do with the latest Arduino v2.x.

First of all, try to use the latest upstream version of the dev-platform with the latest Arduino v2.0.3, for example
```ini
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
board = esp32dev
```
If it doesn't work either, then please try to reproduce the problem using the Arduino IDE. This way we can discard possible problems with the Arduino framework itself.

guestisp, 2022-05-16:
Hi.
Do you mean the HEAD version of the repo ? Because with 2.0.3 i've tried, it's the first one i've tested.

valeros, 2022-05-16:
OK if you already tested v2.0.3 with PlatformIO, we can skip the step with the upstream version of the platform. Then please try the same code in the Arduino IDE. Does it work there as expected?

guestisp, 2022-05-20:
I could try, but it's A LOT of changes, as ArduinoIDE is not compatible with the folder structure used by PIO and also I have to manually download&install a lot of libraries. Not sure the test would be any useful, as there will be too much changes in the codebase. Would be like comparing apples and oranges.

guestisp, 2022-05-20:
Any other test not involving Arduino ide ?

guestisp, 2022-05-24:
Don't know how, as the issue seems to be only related to WiFi or DNSServer.
Everything else seems to work as expected, the only two things i've seen to be buggy are the wifi or the dns server (I don't know how to check, because the issue is during a dns reply with a wifi connection, so it could be one of the two or both at the same time)

closed 2022-08-11 by stale[bot]