platformio/platform-espressif32 #801: Compile issue with Version 4.2.0 on windows installation

ams-hh, 2022-05-12:
Using VSCode 1.6.7 on Windows with platform-espressif V3.4.0.
No issues with compile.
Wanted to upgrade to new V4.2.0 version.
From then on the following:

Building in release mode
Compiling G:\ANDREE\Projekte\PIO\MScadaWS\.pio\build\esp32\src\MScadaWS-V7.12.cpp.o
Generating partitions G:\Projekte\PIO\MScadaWS\.pio\build\esp32\partitions.bin
**>>>  cc1plus.exe: error: too many filenames given.  Type cc1plus.exe --help for usage
Compiling G:\Projekte\PIO\MScadaWS\.pio\build\esp32\lib0cd\EEPROM\EEPROM.cpp.o**
**>>> cc1plus.exe: error: too many filenames given.  Type cc1plus.exe --help for usage
Compiling G:\Projekte\PIO\MScadaWS\.pio\build\esp32\liba72\FS\FS.cpp.o**
Compiling G:\Projekte\PIO\MScadaWS\.pio\build\esp32\liba72\FS\vfs_api.cpp.o
****>>> cc1plus.exe: error: too many filenames given.  Type cc1plus.exe --help for usage**
**>>> cc1plus.exe: error: too many filenames given.  Type cc1plus.exe --help for usage****

after that on runs forever with no more printout ....

Tried a fresh VSCode/platformio/ESP32-Platform installation on a PC that nerver
had seen VSCode before. Same issue. Could not find anything about it in the net.

Any help for this available?
Besr regards
Andree

Jason2866, 2022-05-14:
Do not use special chars in your path. A `.` in Windows is always a bad idea.
`G:\ANDREE\Projekte\PIO\MScadaWS.pio\build\esp32\src\MScadaWS-V7.12.cpp.o`

ams-hh, 2022-05-14:
Hello Jason2866,
agree, but .... what do you mean? 
The hyphen? The dot?

But what I see now: "MScadaWS.pio" is abosolutely wrong. This is obviously constructed by "the script?" wrong. ".pio" is a automatically inserted subdirectory from VSCode/platformio. But there is missing a "\" before, because the tree is "G:\Projekte\PIO\MScadaWS\.pio\build....."

And: "in Windows is always a bad idea" >>> until the former version this was no issue.

ams-hh, 2022-05-14:
Correction:
..... But there is missing a Backslash before.....

ams-hh, 2022-05-14:
Very strange.
Maybe the missing Backslash is (or better "is not") only here in the text. When typeing "\ ." (with NO space between \ and . then both is printed. When not using the space >> \. , the \ is not printed.

Have looked in the original error text. The \ is there. Sorry.

But then as the only "special char" I see the hyphen. Is this really special???

valeros, 2022-05-16:
Hi @ams-hh ! Is there any non-ascii characters in you system paths? Also, please run the `pio system info` command in the IDE terminal in paste the output here.

ams-hh, 2022-05-16:
Hello Valeros,
thank you for takeing time for me!

As you can see in my platformio.ini:

----------------------------------------------------------
[platformio]
default_envs = esp8266-12-1M.FS
lib_dir = g:\andree\projekte\PIO\Libraries\Common

[env]
framework = arduino
monitor_speed = 115200
upload_speed = 921600
monitor_filters = time, default

[env:esp8266-12-1M.FS]
build_flags = 
  -D ESP12 
  -Ig:/andree/projekte/pio
  -Ig:/andree/projekte/pio/libraries
  -Ig:/andree/projekte/pio/MScadaWS/include/ws
  -Ig:/andree/projekte/pio/MScadaWS/include/appli
;  -Wp, -ftrack-macro-expansion=0
;  -Wp, -fno-diagnostics-show-caret
;platform = espressif8266
platform = espressif8266@2.6.3
board = esp12e
lib_extra_dirs = 
  g:\andree\projekte\PIO\Libraries\ESP8266
board_build.ldscript = eagle.flash.4m1m.ld
board_build.filesystem = littlefs

[env:esp32]
build_flags = 
  -D ESP32
  -Ig:/andree/projekte/pio
  -Ig:/andree/projekte/pio/libraries
  -Ig:/andree/projekte/pio/MScadaWS/include/ws
  -Ig:/andree/projekte/pio/MScadaWS/include/appli
  -D CORE_DEBUG_LEVEL=0
  -Wp, -ftrack-macro-expansion=0
  -Wp, -fno-diagnostics-show-caret
platform = espressif32@3.4.0
;platform = espressif32
board = esp32dev
;board = wemos_d1_mini32
lib_extra_dirs = 
  g:\andree\projekte\PIO\Libraries\ESP32
board_build.filesystem = spiffs
----------------------------------------------------------

in env:esp32 I can switch between the platform's newest version (4.2.0) or the older version
(platform = espressif32@3.4.0). In the first case I get the mentioned errors, in the second case everything is allright. So, if non-ascii characters should be really the issue, then, I suppose, the error should occur in both cases !?

In the following as requested:
PS G:\ANDREE\Projekte\PIO\MScadaWS> pio system info
--------------------------  ------------------------------------------------------
PlatformIO Core             5.2.5
Python                      3.9.12-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\amshh\.platformio
PlatformIO Core Executable  C:\Users\amshh\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\amshh\.platformio\penv\Scripts\python.exe    
Global Libraries            0
Development Platforms       3
Tools & Toolchains          20
--------------------------  ------------------------------------------------------

valeros, 2022-05-17:
Could you please try to move your PlatformIO core directory to the root of your disk partition, for example `C:`? Just add the [core_dir ](https://docs.platformio.org/en/latest/projectconf/section_platformio.html#core-dir) option to your project configuration:
```ini
[platformio]
...
core_dir = C:/.platformio
```
Please note that PlatformIO will redownload all required packages, so it might take some amount of time.

ams-hh, 2022-05-17:
Hello Valeros,
I've made you've proposed. Set core_dir to c:.platformio
Sadly no change, same error

Secondly I moved the project too from the network drive to c:\projekte
And another one no change. Same error.

closed 2022-08-11 by stale[bot]