platformio/platform-espressif32 #804: esp32 Takes too long to read from the sd card

arslan437, 2022-05-16:
I have simple code that reads from the SD card around 500 to 600 bytes but it takes more than 3 seconds some times it take 4 seconds.

But in previous version of the platform it was not taking this long.

valeros, 2022-05-16:
Hi @arslan437 ! Could you please try the upstream version of the dev-platform with the latest Arduino core v2.0.3? For example:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
board = esp32dev
monitor_speed = 115200
```

arslan437, 2022-06-15:
@valeros I have compiled the code with the platform link its much better much but not as fast it is in the 3.5.0 version.

valeros, 2022-06-21:
Does it work better in the Arduino IDE?

arslan437, 2022-06-29:
I have not test on Arduino IDE

valeros, 2022-07-28:
@arslan437  Please retest your sketch in Arduino IDE.

closed 2022-09-20 by stale[bot]