platformio/platform-espressif32 #809: Error building TinyUSB MSC example on ESP32-S2

brentru, 2022-05-18:
The current release (v4.2.0) seems to have a build error when attempting to use TinyUSB's MSC on ESP32-S2. 

Sketch: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/MassStorage/msc_external_flash/msc_external_flash.ino

platformio.ini:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:featheresp32-s2]
platform = espressif32
board = featheresp32-s2
framework = arduino
lib_extra_dirs = ~/Documents/Arduino/libraries
```

Output:
```
> Executing task: platformio run <

Processing featheresp32-s2 (platform: espressif32; board: featheresp32-s2; framework: arduino)
------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/featheresp32-s2.html
PLATFORM: Espressif 32 (4.2.0) > Adafruit ESP32-S2 Feather Development Board
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20002.220503 (2.0.2) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch3
Converting msc_external_flash.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 238 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit SPIFlash @ 3.10.0
|   |-- SPI @ 2.0.0
|   |-- SdFat - Adafruit Fork @ 1.5.1
|   |   |-- SPI @ 2.0.0
|-- Adafruit TinyUSB Library @ 1.10.2
|-- SPI @ 2.0.0
|-- SdFat - Adafruit Fork @ 1.5.1
|   |-- SPI @ 2.0.0
Building in release mode
Compiling .pio/build/featheresp32-s2/src/msc_external_flash.ino.cpp.o
In file included from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/tusb_option.h:176,
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/Adafruit_TinyUSB.h:34,
                 from /Users/brentrubell/Documents/PlatformIO/Projects/220518-122318-featheresp32-s2/src/msc_external_flash.ino:28:
/Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/common/tusb_mcu.h:276:4: warning: #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8" [-Wcpp]
   #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8"
    ^~~~~~~
/Users/brentrubell/Documents/PlatformIO/Projects/220518-122318-featheresp32-s2/src/msc_external_flash.ino:78:1: error: 'Adafruit_USBD_MSC' does not name a type; did you mean 'Adafruit_SPIFlash'?
 Adafruit_USBD_MSC usb_msc;
 ^~~~~~~~~~~~~~~~~
 Adafruit_SPIFlash
/Users/brentrubell/Documents/PlatformIO/Projects/220518-122318-featheresp32-s2/src/msc_external_flash.ino: In function 'void setup()':
/Users/brentrubell/Documents/PlatformIO/Projects/220518-122318-featheresp32-s2/src/msc_external_flash.ino:94:3: error: 'usb_msc' was not declared in this scope
   usb_msc.setID("Adafruit", "External Flash", "1.0");
   ^~~~~~~
Compiling .pio/build/featheresp32-s2/lib0fb/Adafruit_TinyUSB_Library/class/cdc/cdc_device.c.o
[... 46 lines cut ...]
   #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8"
    ^~~~~~~
Compiling .pio/build/featheresp32-s2/lib0fb/Adafruit_TinyUSB_Library/class/vendor/vendor_device.c.o
In file included from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/tusb_option.h:176,
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/class/usbtmc/usbtmc_device.c:69:
/Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/common/tusb_mcu.h:276:4: warning: #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8" [-Wcpp]
   #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8"
    ^~~~~~~
*** [.pio/build/featheresp32-s2/src/msc_external_flash.ino.cpp.o] Error 1
Compiling .pio/build/featheresp32-s2/lib0fb/Adafruit_TinyUSB_Library/common/tusb_fifo.c.o
In file included from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/tusb_option.h:176,
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/class/vendor/vendor_device.c:27:
/Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/common/tusb_mcu.h:276:4: warning: #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8" [-Wcpp]
   #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8"
    ^~~~~~~
In file included from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/tusb_option.h:176,
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/common/tusb_common.h:70,
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/osal/osal.h:34,
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/common/tusb_fifo.c:28:
/Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/common/tusb_mcu.h:276:4: warning: #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8" [-Wcpp]
   #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8"
    ^~~~~~~
======================================== [FAILED] Took 7.16 seconds ========================================
The terminal process "platformio 'run'" terminated with exit code: 1.
```

valeros, 2022-05-19:
Hi @brentru ! Unfortunately, that's expected behavior. The problem is caused by a different approach used in handling project include paths in PlatformIO and Arduino. PlatformIO gives libraries a higher priority and therefore they are included prior to the framework includes. In your case the `tusb_config.h` file is present in both  the `Adafruit_TinyUSB` library and the ESP32 Arduino core. The easiest workaround is to rename the `tusb_config.h` file in the library via an extra script (should be placed in the root of your project):
```ini
[env:featheresp32-s2]
platform = espressif32
board = featheresp32-s2
framework = arduino
lib_extra_dirs = ~/Documents/Arduino/libraries
....
extra_scripts = 
    pre:rename_usb_config.py
```

`rename_usb_config.py`:
```py
import os

Import("env")

adafruit_usb_config = os.path.join(
    env.subst("$PROJECT_LIBDEPS_DIR"),
    env.subst("$PIOENV"),
    "Adafruit TinyUSB Library",
    "src",
    "tusb_config.h",
)

if os.path.isfile(adafruit_usb_config):
    os.rename(adafruit_usb_config, adafruit_usb_config + "_backup")
```

closed 2022-05-19 by valeros

brentru, 2022-05-19:
@valeros  Thank you for the info and clarification. I've placed this script within the root of this project. Unfortunately, I do not see it run within the build output, it presents the same error as above.

brentru, 2022-05-19:
I've manually renamed it and this works perfectly, however, I'd love to get it working with the build script.

valeros, 2022-05-19:
Oh, I guess you didn't declare this library in your `lib_deps` and thus PlatformIO is using one from `~/Documents/Arduino/libraries`. Please try adding the following line to you `platformio.ini`:
```ini
lib_deps = 
    adafruit/Adafruit TinyUSB Library@^1.10.3
```
This way you'll have a standalone copy of `Adafruit TinyUSB Library` in your project isolated from the library used in Arduino IDE.

brentru, 2022-05-20:
Works! Thank you.