platformio/platform-espressif32 #810: Framework-arduinoespressif32 @ 3.20003.0 (2.0.3) breaks bootloader file name resolution

foobarhl, 2022-05-19:
Platform IO 6.0.1

```
[env:esp-wrover-kit]
platform = espressif32 @ ^4
board = esp-wrover-kit
framework = arduino
...
platform_packages = 
	platformio/framework-espidf@^3.40302.0
	platformio/framework-arduinoespressif32@^3.20003.0

```
Having a strange problem with framework-arduinoespressif32 version 3.20003.0 (2.0.3). It seems the bootloader path can no longer be found when uploading a newly built firmware to a esp-wrover-kit:

`esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\x\\.platformio\\packages\\framework-arduinoespressif32\\tools\\sdk\\esp32\\bin\\bootloader__40m.bin'`

Right away it looks as though the “dio” is missing from the filename. I checked boards.txt and see build.boot is being set, and also tried adding build_board_boot=dio to my platformio.ini and do not see it trickling down to the esptool call.

I noticed platformio-build-esp32.py in .platformio/packages/framework-arduinoespressif32/tools has some new logic that might be causing the issue:

```
def get_bootloader_image(variants_dir):
    variant_bootloader = join(
        variants_dir, env.BoardConfig().get("build.variant", ""), "bootloader.bin"
    )
    return (
        variant_bootloader
        if isfile(variant_bootloader)
        else join(
            FRAMEWORK_DIR,
            "tools",
            "sdk",
            "esp32",
            "bin",
            "bootloader_${__get_board_boot_mode(__env__)}_${__get_board_f_flash(__env__)}.bin",
        )
    )
```
whereas versions prior to 3.20003 do not use this method and instead use:
```

    FLASH_EXTRA_IMAGES=[
        ("0x1000", join(FRAMEWORK_DIR, "tools", "sdk", "esp32", "bin", "bootloader_${BOARD_FLASH_MODE}_${__get_board_f_flash(__env__)}.bin")),
        ("0x8000", join(env.subst("$BUILD_DIR"), "partitions.bin")),
        ("0xe000", join(FRAMEWORK_DIR, "tools", "partitions", "boot_app0.bin"))
    ]
```
which works.

valeros, 2022-05-19:
Hi @foobarhl ! That's expected, the latest `platformio/framework-arduinoespressif32` package requires the upstream version of the platform:
```ini
[env:esp-wrover-kit]
platform = https://github.com/platformio/platform-espressif32.git
board = esp-wrover-kit
framework = arduino
...
platform_packages = 
	platformio/framework-espidf@^3.40302.0
	platformio/framework-arduinoespressif32@^3.20003.0
```

closed 2022-05-19 by valeros

foobarhl, 2022-05-19:
@valeros My understanding is that would only be needed for tracking the "staging" version - I'm looking to lock down to a specific version. Can you elaborate more please?

valeros, 2022-05-19:
Using the upstream version of the platform is a temporary measure until a new stable release of the platform is published.