platformio/platform-espressif32 #811: Go dead loop using large partions with lastest realeases (4.2.0 ~ 4.3.0 )- no issue with arduino ide or old version (1.0.1)

luc-github, 2022-05-21:
Hi,
Thank you for the 4.3.0 to support latest esp32-arduino  2.0.3, I have issue that did not happen with older releases neither with arduino IDE. 
I have checked doc and usage seems not changed so I am guessing there is a bug.

I am using unmodified wifiAccessPoint.ino from esp32-arduino sample as MCVE and just add platformio.ini to use 16MB partition because my board has 16MB flash
My board is a FYSECT E4 with 16MB flash, so I use esp32dev board, I have also checked with another board I have : Lolin32Pro 16MB flash.
 
Here the configuration file I use
```
[platformio]
default_envs = esp32dev

[env:esp32dev]
platform = espressif32@4.3.0
;platform = espressif32@1.0.1
board = esp32dev
framework = arduino
monitor_speed = 115200
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.partitions = default_16MB.csv
upload_speed = 115200
monitor_speed = 115200
```

I use the default_16MB.csv from esp32-arduino core, but same behavior happen with default_8MB.csv and ok with default 4MB
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x640000,
app1,     app,  ota_1,   0x650000,0x640000,
spiffs,   data, spiffs,  0xc90000,0x370000,
```

Then with 4.3.0 I go to dead loop reboot after flashing

```
rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
ets Jun  8 2016 00:22:57
```
[WiFiAccessPoint.zip](https://github.com/platformio/platform-espressif32/files/8746140/WiFiAccessPoint.zip)

with old version 1.0.1 it is ok 
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90

Configuring access point...
AP IP address: 192.168.4.1
Server started
```
Is it a bug ? or do I miss new setting that is not clearly documented?
I enclose the MCVE
 [WiFiAccessPoint.zip](https://github.com/platformio/platform-espressif32/files/8746140/WiFiAccessPoint.zip)

valeros, 2022-05-23:
Hi @luc-github ! `v1.0.1` is too old version of the platform, a lot has changed since that release. Did I get that right that your project works just fine in Arduino IDE with the latest core v2.0.3? If so, could you please attach here the verbose compilation output from the Arduino IDE?

luc-github, 2022-05-23:
Hi thank you for your answer - yes 1.0.1 is old so need to update ^_^
here you have
[log.txt](https://github.com/platformio/platform-espressif32/files/8753665/log.txt)

Additionnaly from my tests using these from @Jason2866
```
[env:esp32]
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip
platform_packages = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/728/framework-arduinoespressif32-v4.4_dev-d3c108c910.tar.gz
```

works on 8MB and 16MB flash

Jason2866, 2022-05-23:
I see no reason why the official release from platformio (which does now support core 2.0.3) should not work.

valeros, 2022-05-23:
@luc-github Your log file shows that you're using the default partition `C:\\Users\\luc\\Documents\\Arduino\\hardware\\espressif\\esp32\\tools\\partitions\\default.csv`. Could you please verify that it works with 8 and 16MB partitions as well?

luc-github, 2022-05-23:
yes it is - I have modified MCVE to show Chip size - ensure it use the partitions.csv 
[log (2).txt](https://github.com/platformio/platform-espressif32/files/8754865/log.2.txt)
[WiFiAccessPoint (2).zip](https://github.com/platformio/platform-espressif32/files/8754872/WiFiAccessPoint.2.zip)
and output is : 
```
Configuring access point...
AP IP address: 192.168.4.1
Server started
getFlashChipSize: 16777216
```

luc-github, 2022-05-23:
@Jason2866 sorry my understanding is ESP32dev is the generic esp32 and applying a custom partition is indeed overriding the flash size - do I miss something ?

valeros, 2022-05-23:
@luc-github According to your screenshot, you're still using the default partition with 4MB, but in PlatformIO you're using `default_16MB.csv`. Have you tried using this one:
[image]

Jason2866, 2022-05-23:
`board_build.flash_size = 16MB` is possible as override in platformio.ini
Can you try `board_build.flash_mode = dio` ?

luc-github, 2022-05-23:
Thank you I tested as you mentioned and still no success 
```
[platformio]
default_envs = esp32dev

[env:esp32dev]
platform = espressif32@4.3.0

board = esp32dev
framework = arduino
monitor_speed = 115200
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_size = 16MB
build_flags =   -DCORE_DEBUG_LEVEL=0
board_build.partitions = default_16MB.csv
upload_speed = 115200
```
I even removed all entries:
```
[platformio]
default_envs = esp32dev

[env:esp32dev]
platform = espressif32@4.3.0
board = esp32dev
framework = arduino
monitor_speed = 115200
board_build.flash_size = 16MB
board_build.partitions = default_16MB.csv
upload_speed = 115200
```
and same result - dead loop

Also to confirm issue was flash_size not overloaded I used https://docs.platformio.org/en/latest/boards/espressif32/esp32thing_plus.html#board-espressif32-esp32thing-plus
which has 16MB 
```
[platformio]
default_envs = esp32dev

[env:esp32dev]
platform = espressif32@4.3.0

board = esp32thing_plus 
framework = arduino
monitor_speed = 115200
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
build_flags =   -DCORE_DEBUG_LEVEL=0
board_build.partitions = default_16MB.csv
upload_speed = 115200
```

and it is working 

So it is defintly confirmed that the default saze 4MB is not overloaded by `board_build.partitions = default_16MB.csv` neither `board_build.flash_size = 16MB`

valeros, 2022-05-23:
It should be `board_upload.flash_size = 16MB` not `board_build.flash_size = 16MB`.

luc-github, 2022-05-23:
@valeros `board_upload.flash_size = 16MB` did the trick thank you 
I think it worth a note in https://docs.platformio.org/en/latest/platforms/espressif32.html
@Jason2866 thank you for the help too