platformio/platform-espressif32 #812: Upload filesystem options disappear when using ESP32 arduino core 2.0.3

garageeks, 2022-05-21:
Dear all,
I tried to use the newest release of the ESP32 Arduino core, however filesystem related options ("Upload Filesystem Image", "Erase Flash" etc...) disappear from PlatformIO menu.
Here the menu I see:
[image]

And here is the platformio.ini config I used:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3
```

If I revert to 2.0.1 I see the full menu:
[image]

valeros, 2022-05-23:
Hi @garageeks ! Unfortunately I cannor reproduce the problem with the latest release of the platform. The latest version already supports Arduino v2.0.3, so there is no need to install it from custom branches. Simply use the following default config:
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
```

Please note you might need to delete the outdated instances of the platform here:
```ini
; Windows
c:\Users\YOUR_USER\.platformio\platforms
; Linux
/home/YOUR_USER/.platformio/platforms
```

garageeks, 2022-05-23:
Hi @valeros , thank you for your reply.
I have followed your instructions and indeed it worked. It was probably a fluke of that Frankenstein configuration I used to work with beta versions of the ESP32-arduino SDK

Here's the output of a compile, it states 2.0.3 and all the options in the menu are back.

`CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.0 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3`

closed 2022-05-25 by garageeks