platformio/platform-espressif32 #814: The sdkconfig.h include path has been moved with arduino-esp32 core v2.x and later

Hieromon, 2022-05-23:
Thanks for the espressif32 platform for almost catching up with the esp32 arduino core 2.0 head.

The distribution package's folder structure has been changed to support multiple ESP32 architectures in arduino core 2.0 and later. The folder location of **sdkconfig.h** has been moved from `tools/sdk/esp32/include/config` to `tools/sdk/[ARCH]/qspi_qspi/include` and no longer exists with esp32 arduino core 2.0.2. Furthermore, Furthermore, only the ESP32S3 is special, as there are multiple folder hierarchies leading to **sdkconfig.h**, depending on the type of PSRAM interface embedded in the WROVER module.
e.g. `tools/sdk/esp32s3/[INTERFACE]/include`. INTERFACE := '`opi_opi`' | '`opi_qspi`' | '`opi_qspi`' | '`qpi_opi`' | '`qspi_qspi`'

On the other hand, PIO outputs `tools/sdk/[ARCH]/include/config` to the include path and browse path of **c_cpp_properties.json** for VSCODE generated according to the configuration in **platformio.ini**. Its folder should no longer exist.

As a result, VSCODE IntelliSense warns that `tools/sdk/esp32/include/config could not be found`.
That said, there is no real harm done, so we can leave it alone for a while.

**Environment**
- Windows 10 with VSCODE 1.67.1
- PlatformIO IDE v2.4.3
- PIO Core 6.0.1, Home 3.4.1 
- platform-espressif32@4.3.0

**platformio.ini**

```ini
[platformio]
; default_envs = nodemcuv2
default_envs = nodemcu-32s
; default_envs = esp32-c3-devkitm-1

[env]
framework = arduino
lib_extra_dirs = ~/Documents/Arduino/libraries
lib_ldf_mode = chain+
upload_speed = 921600
monitor_speed = 115200

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
board_build.filesystem = littlefs
monitor_filters = esp8266_exception_decoder

[env:nodemcu-32s]
platform = espressif32
; platform = espressif32 @ 3.5.0
board = nodemcu-32s
board_build.filesystem = littlefs
board_build.partitions = min_spiffs.csv
monitor_filters = esp32_exception_decoder

[env:esp32-c3-devkitm-1]
platform = espressif32
board = esp32-c3-devkitm-1
board_build.filesystem = littlefs
board_build.partitions = min_spiffs.csv
monitor_filters = esp32_exception_decoder
```

**c_cpp_properties.json**:

```
{
  "configurations": [
    {
      "name": "PlatformIO",
      "includePath": [
...
        "C:/Users/hiero/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/config",
...
        "C:/Users/hiero/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/qspi_qspi/include",
        "C:/Users/hiero/.platformio/packages/framework-arduinoespressif32/cores/esp32",
        "C:/Users/hiero/.platformio/packages/framework-arduinoespressif32/variants/nodemcu-32s",
        ""
      ],
      "browse": {
        "limitSymbolsToIncludedHeaders": true,
        "path": [
...
          "C:/Users/hiero/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/config",
...
        "C:/Users/hiero/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/qspi_qspi/include",
        "C:/Users/hiero/.platformio/packages/framework-arduinoespressif32/cores/esp32",
        "C:/Users/hiero/.platformio/packages/framework-arduinoespressif32/variants/nodemcu-32s",
          ""
        ]
      },
      "defines": [
        "PLATFORMIO=60001",
        "ARDUINO_NodeMCU_32S",
        "AC_DEBUG",
        "AC_USE_LITTLEFS",
        "PB_USE_LITTLEFS",
        "HAVE_CONFIG_H",
        "MBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"",
        "UNITY_INCLUDE_CONFIG_H",
        "WITH_POSIX",
        "_GNU_SOURCE",
        "IDF_VER=\"v4.4.1-1-gb8050b365e\"",
        "ESP_PLATFORM",
        "_POSIX_READER_WRITER_LOCKS",
        "ARDUINO_ARCH_ESP32",
        "ESP32",
        "F_CPU=240000000L",
        "ARDUINO=10812",
        "ARDUINO_VARIANT=\"nodemcu-32s\"",
        "ARDUINO_BOARD=\"NodeMCU-32S\"",
        "ARDUINO_PARTITION_min_spiffs",
        ""
      ],
      "cStandard": "c99",
      "cppStandard": "c++11",
      "compilerPath": "C:/Users/hiero/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc.exe",
      "compilerArgs": [
        "-mlongcalls",
        ""
      ]
    }
  ],
  "version": 4
}
```

Hieromon, 2022-05-23:
I also have the espressif32 package coexisting in my environment. Is it conceivable that this could be affecting?

[image]

Jason2866, 2022-05-23:
This probably the reason for. You can delete hidden folder `.platformio` in your user folder. Platformio will rebuild.

Hieromon, 2022-05-24:
@Jason2866 I deleted the .platform folder and everything under it and tried again. But still the `framework-arduinoespressif32/tools/sdk/esp32/include/config` entry is genereated into c_cpp_properties.json. The problem reproduces itself in the following minimal project in my environment. And it does not occur with `platform = espressif32@3.5.0`. 

```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
```

```cpp
#include <Arduino.h>

void setup() {}
void loop() {}
```

Given that the problem is not reproduced in your environment, I assume that my environment has a singularity. So, do you have any clues to proceed diagnosing this problem by myself?
Or should I close the issue and proceed with the discussion within the pio community forum?

valeros, 2022-05-24:
Hi @Hieromon ! I managed to reproduce the issue and it seems it's a common issue for both PlatformIO and Arduino IDE as they are autogenerated from the same source. I guess it might be a good idea to ask the maintainers of the Arduino core whether it's expected behavior.

Jason2866, 2022-05-24:
@Hieromon  I use my own fork for building Arduino (libraries) from IDF.
The changes are minimal, but who knews. Same for Platformio, using my fork since
i had some small changes there too. If you want you can test with:
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip
```

Jason2866, 2022-05-24:
Well the line is there!!
`"/Users/hans/.platformio/packages/framework-arduinoespressif32@src-def32c457c76b4d574c70dfd8f6f1217/tools/sdk/esp32/include/config",`
But it does not generate a warning?!

Jason2866, 2022-05-27:
Will be fixed with PR https://github.com/espressif/arduino-esp32/pull/6798

valeros, 2022-06-21:
Should be fixed in the latest Arduino package.

closed 2022-06-21 by valeros