platformio/platform-espressif32 #815: featheresp32-s2 does not work properly

kuravih, 2022-05-25:
## TLDR
Works on arduino IDE but not on platformio

----

I'm using a [Adafruit ESP32-S2 Feather with BME280 Sensor](https://www.adafruit.com/product/5303). I'm trying to access the sensor using the [i2c scanner](https://playground.arduino.cc/Main/I2cScanner/)

```
#include "Arduino.h"
#include <Wire.h>

void setup() {
  Wire.begin();

  Serial.begin(9600);
  while (!Serial);
  Serial.println("\nI2C Scanner");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) {
    // The i2c_scanner uses the return value of the Write.endTransmisstion to see if a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      Serial.print(address, HEX);
      Serial.println("  !");
      nDevices++;
    } else if (error==4) {
      Serial.print("Unknown error at address 0x");
      Serial.println(address,HEX);
    }
    Serial.print("testing address ");
    Serial.println(address,HEX);
  }
  
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);  // wait 5 seconds for next scan
}

```

The `platformio.ini`:

```
[env:featheresp32-s2]
platform = espressif32
board = featheresp32-s2
framework = arduino
```

The board programs fine but the program does not find any of the sensors.

However using the Arduino IDE, and the board definition (`Tools > Board > ESP32 Arduino > Adafruit Feather ESP32-S2`) the program finds the two i2c sensors on the board as expected.

### Another observation:

When programed through the Arduino IDE the programs prints the `testing address xx` lines all in one quick burst, (finds the two connected devices), then holds 5 seconds before doing another loop iteration. (ie. works as expected)

However when programmed using platformio the `testing address xx` lines print one by one in slow succession and does not find any of the connected devices.

kuravih, 2022-05-25:
BTW, Other sketches such as the blink and neopixel and serial port works fine using both programs.

kuravih, 2022-05-26:
@brentru Thank you so much. I did not spot that section, Curiously it worked out of the box with the arduino ide, and not platfromio, so I just assumed it was a platfromio issue.

Marking as resolved,

closed 2022-05-26 by kuravih

maxgerhardt, 2022-05-26:
The Arduino core has the I2C power up code built into an initialization function that is called by the core before setup() in  [adafruit_feather_esp32s2/variant.cpp](https://github.com/espressif/arduino-esp32/blob/master/variants/adafruit_feather_esp32s2/variant.cpp), there should be no different code than in the Arduino IDE needed. It seems PlatformIO has problems linking this weak function and setting `lib_archive = no` in the `platformio.ini` might "fix" it

maxgerhardt, 2022-05-26:
I've opened a pull request that fixes this as linked above.

brentru, 2022-05-26:
Thanks for supporting this, Max.

kuravih, 2022-05-27:
I tested the fix on hardware and verified it to be working. @maxgerhardt and @brentru Thank you very much for your time.