platformio/platform-espressif32 #817: Does not Boot/Start with 8MB partitions

savejeff, 2022-05-26:
First of all, I'm super happy the new 4.3.0 version is out with final official Arduino support for C3 and S2!

I started testing the new version and I found that it works for the standard 4MB partition no problem. Tested Wifi, BLE, SD Card ...
But I found that when switching to 8MB partition the processor does not start
`board_build.partitions = default_8MB.csv`

I've created a basic hello world example that can be tested on any ESP32 with 8MB or more to demonstrate the problem

[Test_ESP32_430.zip](https://github.com/platformio/platform-espressif32/files/8777450/Test_ESP32_430.zip)

My first guess would be that it is caused by different start addresses, maybe together with the OTA partitions second app.

valeros, 2022-05-26:
Hi @savejeff ! Have you updated your flash memory size accordingly? For example:
```ini
...
board_build.partitions = default_8MB.csv
board_upload.flash_size = 8MB
```

savejeff, 2022-05-26:
Hi @valeros ! 
No i have not. Is this a new option? 

It works with with the `board_upload.flash_size = 8MB` very nice.

Thx!

closed 2022-05-26 by savejeff