platformio/platform-espressif32 #819: Support for NodeMCU-ESP-S3-12K

simplest-circuits, 2022-05-28:
Hello,
i bought [this](https://de.aliexpress.com/item/3256803770599325.html?spm=a2g0o.productlist.0.0.5c4d367e4VdDTs&algo_pvid=6a4de0b8-587a-4a71-88be-0ce4613eac49&algo_exp_id=6a4de0b8-587a-4a71-88be-0ce4613eac49-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000027563668858%22%7D&pdp_npi=2%40dis%21EUR%21%216.62%21%21%21%21%21%402100bddf16537236144785807e319e%2112000027563668858%21sea) board and looking for the support.

Compiling works flawlessly, but in case of uploading I´m getting this error:
[image]

Platformio.ini contains this:
[image]

Jason2866, 2022-06-02:
Board NodeMcu-32s is a ESP32 board! So the error message you get is normal. Flashing ESP32 code on a ESP32-S3 is detected from Esptool.

valeros, 2022-06-21:
Hi @simplest-circuits You are using an inappropriate board ID. Please use the board manifest specified above.

closed 2022-06-21 by valeros

MrReSc, 2022-11-27:
I have created the board.json with the content above. Unfortunately it does not work

```
[env:generic]
platform = espressif32
board = ESP-S3-12K
framework = arduino
```

```
Processing generic (platform: espressif32; board: ESP-S3-12K; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ESP-S3-12K.html
PLATFORM: Espressif 32 (5.2.0) > Espressif Generic ESP32-S3 8M Flash
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5) 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/generic/src/main.cpp.o
In file included from /home/remo/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/sys-include/sys/reent.h:503,
                 from /home/remo/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/newlib/platform_include/sys/reent.h:17,
                 from /home/remo/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/sys-include/stdio.h:60,
                 from /home/remo/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:27,
                 from src/main.cpp:1:
/home/remo/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #include <sdkconfig.h>
          ^~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/generic/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/generic/FrameworkArduino/FirmwareMSC.cpp.o
Compiling .pio/build/generic/FrameworkArduino/FunctionalInterrupt.cpp.o
In file included from /home/remo/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/sys-include/sys/reent.h:503,
                 from /home/remo/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/newlib/platform_include/sys/reent.h:17,
                 from /home/remo/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/sys-include/stdio.h:60,
                 from /home/remo/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:27,
                 from /home/remo/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp:20:
/home/remo/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #include <sdkconfig.h>
          ^~~~~~~~~~~~~
[... 252 lines cut ...]
compilation terminated.
*** [.pio/build/generic/FrameworkArduino/StreamString.cpp.o] Error 1
*** [.pio/build/generic/FrameworkArduino/Tone.cpp.o] Error 1
*** [.pio/build/generic/FrameworkArduino/USB.cpp.o] Error 1
*** [.pio/build/generic/FrameworkArduino/USBCDC.cpp.o] Error 1
In file included from /home/remo/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBMSC.cpp:14:
/home/remo/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBMSC.h:18:10: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #include "sdkconfig.h"
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/generic/FrameworkArduino/USBMSC.cpp.o] Error 1
============================================================================ [FAILED] Took 1.34 seconds ============================================================================

 *  Der Terminalprozess "platformio 'run', '--target', 'upload'" wurde mit folgendem Exitcode beendet: 1. 
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste. 
```

Jason2866, 2022-11-27:
Syntax changed for flash and psram type since summer. 
```
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "memory_type": "qio_qspi"
    },
    "core": "esp32",
    "extra_flags": "-DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=0 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0",
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dio",
    "mcu": "esp32s3",
    "variant": "esp32s3",
    "partitions": "default_8MB.csv"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "espidf",
    "arduino"
  ],
  "name": "Espressif Generic ESP32-S3 8M QIO Flash, QIO PSRAM",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/",
  "vendor": "Espressif"
}
```