platformio/platform-espressif32 #820: esp32-C3 Bluetooth does not connect properly

223222981, 2022-05-28:
# Here is the code I burned

```C++
#include <Arduino.h>
#include <BleKeyboard.h>
#include <WiFi.h>

const int board_btn = 9;
const int board_led_left = 12;
const int board_led_right = 13;
BleKeyboard bleKeyboard("ESP32-Echo", "Espressif", 100); 

void setup()
{
  pinMode(board_btn, INPUT);
  pinMode(board_led_left, OUTPUT);
  pinMode(board_led_right, OUTPUT);
  digitalWrite(board_led_left, LOW);
  digitalWrite(board_led_right, LOW);

  Serial.begin(9600);
  Serial.println("Starting BLE work!");
  bleKeyboard.begin();
}

void loop()
{
  if (bleKeyboard.isConnected())
  {
    if (digitalRead(board_btn) == LOW)
    {
      delay(50);
      Serial.println("A + Enter");
      bleKeyboard.print("A");
      bleKeyboard.write(KEY_RETURN);
      delay(100);
    }
    digitalWrite(board_led_right, HIGH);
  }
  digitalWrite(board_led_right, LOW);

  Serial.println("Waiting 3 seconds...");
  delay(3000);
}
// 串口报错提示
// E (24463) BT_HCI: CC evt: op=0x2022, status=0x2
// E (34735) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
// E (34736) BT_SMP: smp_derive_link_key_from_long_term_key failed

// E (34791) BT_BTM: btm_proc_smp_cback received for unknown device
// E (34972) BT_BTM: BTM_GetSecurityFlags false
```

# Prompt messages for the download process

```bash
Processing esp32-c3-devkitm-1 (platform: espressif32; board: esp32-c3-devkitm-1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (4.3.0) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.0 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32 BLE Keyboard @ 0.3.2
|   |-- ESP32 BLE Arduino @ 2.0.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\esp32-c3-devkitm-1\src\main.cpp.o
Linking .pio\build\esp32-c3-devkitm-1\firmware.elf
Retrieving maximum program size .pio\build\esp32-c3-devkitm-1\firmware.elf
Checking size .pio\build\esp32-c3-devkitm-1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   8.3% (used 27212 bytes from 327680 bytes)
Flash: [======    ]  60.0% (used 786578 bytes from 1310720 bytes)
Building .pio\build\esp32-c3-devkitm-1\firmware.bin
esptool.py v3.3
Creating esp32c3 image...
Merged 2 ELF sections
Successfully created esp32c3 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM11
Uploading .pio\build\esp32-c3-devkitm-1\firmware.bin
esptool.py v3.3
Serial port COM11
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:77:6a:50
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00002fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000d7fff...
Flash params set to 0x022f
Compressed 11744 bytes to 8491...
Writing at 0x00000000... (100 %)
Wrote 11744 bytes (8491 compressed) at 0x00000000 in 0.4 seconds (effective 265.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 437.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 677.0 kbit/s)...
Hash of data verified.
Compressed 817216 bytes to 452414...
Writing at 0x00010000... (3 %)
Writing at 0x0001bf2b... (7 %)
Writing at 0x00028f19... (10 %)
Writing at 0x00033d84... (14 %)
Writing at 0x0003b890... (17 %)
Writing at 0x00041400... (21 %)
Writing at 0x00047c7a... (25 %)
Writing at 0x0004ee02... (28 %)
Writing at 0x0005533e... (32 %)
Writing at 0x0005b5b4... (35 %)
Writing at 0x00062e84... (39 %)
Writing at 0x00069548... (42 %)
Writing at 0x0006ffec... (46 %)
Writing at 0x00076271... (50 %)
Writing at 0x0007c5d3... (53 %)
Writing at 0x00083c11... (57 %)
Writing at 0x0008a4c8... (60 %)
Writing at 0x00091f00... (64 %)
Writing at 0x000988e8... (67 %)
Writing at 0x0009ebab... (71 %)
Writing at 0x000a5828... (75 %)
Writing at 0x000ac4f8... (78 %)
Writing at 0x000b2d1a... (82 %)
Writing at 0x000b8ec6... (85 %)
Writing at 0x000bec80... (89 %)
Writing at 0x000c53e7... (92 %)
Writing at 0x000ccb25... (96 %)
Writing at 0x000d3938... (100 %)
Wrote 817216 bytes (452414 compressed) at 0x00010000 in 13.3 seconds (effective 491.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

# The following is the prompt message output from the serial port

```bash
Waiting 3 seconds...
Waiting 3 seconds...
Waiting 3 seconds...
Waiting 3 seconds...
E (37143) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
E (37144) BT_SMP: smp_derive_link_key_from_long_term_key failed

E (37199) BT_BTM: btm_proc_smp_cback received for unknown device
E (37266) BT_BTM: BTM_GetSecurityFlags false

E (37321) BT_BTM: Device not found

Waiting 3 seconds...
Waiting 3 seconds...
Waiting 3 seconds...
�*)�    �␄�}�))ɥ������
�)91␕�1��Starting BLE work!
Waiting 3 seconds...
Waiting 3 seconds...
Waiting 3 seconds...
Waiting 3 seconds...
Waiting 3 seconds...
E (14969) BT_HCI: CC evt: op=0x2022, status=0x2
Waiting 3 seconds...
Waiting 3 seconds...
E (19949) BT_HCI: CC evt: op=0x2022, status=0x2
```

223222981, 2022-05-28:
This is the only third-party library used in the code

[ESP32-BLE-Keyboard](https://github.com/T-vK/ESP32-BLE-Keyboard)

223222981, 2022-05-28:
This is the debug message I output when I try to connect to Bluetooth via my phone

```bash
E (137346) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 
= 0x00
E (137347) BT_SMP: smp_derive_link_key_from_long_term_key failed

E (137404) BT_BTM: btm_proc_smp_cback received for unknown device
E (137583) BT_BTM: BTM_GetSecurityFlags false

E (137621) BT_BTM: BTM_GetSecurityFlags false

E (137636) BT_BTM: BTM_GetSecurityFlags false

E (137680) BT_BTM: BTM_GetSecurityFlags false

E (137696) BT_BTM: BTM_GetSecurityFlags false

E (137741) BT_BTM: BTM_GetSecurityFlags false

E (137808) BT_BTM: BTM_GetSecurityFlags false

E (137846) BT_BTM: BTM_GetSecurityFlags false

E (137891) BT_BTM: BTM_GetSecurityFlags false

E (137958) BT_BTM: BTM_GetSecurityFlags false

E (137996) BT_BTM: BTM_GetSecurityFlags false

E (138048) BT_BTM: BTM_GetSecurityFlags false

E (138101) BT_BTM: BTM_GetSecurityFlags false

E (138146) BT_BTM: BTM_GetSecurityFlags false

E (138258) BT_BTM: BTM_GetSecurityFlags false

E (138273) BT_BTM: BTM_GetSecurityFlags false

E (138303) BT_BTM: BTM_GetSecurityFlags false

E (138431) BT_BTM: btm_proc_smp_cback received for unknown device
Waiting 3 seconds...
Waiting 3 seconds...
Waiting 3 seconds...
Waiting 3 seconds...
```

Jason2866, 2022-06-02:
This is not a Platformio issue. You should open a issue in [Espressif Arduino Github](https://github.com/espressif/arduino-esp32/issues)

valeros, 2022-06-21:
Please try to compile your project using the Arduino IDE. Feel free to reopen the issue if it works there, but not with PlatformIO.

closed 2022-06-21 by valeros