platformio/platform-espressif32 #821: ESP32-C3 bootloop

jancoow, 2022-05-29:
Hi,

I've an ESP32-C3 (2MB) but I'm unable to flash this through platformIO. It keeps resetting in a bootloop:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d0f62
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d0f62
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
entry 0x403ce000
```

I've tried setting 
`platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip`

But this does not work.

My full config:
```
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.3-rc1/platform-espressif32-2.0.3-rc1.zip
board = esp32-c3-devkitm-1
board_build.filesystem = littlefs
framework = arduino
monitor_speed = 115200
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
board_upload.flash_size = "2MB"
board_upload.maximum_size = 2097152
board_build.flash_mode = dio
board_build.f_flash = 40000000L
```

Jason2866, 2022-06-02:
Please use release Platformio Arduino core 2.0.3 
```
platform = espressif32 @ 4.4.0
```

Jason2866, 2022-06-18:
Have you defined a valid partition scheme? The standard is a 4MB one and will fail with a 2MB device. I see no entry for a custom partition scheme in your platformio.ini

jancoow, 2022-06-20:
I did not. Can you specify how I can set that?

valeros, 2022-06-21:
I guess you need to use the `minimal.csv` partition:
```ini
[env:esp32-c3-devkitm-1]
board = esp32-c3-devkitm-1
board_build.filesystem = littlefs
...
board_build.partitions = minimal.csv
```

thijstriemstra, 2022-06-27:
> I guess you need to use the minimal.csv partition:

This unfortunately doesn't help either, still seeing the bootloop:

```
entry 0x403ce000
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d0f62
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
entry 0x403ce000
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d0f62
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
entry 0x403ce000
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d0f62
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
entry 0x403ce000
```

with this config:

```ini
[env:foobar]
platform = espressif32
board = esp32-c3-devkitm-1
framework = arduino
monitor_speed = 115200
board_build.filesystem = littlefs
board_build.partitions = minimal.csv
board_upload.flash_size = "2MB"
board_upload.maximum_size = 2097152
board_build.flash_mode = dio
board_build.f_flash = 40000000L
```

Jason2866, 2022-06-27:
Can you try `board_build.flash_mode = qio` ?

Jason2866, 2022-07-11:
@hpsaturn The bug is fixed in Arduino core 2.0.4  See comment https://github.com/espressif/arduino-esp32/issues/6862#issuecomment-1175926313

hpsaturn, 2022-07-13:
I have a doubt, the version name of `framework-arduinoespressif32` on PlatformIO it's different to the `arduino-esp32` Github repo, but the entry on the PlatformIO registry of this framework has the same Github repo target, that to say is the same, but with version names different. How I know how Platformio name the versions?

## PlatformIO Package and tools
```python
pio pkg list

Resolving esp32-c3 dependencies...
Platform espressif32 @ 5.0.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20003.220626 (required: platformio/framework-arduinoespressif32 @ ~3.20003.0)
├── tool-cmake @ 3.16.4 (required: platformio/tool-cmake @ ~3.16.0)
├── tool-esptoolpy @ 1.30300.0 (required: platformio/tool-esptoolpy @ ~1.30300.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.7.1 (required: platformio/tool-ninja @ ^1.7.0)
├── tool-openocd-esp32 @ 2.1100.20220411 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
└── toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3)
```
that to say, version 3.2 of 220626. This package is published here:
https://registry.platformio.org/tools/platformio/framework-arduinoespressif32/installation
with repo target:
https://github.com/espressif/arduino-esp32 (last release, 2.0.4)

Whats means `3.20003.220626`? **What hash commit? what version is it? somebody knows?**

Jason2866, 2022-07-14:
`3.20003.220626` is the release version from Espressif Arduino Core 2.0.3
Espressif Arduino Core 2.0.4 is not yet released from Platformio

hpsaturn, 2022-07-14:
thanks for clarify it. But, one question, how you know that? that to say, where I looking for where is the current target of a PlatformIO framework release?

Jason2866, 2022-07-14:
Platformio crew only do releases from release versions

hpsaturn, 2022-08-19:
Hi,

FYI, in the last version of PlatformIO, the issue persist:

```python
Resolving esp32c3 dependencies...
Platform espressif32 @ 5.1.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20004.220818 (required: platformio/framework-arduinoespressif32 @ ~3.20004.0)
├── tool-esptoolpy @ 1.30300.0 (required: platformio/tool-esptoolpy @ ~1.30300.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
└── toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3)
```

hpsaturn, 2022-08-19:
well, the bootloop execution issue described here in the issue. The output is similar:

``` 
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380000
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
```

Jason2866, 2022-08-19:
Which device? Please post your platformio config

hpsaturn, 2022-08-19:
My ini file:

```python
[env]
framework = arduino
upload_speed = 1500000
monitor_speed = 115200
build_flags =
    -D CORE_DEBUG_LEVEL=0
lib_deps =
    https://github.com/hpsaturn/SerialTerminal.git

[esp32_common]
platform = espressif32
board = esp32dev
framework = ${env.framework}
upload_speed = ${env.upload_speed}
monitor_speed = ${env.monitor_speed}
lib_deps = ${env.lib_deps}
build_flags =
    ${env.build_flags}

[env:esp32c3]
extends = esp32_common
platform = espressif32
board = esp32-c3-devkitm-1
framework = arduino
build_src_filter = -<*> +<advanced/>
lib_deps = 
    ${esp32_common.lib_deps}
``` 

**NOTE:** The board is a `M5Stamp C3` that some time ago it worked with this board ID.

Esptool output:

```bash
esptool.py v3.3
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 84:f7:03:27:fe:40
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
```

hpsaturn, 2022-08-19:
The complete source code is here:

[esp32-wifi-cli](https://github.com/hpsaturn/esp32-wifi-cli)

Jason2866, 2022-08-19:
Your code is doing the WDTs. Try the examples from [here](https://github.com/platformio/platform-espressif32/tree/develop/examples). Just tested with the RMT blink. Works.

Jason2866, 2022-08-19:
Here is a C3 bootlog from starting Tasmota (same build as current release of Platformio, only bootlog enabled)
It can be seen that the 2nd stage bootloader correctly switches to mode `qio`
```
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x1278
load:0x403cc710,len:0xb9c
load:0x403ce710,len:0x2b50
SHA-256 comparison failed:
Calculated: f8466c8fb1e87897a6eda5234667ae5bff36fd173f25a4760ddbf329ad720eaa
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403cc710
I (48) boot: ESP-IDF v4.4.3-146-geafe239872 2nd stage bootloader
I (49) boot: compile time 18:46:48
I (49) boot: chip revision: 3
I (49) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (55) qio_mode: Enabling default flash chip QIO
I (59) boot.esp32c3: SPI Speed      : 80MHz
I (63) boot.esp32c3: SPI Mode       : QIO
I (67) boot.esp32c3: SPI Flash Size : 4MB
I (71) boot: Enabling RNG early entropy source...
I (75) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (90) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (97) boot:  2 safeboot         factory app      00 00 00010000 000d0000
I (103) boot:  3 app0             OTA app          00 10 000e0000 002d0000
I (110) boot:  4 spiffs           Unknown data     01 82 003b0000 00050000
I (117) boot: End of partition table
I (120) esp_image: segment 0: paddr=000e0020 vaddr=3c100020 size=375b8h (226744) map
I (127) esp_image: segment 1: paddr=001175e0 vaddr=3fc8cc00 size=02dfch ( 11772) load
I (135) esp_image: segment 2: paddr=0011a3e4 vaddr=40380000 size=05c34h ( 23604) load
I (144) esp_image: segment 3: paddr=00120020 vaddr=42000020 size=f43a4h (1000356) map
I (150) esp_image: segment 4: paddr=002143cc vaddr=40385c34 size=06f38h ( 28472) load
I (159) esp_image: segment 5: paddr=0021b30c vaddr=50000010 size=00010h (    16) load
I (165) boot: Loaded app from partition at offset 0xe0000
I (170) boot: Disabling RNG early entropy source...

00:00:00.001 HDW: ESP32-C3 
00:00:00.008 UFS: FlashFS mounted with 304 kB free
00:00:00.020 CFG: Loaded from File, Count 5
```

hpsaturn, 2022-08-20:
Thanks for your answer and help.

Whats means `RTM Blink`? I don't found a sample whit this name. Don't matter, I tested `arduino-blink` C3 sample, and yes, it don't enter to infinite boot loop, but didn't work too. For confirm that, I changed the sample code to:

```cpp
#include <Arduino.h>

// Set LED_BUILTIN if it is not defined by Arduino framework
#ifndef LED_BUILTIN
  ¦ #define LED_BUILTIN 2
#endif

void setup()
{
  Serial.begin(115200);
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  Serial.println("HIGH");
  // wait for a second
  delay(1000);
  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);
  Serial.println("LOW");
  ¦// wait for a second
  delay(1000);
}
``` 

And the output is similar but without loop, only it is freezes:

```bash
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
``` 
**The weird is that I remember use this board with my code some months ago, something change..**

Jason2866, 2022-08-20:
Really strange!

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
HIGH
LOW
HIGH
LOW
HIGH
LOW
HIGH
LOW
HIGH
LOW
HIGH
LOW
```

sketch
```
#include <Arduino.h>

// Set LED_BUILTIN if it is not defined by Arduino framework
#ifndef LED_BUILTIN
  #define LED_BUILTIN 3
#endif

void setup()
{
  Serial.begin(115200);
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  Serial.println("HIGH");
  // wait for a second
  delay(1000);
  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);
  Serial.println("LOW");
  // wait for a second
  delay(1000);
}
```

hpsaturn, 2022-08-20:
Oh! well, these parameters, `quio` and `esp32-c3-devkitm-1`, aren't in the example directory that you referenced. I tried again, and it works, it show the serial messages. Now I need understand why in my old code it doesn't work and it enter to infinity loop with the same parameters.

Jason2866, 2022-08-20:
Tried all other modes (dio, qout, dout) too. All flash modes worked.

tablatronix, 2022-08-23:
What is the actual issue here?
I cannot get my C3 dev kit to boot at all, its just this loop...

I can get a boot with 
platform = espressif32 @ 5.1.0
but then I get crashes if I try to do anything like WiFi.mode() etc

Jason2866, 2022-08-23:
@tablatronix You have not latest framework. There was a silent update.  Delete your hidden folder `.platformio/.cache`and delete folder `.platformio/packages`. After this it will download latest arduino framework.
This solves the flashing of wrong bootloader. Your device trys to start in mode `qio` That is not possible and will fail
```
Saved PC:0x400492ac
SPIWP:0xee
mode:QIO, clock div:1
```

tablatronix, 2022-08-23:
Fork works! I gotta clean this cache and try official now, thanks!

tablatronix, 2022-08-23:
Looks like this bootloader flash mode also fixes my S2 flash saving issue, sigh

Jason2866, 2022-08-23:
Yeah, using a wrong bootloader places Flash in wrong mode which can results in flash corruption when writing

tablatronix, 2022-09-01:
I cannot get pio release w arduino 2.0.3 to work at all, anyone know how to fix this?
How can I get esp 5.0.0 working ?

Jason2866, 2022-09-01:
core 2.0.3 does not work without changes with the ESP32-C3. But there is no reason to use 2.0.3. Core 2.0.4 has many fixes. The upcoming 2.0.5 will fix the last relevant ones.
Imho 2.0.5 will be a good base for long term projects.

tablatronix, 2022-09-01:
Interestingly I have 2.0.4 working but C3 crashes instantly if i enable wifi, however @Jason2866 branch works perfectly, i see its using IDF 4.4.3 though, so not sure whats wrong with 4.4.1

I am pretty much done working with this at this point, talk about aggrevating

[image]

Jason2866, 2022-09-01:
@tablatronix Which C3 board do you have? There are some C3 boards which have PCB design flaws. The most "famous" example is the Lolin C3 mini. The board is borked.

tablatronix, 2022-09-01:
This is the C3 mini devkit

tablatronix, 2022-09-01:
yup its the devkit!!

Just got it working, i had a G3V01 and its alive!
[image]

hpsaturn, 2022-09-07:
Yes, it is working with this kind of boards, I have the OLED version, and works, I did this video:

[image]

But with the M5Stack C3 and complex code I have this issue..

hpsaturn, 2022-09-08:
I understand that but:

1) My `esp32-c3-devkitm-1` is official board, it was a group of boards that people into the Espressif company give a some developers. You remember this December when Espressif gift this boards? And also this board was working.
2) My M5StampC3 was working in previous versions of Espressif tools, I remember it some months ago. And also is relative new. On the other hand is weird that the basic hello world works but extended code doesn't works.

Jason2866, 2022-09-08:
I never got a free board from anyone... @hpsaturn  From which "extended code" you are talking?

tablatronix, 2022-09-08:
Yes my devkit is the same, directly from espressif, but I think there was a rev 2 cause and these early ones were buggy I think i remember there being something about it, either way newer IDF works it seems.

hpsaturn, 2022-09-08:
> I never got a free board from anyone... @hpsaturn From which "extended code" you are talking?

With a basic code that you suggested, it works, I confirmed here: 
https://github.com/platformio/platform-espressif32/issues/821#issuecomment-1221316826

And my extended code is this:
https://github.com/hpsaturn/esp32-wifi-cli/

For the test you don't need any, because it is a simple CLI, only run and upload normally:  

`pio run -e esp32c3 --target upload`

This code is working on:
- [x] esp32c3 pico OLED
- [x] esp32dev variants included M5Atom
- [ ] M5StampC3
- [ ] esp32-c3-devkitm-1 (The same board in the [photo](https://github.com/platformio/platform-espressif32/issues/821#issuecomment-1234746457) of @tablatronix

Jason2866, 2022-09-08:
Ouuch, the advanced example uses a not possible GPIO in line23 `int LED_PIN = 13;` when mode `qio` is used.
GPIO 12 and GPIO13 are needed for QIO with SPI

hpsaturn, 2022-09-08:
I remember test it without these lines. Let me check again it..

hpsaturn, 2022-09-08:
It is working on M5StampC3 now!  I think that was many variables, toolchain version, dio, and the LED issue. But in the dev-kit it doesn't works. Details from these boards:

**M5StampC3**:

```python
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6ac
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 697c37ed1f62ec376bac84a4be3161309fd96b67171cac136349db0d3925a5d8
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000

SSID is empty, please set a valid SSID into quotes

SerialTerm v1.1.2
(C) 2022, MikO - Hpsaturn
  available commands:
        help            show detail usage information
        setSSID         set the Wifi SSID
        setPASW         set the WiFi password
        connect         save and connect to WiFi network
        list            list saved WiFi networks
        select          select the default AP (default: last)
        mode            set the default operation single/multi AP (slow)
        scan            scan WiFi networks
        status          WiFi status information
        disconnect      WiFi disconnect
        delete          remove saved WiFi network by SSID

        sleep           <mode> <time> ESP32 will enter to sleep mode
        echo            "message" Echo the msg. Parameter into quotes
        setLED          <PIN> config the LED GPIO for blink
        reboot          perform a ESP32 reboot
st> 
```
**esp32-c3-devkitm-1**:

```python
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381976
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6ac
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 697c37ed1f62ec376bac84a4be3161309fd96b67171cac136349db0d3925a5d8
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40001be4  RA      : 0x420680a2  SP      : 0x3fca2a00  GP      : 0x3fc8f000  
TP      : 0x3fc8f5b8  T0      : 0x40057fa6  T1      : 0x40383b3e  T2      : 0xffffffff  
S0/FP   : 0x3fc95e08  S1      : 0x40384016  A0      : 0x3fc900cc  A1      : 0x40383f18  
A2      : 0x4038579e  A3      : 0x40383c5a  A4      : 0x40383c92  A5      : 0x3fcdf4f8  
A6      : 0x40383d28  A7      : 0x40383cfc  S2      : 0x3c08863c  S3      : 0x00000002  
S4      : 0x3fc96000  S5      : 0x3fce0000  S6      : 0x3fce0000  S7      : 0x3fce0000  
S8      : 0x3ff1b000  S9      : 0x3fce0000  S10     : 0x3fcdf8d4  S11     : 0x00000000  
T3      : 0x40384474  T4      : 0x4038420a  T5      : 0x3fc90000  T6      : 0x4203e3a2  
[... 16 lines cut ...]
3fca2b80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca2ba0: 0x27146414 0x01010107 0x00000001 0x000001ff 0x00000000 0x00000000 0xffffffff 0x7fefffff
3fca2bc0: 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3fe00000 0x00000000 0x3ff80000
3fca2be0: 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000 0x3ff00000
3fca2c00: 0x00000000 0x40240000 0x00000000 0x401c0000 0x00000000 0x40140000 0x00000000 0x43500000
3fca2c20: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000160 0xabba1234 0x00000154
3fca2c40: 0x3fca2a70 0x6d09f49d 0x3fc927a4 0x3fc927a4 0x3fca2c40 0x3fc9279c 0x00000002 0x3fca0b54
3fca2c60: 0x3fca0b54 0x3fca2c40 0x00000000 0x00000017 0x3fca1230 0x69666977 0xed7a6500 0xc27194eb
3fca2c80: 0x00fc2efc 0x00000000 0x3fca2c20 0x00000017 0x00000001 0x00000000 0x00000000 0x00000000
3fca2ca0: 0x3fc97554 0x3fc975bc 0x3fc97624 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
3fca2cc0: 0x00000000 0x4207207a 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2ce0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2d00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2d20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2d40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2d60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2d80: 0x00000000 0x00000000 0x00000000 0x00000000 0x69000000 0xbaad5678 0x00000060 0xabba1234
3fca2da0: 0x00000054 0x3fca2da4 0x3fca2da4 0x3fca2da4 0x3fca2da4 0x00000000 0x3fca2dbc 0xffffffff
3fca2dc0: 0x3fca2dbc 0x3fca2dbc 0x00000000 0x3fca2dd0 0xffffffff 0x3fca2dd0 0x3fca2dd0 0x00000000
3fca2de0: 0x00000001 0x00000000 0x1500ffff 0x00000000 0xb33fffff 0x00000000 0xbaad5678 0x00000018

ELF file SHA256: 0000000000000000

Rebooting...
```

Jason2866, 2022-09-08:
Can not reproduce. Using 
```
[env:esp32c3]
extends = esp32_common
board = esp32-c3-devkitm-1
board_build.flash_mode  = dio
build_src_filter = -<*> +<advanced/>
lib_deps = 
	${esp32_common.lib_deps}
```
to free GPIO 12 / 13 for LED usage
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0xf4
load:0x403cc710,len:0x8a0
load:0x403ce710,len:0x216c
SHA-256 comparison failed:
Calculated: 236799dd10e5b883b047aa7266eb6dff7e484fbc6e35190f5e5bc5035caac734
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403cc710

Connecting to Jason_Home_WLAN.....connected!

WiFi SSID .: [Jason_Home_WLAN]
IP address  .: 192.168.2.168
RSSI signal .: -81
MAC Address.: 84:F7:03:3B:08:14
Hostname .: esp32c3-3B0814

SerialTerm v1.1.2
(C) 2022, MikO - Hpsaturn
  available commands:
.help..show detail usage information
.setSSID..set the Wifi SSID
.setPASW..set the WiFi password
.connect..save and connect to WiFi network
.list..list saved WiFi networks
.select..select the default AP (default: last)
.mode..set the default operation single/multi AP (slow)
.scan..scan WiFi networks
.status..WiFi status information
.disconnect.WiFi disconnect
.delete..remove saved WiFi network by SSID

.sleep..<mode> <time> ESP32 will enter to sleep mode
.echo.."message" Echo the msg. Parameter into quotes
.setLED..<PIN> config the LED GPIO for blink
.blink..<times> <millis> LED blink x times each x millis
.reboot..perform a ESP32 reboot
st>
```

Jason2866, 2022-09-08:
The devkit board you have has a very old build date `ESP-ROM:esp32c3-20200918`
Which revision is it? Arduino does support only rev. 3 and newer!

hpsaturn, 2022-09-08:
> The devkit board you have has a very old build date `ESP-ROM:esp32c3-20200918` Which revision is it? Arduino does support only rev. 3 and newer! You can check the revision with esptool.py My fork does support the old C3 from rev. 0 ;-)

```
esptool.py --port /dev/ttyUSB0 chip_id
esptool.py v3.3
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 2)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:42:f4:d0
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-C3 has no Chip ID. Reading MAC instead.
MAC: 7c:df:a1:42:f4:d0
Hard resetting via RTS pin...
```

Jason2866, 2022-09-08:
So verified, you have a board rev which is not supported from Arduino.

Jason2866, 2022-09-08:
@tablatronix It is the same for you not working board. It is a not supported old revision.
You can check with esptool.py. To work with Arduino it has to be rev. 3 or newer.

All open cases/questions in this issue thread solved.
@valeros Issue can be closed.

tablatronix, 2022-09-08:
@Jason2866 how exactly does your fork support this board? Curious

Jason2866, 2022-09-08:
@tablatronix The arduino framework is builded with the sdkonfig set to support the c3 from rev.0.

chegewara, 2022-09-14:
> @tablatronix The arduino framework is builded with the sdkonfig set to support the c3 from rev.0. which is used in the platform

I think its not, anymore. I believe that bootloader is built to support rev1, since esp-idf is no longer supporting esp32 C3 rev0, at least not the esp_wifi component.

Jason2866, 2022-09-14:
right i had a typo ` not` is missing

Jason2866, 2022-09-14:
I did not write it is supported. The setting for arduino is rev. 3

mulorc, 2024-08-28:
Selected an incorrect board,

[env:airm2m_core_esp32c3]
platform = espressif32
board = airm2m_core_esp32c3
framework = arduino
monitor_speed = 115200

tyirvine, 2025-03-13:
I was also experiencing a boot loop. After the program was built and uploaded I would get the following message, which was similar to yours. The message repeated till I closed the process.
```
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf94c
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
```
Even though I set the flash size to the correct size, I didn't set the partition scheme to match it. **I ended up needing to use the `minimal.csv` partition scheme because the chip I'm using only has 2 MB of storage.**

### Viewing All Partitions
You can view all the partition schemes in ⤵︎
```
~/Library/Arduino15/packages/esp32/hardware/esp32/<version>/tools/partitions/
```

### Side Note: Board Configurations
I was also having troubles getting my logs to appear in the serial monitor. I was trying to enable 'USB CDC on Boot' using the `build_flags` but it turns out they weren't being run through there for some reason.

Instead, I had to make a custom board configuration. To do so, create a directory at the root called `boards`. Then make a new `*.json` file and label it. The name you give it will be what PlatformIO uses to identify the board. In your `platform.ini` set the `board` parameter to your custom board configuration.

It's a bit easier using an existing board configuration as a starting point. You can find them all in ⤵︎
```
~/.platformio/platforms/espressif32/boards/
```

I found that when I placed my build flags in the board configuration they worked.
```
        ...
        "extra_flags": [
            "-DARDUINO_USB_MODE=1",
            "-DARDUINO_USB_CDC_ON_BOOT=1"
        ],
        ...
```