platformio/platform-espressif32 #826: Can’t build for ESP32S3 - KeyError: 'toolchain-xtensa-esp32'

marcosp32, 2022-06-02:
I am having issues compiling for a new esp module ESP32-WROOM-1U.

I can build just fine using idf.py build

I am using esp-idf.

This is the output:

```
> Executing task: C:\Users\Someone\.platformio\penv\Scripts\platformio.exe run --target upload --target monitor --environment esp32s3box <

Processing esp32s3box (platform: espressif32; board: esp32s3box; framework: espidf)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32s3box.html
PLATFORM: Espressif 32 (4.4.0) > Espressif ESP32-S3-DevKitC-1
HARDWARE: ESP32S3 240MHz, 512KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.40302.0 (4.3.2)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
KeyError: 'toolchain-xtensa-esp32':
  File "C:\Users\Someone\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 184:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Someone\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Someone\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Someone\.platformio\platforms\espressif32\builder\main.py", line 283:
    target_elf = env.BuildProgram()
  File "C:\Users\Someone\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Someone\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 63:
    env.ProcessProgramDeps()
  File "C:\Users\Someone\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Someone\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 126:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Someone\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Someone\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 346:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\Someone\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 660:
    return method(*args, **kw)
  File "C:\Users\Someone\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Someone\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Someone\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 58:
    TOOLCHAIN_DIR = platform.get_package_dir(
  File "C:\Users\Someone\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 32:
    pkg = self.get_package(name)
  File "C:\Users\Someone\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 29:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "C:\Users\Someone\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 21:
    owner=self.packages[name].get("owner"),
====================================================================================== [FAILED] Took 0.66 seconds ======================================================================================
The terminal process "C:\Users\Someone\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--target', 'monitor', '--environment', 'esp32s3box'" terminated with exit code: 1.
```

Board file:
```
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "memory_type": "qspi_qspi"
    },
    "core": "esp32",
    "extra_flags": "-DBOARD_HAS_PSRAM",
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dio",
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif ESP32-S3-DevKitC-1",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 524288,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html",
  "vendor": "Espressif"
}
```

valeros, 2022-06-02:
Hi @marcosp32 ! ESP-IDF v4.4 is not supported yet and hence S3-based boards cannot be used with this framework.

closed 2022-06-02 by valeros

martinberlin, 2022-11-24:
Hi @valeros 
Any idea if it's supported already?
I'm hitting this same error today 5 months afterwards.

Jason2866, 2022-11-25:
@martinberlin  Compiling for S3 works without any issues.

martinberlin, 2022-11-25:
Can you please point me to a platformio.ini file where an ESP32 S3 board is setup along with esp-idf framework?
Thanks a lot. I've lost a lot of time yesterday and could not make it work yet

Jason2866, 2022-11-25:
@martinberlin  This example is working https://github.com/tasmota/platform-espressif32/tree/Tasmota/205x/examples/espidf-arduino-blink
You may have to change the Gpio for the Led

martinberlin, 2022-11-25:
Ok great will try those. 
When I try like it's on the Platformio examples, get the same as @marcosp32 

```
KeyError: 'toolchain-xtensa-esp32':
  File "/home/martin/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 188:
    env.SConscript("$BUILD_SCRIPT")   ...
```

I would like to understand how to fix that too!

martinberlin, 2022-11-25:
Thanks Jason. Here it goes:

- Python 3.8.10
- PlatformIO 6.1.5 is currently the newest version available (pio upgrade)

Deleted:  rm -fr .platformio

Still opening it again:

```
PACKAGES: 
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - tool-ninja @ 1.7.1 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
KeyError: 'toolchain-xtensa-esp32':
  File "/home/martin/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 188:
    env.SConscript("$BUILD_SCRIPT")
```

Some more details

Distribution:	Ubuntu 20.04.5 LTS

Thanks a lot for trying to help me. Other things like building esp32-arduino stuff goes fine. This happens only when trying to build espidf using esp32s3. If I build with any other environments it works correctly, so there has to be something with this board / MCU configuration. Just pushed the **platformio.ini** file:
https://github.com/martinberlin/IT8951-esp32-epub-reader/blob/main/platformio.ini

Jason2866, 2022-11-25:
Ahh, you are the ePub guy :-)  Now I know why this one looked somehow familiar to me
Tbh, i am out of ideas at the moment. Try my fork, i changed a lot under the hood (using released IDF 4.4.3, toolchain patch5 and some others changes...)
```
[env:it8951_paper]
platform = https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.5/platform-espressif32-v.2.0.5.zip
board = esp32-s3-devkitc-1
framework = espidf
; change microcontroller
board_build.mcu = esp32s3
;upload_port = /dev/cu.SLAB_USBtoUART 
;monitor_port = /dev/cu.SLAB_USBtoUART
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
board_build.partitions = partitions.csv
lib_deps =
  https://github.com/leethomason/tinyxml2.git
build_flags =
  ; This is an M5 EPD display
  -DBOARD_TYPE_M5_PAPER
  ; maximim speed
  -Ofast
  ; needed to stop the png library tying to pull in Arduino.h
  -D__MCUXPRESSO
  ; stop miniz conflicting with zlib which is used by the png library
  -DMINIZ_NO_ZLIB_COMPATIBLE_NAMES
  ; device has PRSRAM
  ; and should be used for ram intensive display work
  -DBOARD_HAS_PSRAM
  ; Setup display format and model via build flags - we're only using part of the epdiy codebase
  ; mainly the drawing code - but these settings will get us to compile and give us the correct 
  ; display size
  -DCONFIG_EPD_DISPLAY_TYPE_ED047TC2
  -DCONFIG_EPD_BOARD_REVISION_LILYGO_T5_47
  ; setup the pins for the SDCard
  -DSD_CARD_PIN_NUM_MISO=GPIO_NUM_13
  -DSD_CARD_PIN_NUM_MOSI=GPIO_NUM_12
  -DSD_CARD_PIN_NUM_CLK=GPIO_NUM_14
  -DSD_CARD_PIN_NUM_CS=GPIO_NUM_4
  ; the adc channel that is connected to the battery voltage divider - this is GPIO_NUM_35
  -DBATTERY_ADC_CHANNEL=ADC1_CHANNEL_7
  ; use SPIFFS - experimental feature
  -D USE_SPIFFS
  ; Logging. Leave enabled for first builds and debugging. Comment to disable
  -D LOG_ENABLED
lib_ignore = 
  touch
  FT6X36
```

Have you transplanted the Lillygo ePaper display to an S3 ?

Jason2866, 2022-11-25:
I will clone the repo (tomorrow) and try to compile the env variant which fails.

martinberlin, 2022-11-25:
I will try your take also. I’m using something similar to M5 but with a PCB that is sold by Goodisplay (called DEXA-C097) my own PCB is based on S3 and it’s just a very basic SPI master with RTC and 5V booster.
Just for fun trying to make a 9.7 inch EPub reader. Thanks a lot for trying to help will try more also until I find what it is, is just that the error, does not give much hints

Jason2866, 2022-11-25:
The `board = esp32-s3-devkitc-1` does not have PSRAM.
S3 Flash / Psram with type QIO / OPI has to be 100% correct, if not device will not boot or bootloop. Whats the flash size and type and PSRAM size and type of your board?
With this let`s define a custom boards json. It is easy with platformio :-)
But this is not the reason for compile fail. Thats the next step of fun with the S3...

martinberlin, 2022-11-25:
Thanks a lot @Jason2866 
The Schematics of my [S3 board are here](https://github.com/martinberlin/H-cinread-it8951/tree/main/Schematic)
With this I can already go forward. Created an [Issue there in my fork](https://github.com/martinberlin/IT8951-esp32-epub-reader/issues/1) so you can track my advance since I have the feeling we are communicating over here and it's starting to be Offtopic for the nice Platformio people.

Jason2866, 2022-11-26:
Solution: The problem was a faulty Pio setup.