platformio/platform-espressif32 #828: ESP32-S3 Built-in debugging support

maxgerhardt, 2022-06-06:
Per https://community.platformio.org/t/esp32-s3-jtag-debugging-over-usb/28182

ESP32S3 boards, similiar to ESP32C3 boards in https://github.com/platformio/platform-espressif32/issues/651#issuecomment-1086863280, have a USB JTAG configuration built into them:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html#selecting-jtag-adapter

PlatformIO should use https://github.com/espressif/openocd-esp32/blob/master/tcl/board/esp32s3-builtin.cfg to support it.

Jason2866, 2022-06-07:
@maxgerhardt (not at home so asking) Is just the change `"openocd_target": "esp32s3.cfg"` in boards.json to `interface/esp_usb_jtag.cfg` not working?

maxgerhardt, 2022-06-07:
Sadly I have no board to test, but essentially I've recommended the user to `esp32s3-builtin.cfg`.

closed 2022-06-14 by valeros

peteDDD, 2022-10-01:
Any further update on this?

Jason2866, 2022-10-01:
Tried and does work. BUT there is general issue in Platformio at the moment (does hit ESP32 with ESP-PROG too). You can not upload and debug. Upload. After that use debug without upload. Works with S3 and C3.

Pixel-Soup, 2022-11-03:
Hi, 
I tried to run Platformio debug on the esp32-s3 with the on-chip usb jtag without success.  Upload using esp-builtin works.  Blink program runs fine after a manual reset.  When I run the debugger, openocd initialization looks fine until it is suppose to execute tbreak setup.  It does not break.  If I press the "pause" button, it reports:   

Error: FreeRTOS maximum used priority is unreasonably big, not proceeding: 231
Program
 received signal SIGINT, Interrupt.
0x40041a76 in ?? ()

I have openocd v0.11.0-esp32-20220706.  Esp-Prog works fine (on a non-S3 board), so I assume the issue is only with the S3 usb jtag.  Has anyone else get theirs working?   Any help would be appreciated.

Hieromon, 2022-11-07:
Is this issue related to #793?
I too have not been able to solve this problem. I think the solution with the updated ```builtin.cfg``` is not enough and it would be undesirable to close the topic to continue the investigation. We need to narrow down the conditions under which the problem occurs.

Pixel-Soup, 2022-11-07:
Unfortunately, my setup fails 100% of the time.  Have you been able to make it work sometimes?
My Esp-prog has fail in the same way, but was corrected with a Platformio update.  I see a few people reporting success with the built-in jtag, but I have not been able to duplicate their results, using their procedures.

maxgerhardt, 2022-11-07:
>After that use debug without upload

So y'all are you using 
```ini
debug_tool = esp-builtin
debug_load_mode = manual
```
in the `platformio.ini`, use the regular "Upload" task, then start debugging?

Pixel-Soup, 2022-11-07:
Yes.  Also, I start the debugger using the Platformio "Debug without uploading" menu, after doing a reset with the boot button pressed to put the chip in the usb jtag mode.

Jason2866, 2022-11-07:
@maxgerhardt @Pixel-Soup  got debugging working WITH uploading in my fork for ESP32-S3 inbuilt jtag.
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.5.2/platform-espressif32-2.0.5.2.zip
```

Pixel-Soup, 2022-11-07:
I just tried that without success.  Here's the messages below.  The tool manager complained about the files sizes not matching, and then downloaded from another mirror site.  I don't know if that is problematic, but it no longer recognizes esp-builtin as a debug tool.

=====================================================
Installing https://github.com/tasmota/arduino-esp32/releases/download/2.0.5.2/framework-arduinoespressif32.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoespressif32@2.0.5 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Downloading  [####################################]  100%

Tool Manager: Warning! Package Mirror: The size (79675042 bytes) of downloaded file 'toolchain-xtensa-esp32-windows_amd64-8.4.0+2021r2-patch5.tar.gz' is not equal to 
remote size (120571045 bytes)
Tool Manager: Looking for another mirror...
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch5 has been installed!

Tool Manager: Installing https://github.com/tasmota/esptool/releases/download/v4.3.1/esptool-4.3.1.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40301.221017 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
DebugInvalidOptionsError: Unknown debug tool `esp-builtin`. Please use one of `cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa` or `custom`:

Jason2866, 2022-11-07:
@Pixel-Soup Weird, since i use `esp-builtin`. Delete the hidden folder `.platformio`.
So Platformio installs everything fresh and no old stuff is left.

Pixel-Soup, 2022-11-08:
I reinstalled Platformio and managed to get esp-builtin recognized again.  Unfortunately, debugging still didn't work.  The debugger runs, but seems to put the breakpoint where the code isn't running.   
Can you clarify what the issue is?   I didn't see any obvious settings in the Tasmota board jsons which I could incorporate to possibly fix the problem.  

So far, the best combination for the S3 is  platform = espressif32.  The code uploads over the builtin jtag and runs.  When I used the Tasmota branch, it uploads, but no longer ran.

Jason2866, 2022-11-08:
Mhhh
[image]

Pixel-Soup, 2022-12-14:
I was eventually able to make it work on a "bare" esp32-s3-wroom, using the standard version, "platform = espressif32".  The main issue was related to my board definition file enabling the CDC usb serial port.  This conflicted with the usb jtag.  It required a tedious sequence to exorcise the windows driver that attaches itself when the esp32 usb is detected.

davidtheITguy, 2022-12-18:
> I was eventually able to make it work on a "bare" esp32-s3-wroom, using the standard version, "platform = espressif32". The main issue was related to my board definition file enabling the CDC usb [...]

Any chance you can post your steps?

Hieromon, 2022-12-18:
Various success stories have been reported, but in the end, it does not seem to be possible to start debugging in a stable manner. It does not break at setup(). I am not clear on the stable debug start procedure and environment.

[image]

Pixel-Soup, 2022-12-18:
@davidtheITguy
If you're using the esp32-s3-wroom module, then you can copy my custom board def into the platformio board folder.  In my case, it's    c:\.platformio\platforms\espressif32\boards      
Modify your platformio.ini to incorporate the following:
platform = espressif32
board = CUSTOM_S3
framework = arduino
upload_protocol = esp-builtin 
debug_tool = esp-builtin
debug_speed = 12000
build_type = debug

Assuming you have the wroom module wired correctly and running the latest version of the platformio esp32, you can build and upload via the usb jtag with the arrow icon as usual.  Then use "debug without uploading" to run the debugger. 

@Hieromon 
"tbreak setup" doesn't appear to work consistently.  However, it does not matter.  The debugger will stop in app_main just before running setup.  You can use the gui to add a break point at setup.  When you press the "continue" button, execution should hit the breakpoint.  Then you can debug as usual.

[custom_S3.zip](https://github.com/platformio/platform-espressif32/files/10254558/custom_S3.zip)

davidtheITguy, 2022-12-18:
Will try and report back thank you!

———————
David Eisner
240.994.5900

davidtheITguy, 2022-12-19:
@Pixel-Soup Ok so I followed all the directions, used the custom board profile (thank you again), updated the platformio file and it ALMOST worked on my test blink sketch.

After uploading and then starting debugging (without uploading), it appears that the internal debug probe is stuck in a loop just before it would try to hit the breakpoint (or stop in setup):

**loops forever:**
`Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.

[esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Core was reset.

Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.

[esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Core was reset.

Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
`

Steps I've checked:
1. probe USB cable wired correctly and tested (recognized)
2. Zadig driver installed
3. Upload and monitoring works via separate USB port
4. code works

The dev board I am using is an ESP32-S3-WROOM-1 **N8R8**

I'm wondering aloud if one of the parameters in the custom board definition you provided doesn't work on
the N8S8 DevKit board...

Anyway I'll keep plugging away and see if I can get to the bottom of it. Thanks again for pointing me in the right direction

Pixel-Soup, 2022-12-19:
Try using       board = esp32-s3-devkitc-1      in platformio.ini.  It supports 8M of flash.
If that doesn't work, please post the first 30 lines of startup messages from the debug console.  And your complete platformio.ini.
View the output console as well.  It will show a code that indicate the reason for the resets.  That can provide a clue.

davidtheITguy, 2022-12-19:
Ok got it to work (as advertised). It appears the issue came down not to board definition but to the `upload_protocol` setting in my platformio.

Generally, it appears that I needed to be very explicit about separating the upload/monitor ports from the internal JTAG interface. I am using two USB ports/cables: 1 for the UART upload and monitoring and 1 for the internal JTAG probe. 

Here is my working platformio:

```
[env:esp32dev]
platform = espressif32
;board = esp32dev
;board = esp32-s3-devkitc-1
board = ESP32-S3-N8R8_custom   ; modified custom def to correct max flash size for my S3 N8R8
framework = arduino
debug_tool = esp-builtin
;upload_protocol = esp-builtin               <-- this doesn't work when enabled!!!
;debug_init_break = tbreak setup
build_type = debug
;debug_speed = 500
monitor_speed = 115200
upload_port = COM3               ; <-- USB cable to board's UART
monitor_port = COM3             ; <-- USB cable to board's UART
```

@Pixel-Soup Thank you again for your help. Hope all this helps others.

Jason2866, 2022-12-19:
Seems, it is a Windows issue. It works to upload and debug via `esp-builtin` on Mac

Pixel-Soup, 2022-12-19:
Glad it's working!  
Upload using esp-builtin works for me on Windows.  My bread board does not have any connection to the hardware serial port, just usb d+/d- and 3v3 power.  
The jtag speed defaults to 40Mhz, far too fast for the usb jtag.  "Debug_speed =12000" sets it to a more reasonable value.  Otherwise, I get inconsistent results in both uploading and debugging. 
If only there was a faster way to upload.  It takes 3 seconds to  rebuild, but 17 seconds to upload.  =)

[image]

Hieromon, 2022-12-20:
@Pixel-Soup Thanks for your advice. However, my problem is that the breakpoint in the setup function doesn't work, and it still hasn't improved. It doesn't matter if tbreak setup is alive or not, it doesn't stop at app_main in the first place. That's what I mean by "unstable".
And it does not mean that the upload process indicated by upload_protocol is incomplete. In my environment, uploading works fine with both upload_protocol esp-tool and esp-builtin in the following platformio.ini. That is different from the situation you have encountered.

The following screenshots shows the situation I am encountering. It is somewhat redundant, but I dared to show it in the hope that the experts who are looking here will grasp the situation. In all situations, platform.ini is assumed to be the same in all cases as follows:

**Environment:**

```
pio system info
--------------------------  ------------------------------------------------------
PlatformIO Core             6.1.5
Python                      3.9.10-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp932
PlatformIO Core Directory   C:\Users\hiero\.platformio
PlatformIO Core Executable  C:\Users\hiero\.platformio\penv\Scripts\platformio.exe
Global Libraries            0
Development Platforms       4
Tools & Toolchains          25
--------------------------  ------------------------------------------------------
```

```
pio pkg list
Resolving esp32-s3-devkitc-1 dependencies...
Platform espressif32 @ 5.2.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20005.220925 (required: platformio/framework-arduinoespressif32 @ ~3.20005.0)
├── tool-cmake @ 3.16.4 (required: platformio/tool-cmake @ ~3.16.0)
├── tool-esptoolpy @ 1.40201.0 (required: platformio/tool-esptoolpy @ ~1.40201.0)
├── tool-idf @ 1.0.1 (required: platformio/tool-idf @ ~1.0.1)
├── tool-mconf @ 1.4060000.20190628 (required: platformio/tool-mconf @ ~1.4060000.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.9.0 (required: platformio/tool-ninja @ ^1.7.0)
├── tool-openocd-esp32 @ 2.1100.20220706 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
├── toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3)
└── toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3)
```

**platformio.ini:**

```ini
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
framework = arduino
build_type = debug
debug_speed = 12000
debug_tool = esp-builtin
debug_init_break = tbreak setup
upload_protocol = esp-builtin
; upload_protocol = esptool
upload_speed = 921600
monitor_port = COM9
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
```

#### Step 1

Build in debug mode, upload it. Then set break point at `setup()` and `loop()` And run PIO Debug.
Build in debug mode and upload. Then set breakpoints in setup() and loop() and run PIO Debug.
But, `tbreak` is ignored and the CPU continues to run. At first glance, it looks like breakpoints are not working. 

[image]
[... 331 lines cut ...]
[New Thread 1070550176]
[New Thread 1070538140]
[New Thread 1070543424]
[New Thread 1070536440]
[Switching to Thread 1070548776]
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001

Thread 1 "main" hit Breakpoint 9, app_main () at C:/Users/hiero/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:69
69	    loopTaskWDTEnabled = false;
```

I don't think this phenomenon is related to debugging speed or CDC driver imperfection. If the debugging speed is too fast than the adapter's clock, it would induce another error. For example, the following:

```
Error: [esp32s3.cpu0] DSR (8080CC19) indicates DIR instruction generated an overrun!
Warn : [esp32s3.cpu0] Failed reading 32 bytes at address 0x420214d4
```

To me, the cause of this problem appears to be more basic. It could be something as basic as an incorrect address map for the symbol table that GDB relies on.
However, if this phenomenon occurs only on Windows, it may be due to OpenOCD porting to Windows. Might be a little deep-rooted...

davidtheITguy, 2022-12-20:
@Jason2866 quick follow up from me on my previous success on this. It turns out that on Windows I was able to get the programming upload to work as well. The issue for me was being connected to the ESP32 UART via a separate USB cable and also specifying upload ports in the platformio. Once I removed that and only used the internal JTAG probe, all is working!

Pixel-Soup, 2022-12-20:
@Hieromon 
I think the debugger is not allocating the 2 hardware breakpoints correctly during startup.   Try deleting all breaks, especially the "tbreak setup" in platformio.ini.  Then manually set a breakpoint at setup().  That should cause it to break in app_main() where you can then press "continue" and run to in setup();

zackees, 2022-12-23:
On windows 10, I get this error:

```
DebugInvalidOptionsError: Unknown debug tool `esp-builtin`. Please use one of `cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa` or `custom`:
```

Here is my platformio

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:WT32-SC01]
platform = espressif32
board = wt32-eth01  # Note that to use psram board must use esp32dev
debug_speed = 12000
debug_tool = esp-builtin
debug_init_break = tbreak setup
debug_load_mode = manual
;upload_protocol = esp-builtin
build_type = debug
framework = arduino
upload_speed = 921600
monitor_speed = 115200
monitor_filters = 
	default
	esp32_exception_decoder
;board_build.f_flash = 80000000L
board_build.f_cpu = 240000000L
build_flags =
  ;-O3
	;-ffast-math
  -DCORE_DEBUG_LEVEL=5
	-DBOARD_HAS_PSRAM
	-DUSE_STACK_HACK_RECYCLE_ALLOCATION_SINGLE_THREADED=1
	-mfix-esp32-psram-cache-issue
	-I src/
	-I src/third_party/arduino-liblame
	-I src/third_party/snprintf
	-DLAME_NO_LOGGING
	-DLAME_LOGGING_ACTIVE=0
;build_flags =
;  -DUSER_SETUP_LOADED=1
;  -DST7796_DRIVER=1
;  -DTFT_WIDTH=480
;  -DTFT_HEIGHT=320
;  -DUSE_HSPI_PORT=1
;  -DPIN_SDA=18
;  -DPIN_SCL=19
;  -DTFT_MISO=12
;  -DTFT_MOSI=13
;  -DTFT_SCLK=14
;  -DTFT_CS=15
;  -DTFT_DC=21
;  -DTFT_RST=22
;  -DTFT_BL=23
;  -DLOAD_GLCD
;  -DLOAD_FONT2
;  -DLOAD_FONT4
;  -DLOAD_FONT6
;  -DLOAD_FONT7
;  -DLOAD_FONT8
;  -DLOAD_GFXFF
;  -DSPI_FREQUENCY=27000000
;  -DSPI_READ_FREQUENCY=20000000
;  -DSPI_TOUCH_FREQUENCY=2500000
;  -DSMOOTH_FONT
;lib_deps =
;    Bodmer/TFT_eSPI
```

davidtheITguy, 2022-12-23:
Hey @zackees think your problem is that your wt32-eth01 board doesn't have an internal JTAG debugger. You need to use an external probe. From Platformio docs: _WT32-ETH01 Ethernet Module does not have on-board debug probe and IS NOT READY for debugging._

zackees, 2022-12-23:
Thanks! I'm actually using a WT32-SCO board and this is the closest thing I could find to it.

Do you know if the WT32-SCO has an internal JTAG debugger.

Thanks!

Jason2866, 2022-12-23:
Internal debugging possibilties do have ONLY S3 and C3. All other MCUs do not support!

Pixel-Soup, 2022-12-23:
The usb port on the WT32-SCO cannot be used for JTAG debug.   If you want debugging, get an ESP-PROG unit from Aliexpress and solder a jtag header on to your board.

Pixel-Soup, 2022-12-24:
@zackees   I just noticed a Wt32-sc01 Plus version that has a builtin jtag.  Which version do you actually have?

zackees, 2022-12-24:
I am currently using the non Plus version.

tyeth, 2023-02-15:
> On windows 10, I get this error:
> 
> ```
> DebugInvalidOptionsError: Unknown debug tool `esp-builtin`. Please use one of `cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa` or `custom`:
> ```

looks similar to what I experienced, the docs were out of date with latest, just use `esp-bridge`
 https://github.com/platformio/platform-espressif32/issues/969

maxgerhardt, 2023-02-15:
No. esp-bridge and esp-builtin are two different things. 

https://github.com/platformio/platform-espressif32/blob/764527eb03c0e937c6014b7f03a4131a03c50188/platform.py#L206-L209

If the debug tool is unknown you need to **update** your platform version, e.g. using the CLI (https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and
```
pio pkg update -g -p espressif32
```

tyeth, 2023-02-15:
> No. esp-bridge and esp-builtin are two different things.
>
> https://github.com/platformio/platform-espressif32/blob/764527eb03c0e937c6014b7f03a4131a03c50188/platform.py#L206-L209
>
> If the debug tool is [...]

Ah, legend! I did update also :)

tranvnhan, 2023-05-03:
> @Hieromon I think the debugger is not allocating the 2 hardware breakpoints correctly during startup. Try deleting all breaks, especially the "tbreak setup" in platformio.ini. Then manually set a [...]

This worked for me. After removing the "break setup" in platform.ini, and manually set the break point at setup(), I can debug normally, and it can actually break in app_main().
Thanks for your comment!

BlackOilCoder, 2023-11-13:
I've tried to follow this thread (along with others).  I'm stuck on my TinsyS3 board.  I believe I sorted the USB driver issue out, but perhaps not.  For me, when I try to upload and debug, it is as if the code never runs and none of the debugging work.

Platformio.ini
`[env:um_tinys3]
platform = espressif32
board = um_tinys3
framework = arduino

build_type = debug
debug_speed = 12000
debug_tool = esp-builtin

upload_speed = 921600
upload_port = COM3
monitor_port = COM3
monitor_speed = 115200`

Debug Console Output:
`undefinedC:\Users\mattd\.platformio\packages\toolchain-xtensa-esp32s3\bin\xtensa-esp32s3-elf-gdb.exe: warning: Couldn't determine a path for the index cache directory.
Reading symbols from c:\Users\mattd\OneDrive\(1) Personal\Hobby\Software\Blink\.pio\build\um_tinys3\firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 12000 kHz

Info : tcl server disabled
Info : telnet server disabled
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: serial (DC:54:75:CB:02:4C)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 13333 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Error: Unexpected OCD_ID = 00000000
Error: Unexpected OCD_ID = 00000000
Error: Unexpected OCD_ID = 00000000
Error: Unexpected OCD_ID = 00000000
Warn : target esp32s3.cpu1 examination failed
Error: Unexpected OCD_ID = 00000000
Error: Unexpected OCD_ID = 00000000
Info : starting gdb server for esp32s3.cpu0 on pipe
Info : accepting 'gdb' connection from pipe
Info : [esp32s3.cpu0] Target halted, PC=0x40041A76, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 53 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 161 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 53 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 161 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 164 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 53 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 161 KB
[... 218 lines cut ...]
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
[esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu1] requesting target halt and executing a soft reset
[esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Hardware assisted breakpoint 1 at 0x42002384: file C:/Users/mattd/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp, line 56.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = thb app_main`
PlatformIO: More configuration options -> https://bit.ly/pio-debug
`

I'm thinking maybe the undefined path to GBD and the libusd_open() failure is my issue, but I'm at a loss to solve.

commit f84dc9a3 (2022-06-14): Add ESP-based debug probes

Resolve #828