platformio/platform-espressif32 #832: Adafruit boards using tinyuf2 are missing partitions-4MB-tinyuf2.csv in boards.json

Jason2866, 2022-06-12:
which is in folder https://github.com/espressif/arduino-esp32/blob/master/variants/adafruit_feather_esp32s2/partitions-4MB-tinyuf2.csv

valeros, 2022-06-21:
I believe it's just renamed `partitions.csv` currently available in 2.0.3:
[image]

closed 2022-06-21 by valeros

Jason2866, 2022-06-21:
No, it is not a renamed partitions.csv
```

# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
# bootloader.bin,,          0x1000, 32K
# partition table,          0x8000, 4K

nvs,      data, nvs,      0x9000,  20K,
otadata,  data, ota,      0xe000,  8K,
ota_0,    0,    ota_0,   0x10000,  1408K,
ota_1,    0,    ota_1,  0x170000,  1408K,
uf2,      app,  factory,0x2d0000,  256K,
ffat,     data, fat,    0x310000,  960K,
```

valeros, 2022-06-21:
Got it, will be updated once a new stable Arduino core is released.

Jason2866, 2022-06-21:
Maybe "scanning" the Arduino `variants` dir after a *.csv is a way to do (in `platformio-build.py`)?

valeros, 2022-06-21:
Related PR https://github.com/espressif/arduino-esp32/pull/6887 has been submitted.