platformio/platform-espressif32 #833: Upload error with esp32-wroom-32

michaelwebfischer, 2022-06-15:
Hi!

I'm not sure if this is the right place to report my issue. If not so, maybe someone can point me in the right direction.

On uploading to a ESP32-WROOM-32 PlatformIO shows:
`Serial port /dev/ttyUSB0
Connecting.....
Traceback (most recent call last):
  File "/home/fimi/.platformio/packages/tool-esptoolpy/esptool.py", line 468, in command
    p = self.read()
  File "/home/fimi/.platformio/packages/tool-esptoolpy/esptool.py", line 413, in read
    return next(self._slip_reader)
  File "/home/fimi/.platformio/packages/tool-esptoolpy/esptool.py", line 3571, in slip_reader
    read_bytes = port.read(1 if waiting == 0 else waiting)
  File "/home/fimi/.platformio/penv/lib/python3.10/site-packages/serial/serialposix.py", line 595, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/fimi/.platformio/penv/lib/python3.10/site-packages/serial/serialposix.py", line 398, in _reconfigure_port
    orig_attr = termios.tcgetattr(self.fd)
termios.error: (5, 'Input/output error')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/fimi/.platformio/packages/tool-esptoolpy/esptool.py", line 5387, in <module>
    _main()
  File "/home/fimi/.platformio/packages/tool-esptoolpy/esptool.py", line 5380, in _main
    main()
  File "/home/fimi/.platformio/packages/tool-esptoolpy/esptool.py", line 4687, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/home/fimi/.platformio/packages/tool-esptoolpy/esptool.py", line 115, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/home/fimi/.platformio/packages/tool-esptoolpy/esptool.py", line 664, in connect
    last_error = self._connect_attempt(mode=mode, usb_jtag_serial=usb_jtag_serial, extra_delay=extra_delay)
  File "/home/fimi/.platformio/packages/tool-esptoolpy/esptool.py", line 629, in _connect_attempt
    self.sync()
  File "/home/fimi/.platformio/packages/tool-esptoolpy/esptool.py", line 519, in sync
    val, _ = self.command(self.ESP_SYNC, b'\x07\x07\x12\x20' + 32 * b'\x55',
  File "/home/fimi/.platformio/packages/tool-esptoolpy/esptool.py", line 484, in command
    self._port.timeout = saved_timeout
  File "/home/fimi/.platformio/penv/lib/python3.10/site-packages/serial/serialutil.py", line 372, in timeout
    self._reconfigure_port()
  File "/home/fimi/.platformio/penv/lib/python3.10/site-packages/serial/serialposix.py", line 401, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (5, 'Input/output error')
*** [upload] Error 1`

After that /dev/ttyUSB0 disappears and won't come back even if the USB cable is replugged - only a Reboot makes it appear again.

The dmesg shows:
`[  224.455788] usb 3-4: new full-speed USB device number 6 using xhci_hcd
[  224.705351] usb 3-4: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[  224.705362] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  224.705366] usb 3-4: Product: CP2102 USB to UART Bridge Controller
[  224.705369] usb 3-4: Manufacturer: Silicon Labs
[  224.705372] usb 3-4: SerialNumber: 0001
[  224.745409] usbcore: registered new interface driver usbserial_generic
[  224.745416] usbserial: USB Serial support registered for generic
[  224.747266] usbcore: registered new interface driver cp210x
[  224.747276] usbserial: USB Serial support registered for cp210x
[  224.747294] cp210x 3-4:1.0: cp210x converter detected
[  224.748155] usb 3-4: cp210x converter now attached to ttyUSB0
[  233.207288] usb usb2-port3: attempt power cycle
[  241.722782] usb usb2-port3: unable to enumerate USB device
[  249.963024] usb 3-4: USB disconnect, device number 6
[  249.963372] cp210x ttyUSB0: failed set request 0x7 status: -19
[  249.963383] cp210x ttyUSB0: failed set request 0x12 status: -19
[  249.963387] cp210x ttyUSB0: failed set request 0x0 status: -19
[  249.963572] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[  249.963594] cp210x 3-4:1.0: device disconnected
[  250.541987] usb 3-4: new full-speed USB device number 7 using xhci_hcd
[  250.777627] usb 3-4: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[  250.777637] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  250.779523] usb 3-4: can't set config #1, error -71
[  250.779669] usb 3-4: USB disconnect, device number 7`

System information:
Debian sid: `Linux framichi 5.18.0-1-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.2-1 (2022-06-06) x86_64 GNU/Linux
`Version: 1.68.0
Commit: 4af164ea3a06f701fe3e89a2bcbb421d2026b68f
Date: 2022-06-08T11:49:57.055Z
Electron: 17.4.7
Chromium: 98.0.4758.141
Node.js: 16.13.0
V8: 9.8.177.13-electron.0
OS: Linux x64 5.18.0-1-amd64`

PlatformIO: Core 6.0.2, Home 3.4.2

Jason2866, 2022-06-18:
Looks like you have not configured Platformio to have the right to access serial port.

michaelwebfischer, 2022-06-20:
> Looks like you have not configured Platformio to have the right to access serial port.

I don't think so, even after a "sudo chmod 777 /dev/ttyUSB0" the same error occurs...

Jason2866, 2022-06-20:
It looks like a problem with your Linux installation -> USB Serial driver. This part is not from Platformio.

valeros, 2022-06-21:
Does uploading work in the Arduino IDE?