platformio/platform-espressif32 #834: AddPreAction buildfs runs after the build

MartinVerges, 2022-06-17:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Linux (Ubuntu)

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.0.2

### Description of problem

When I add a `env.AddPreAction("buildfs", ...)` it runs after the mklittlefs instead of before.

#### Steps to Reproduce

Create a simple script (from the docs):
```
Import("env")

def before_buildfs(source, target, env):
    print("before_buildfs")

def after_buildfs(source, target, env):
    print("after_buildfs")

env.AddPreAction("buildfs", before_buildfs)
env.AddPostAction("buildfs", after_buildfs)
```

Add to the `platformio.ini` file env:target:
```
board_build.filesystem = littlefs
extra_scripts =  extrascript.py
```

Run `pio run --target buildfs -v` and see that it starts to build the FS and then outputs the pre and post script.

### Actual Results

```
Building in release mode
"mklittlefs_linux" -c ui/build -s 786432 -p 256 -b 4096 .pio/build/esp32dev/littlefs.bin
...
/index.html
before_buildfs(["buildfs"], [".pio/build/esp32dev/littlefs.bin"])
before_buildfs
after_buildfs(["buildfs"], [".pio/build/esp32dev/littlefs.bin"])
after_buildfs
```

### Expected Results

```
Building in release mode
before_buildfs(["buildfs"], [".pio/build/esp32dev/littlefs.bin"])
before_buildfs
"mklittlefs_linux" -c ui/build -s 786432 -p 256 -b 4096 .pio/build/esp32dev/littlefs.bin
...
/index.html
after_buildfs(["buildfs"], [".pio/build/esp32dev/littlefs.bin"])
after_buildfs
```

ivankravets, 2022-06-17:
This is a feature, not a bug. See code https://github.com/platformio/platform-espressif32/blob/develop/builder/main.py#L285

The `env.DataToBin`is called before `buildfs`. Could you try 

```python

env.AddPreAction(".pio/build/esp32dev/littlefs.bin", before_buildfs)
```

closed 2022-08-11 by stale[bot]

kel30a, 2023-02-12:
Could this be reopened @ivankravets? I just tested the following in espressif32@6.0.1

The following doesn't run at all:
`env.AddPreAction(".pio/build/esp32dev/littlefs.bin", before_buildfs)`
and this runs after the build:
`env.AddPreAction("buildfs", before_buildfs)`

kel30a, 2023-02-12:
> This is a feature, not a bug. See code https://github.com/platformio/platform-espressif32/blob/develop/builder/main.py#L285
> 
> The `env.DataToBin`is called before `buildfs`. Could you try
> 
> ```python
> env.AddPreAction(".pio/build/esp32dev/littlefs.bin", before_buildfs)
> ```

Could this be reopened @ivankravets? I just tested the following in espressif32@6.0.1 with PlatformIO Core v6.1.6.

The following doesn't run at all:
`env.AddPreAction(".pio/build/esp32dev/littlefs.bin", before_buildfs)`
and this runs after the build:
`env.AddPreAction("buildfs", before_buildfs)`

ivankravets, 2023-02-24:
@kel30a , do you have `esp32dev` env in your `platformio.ini`? To make the code more generic, I recommend this:

```python
env.AddPreAction("$BUILD_DIR/littlefs.bin", before_buildfs)
```

kel30a, 2023-04-07:
@ivankravets,  Sorry for the slow response, yes I have: `[env:esp32dev]` at the top of the platformio.ini file.

I agree with using the $BUILD_DIR variable but I tried both and neither run the function.

My function is just a  print statement at this point - I assume that it should show up? I can see the print statement when I add the following: 
`env.AddPreAction("buildfs", before_buildfs) `

bmedici, 2023-07-26:
Hi, same situation here, where the execution order is:
1) Building FS image from 'data' directory to .pio/build/ser/spiffs.bin
2) before_buildfs(["buildfs"], [".pio/build/ser/spiffs.bin"])
3) after_buildfs(["buildfs"], [".pio/build/ser/spiffs.bin"])

Which is not the expected order.

NCPlyn, 2024-07-14:
Still the same problem and it's quite irritating that the pre and post actions are basically the same....
How am I supposed to do something with the files before the filesystem gets build when the pre action runs after the build is done?!

olicooper, 2024-11-25:
Just experienced the same issue myself.. I need to minify and copy web files before building the filesystem image but none of the above actions are working correctly.

DeeEmm, 2024-12-13:
> Still the same problem and it's quite irritating that the pre and post actions are basically the same.... How am I supposed to do something with the files before the filesystem gets build when the pre action runs after the build is done?!

I agree. Just junk the prefixes as they do nothing and imply too much. The only purpose they appear to serve is to waste lots of time for multiple people. Documentation is completely lacking here too.

I'm experiencing a similar / related issue - https://community.platformio.org/t/opening-serial-monitor-runs-pio-extra-tasks/44635/5

IMHO relying on tying preActions to specific files is folly in this case, as it only works for single files / single threaded execution. Fine if you use an Arduino I guess, but not an ESP32

```
env.AddPreAction("$BUILD_DIR/${ESP32_FS_IMAGE_NAME}.bin", before_buildfs)
```

This is useless in my case

I'm intrigued as to why are there no system defined hooks for prebuild? i.e. BEFORE we call the compiler! And if there are, why are they not documented?

The documentation suggest this...

```
#
# Custom actions when building program/firmware
#

env.AddPreAction("buildprog", callback...)
```

But this hook is not called until AFTER the code is built

'Desired behaviour'? I think not. A poor choice of language perhaps?

I note that we print "Building in release mode" right after we scan the dependencies. Seems like an appropriate place to create a system defined pre build hook.

[image]

Or even better why not fire the pre: extra_script at this point. It's pretty much where everyone who uses this 'feature' expects the task to be fired.

A trivial solution really.

ivankravets, 2024-12-13:
If you want to perform any actions before the compiler starts, you can leverage PRE-type scripts https://docs.platformio.org/en/latest/scripting/launch_types.html

DeeEmm, 2024-12-13:
I have already tried this. Does not work for my case.

The issue for me is that not only is the PRE: script run prior to the build, but opening the serial monitor also runs the PRE: script, and I have read other users who note that it also runs when they modify platformio.ini or other non system tasks that call platformio (such as the esp32_exception_decoder)

The nomenclature **'pre'** and the documentation the specifically states **"PRE - executes before the main script of Development Platforms"** Clearly implies that this is the **only** time that PRE scripts should run. So the behaviour is completely unexpected.

I have seen you refer to this as 'desired behaviour' elsewhere, but the sheer number of related posts seem to suggest otherwise.

The problem remains that to prevent this behaviour it is necessary to include some form of procedural code within the PRE:extra_script to determine what called the script.

This is the prescribed solution, which does not work...

```
#
# Custom actions when building program/firmware
#

env.AddPreAction("buildprog", callback...)
```

There is no way that this will execute before the main script of the Development Platform

It would work if I could use a hook that specifically links its operation to before the build process takes place, but none exist. Or at least none that I could find.

I note that there are many (many) posts where users all face a similar challenge. Specifically that the assumed functionality of PRE: scripts and preActions is very different from its actual functionality. And that the examples given are ambiguous.

A pre-build hook would do the trick

```
env.AddPreAction("pre-build", callback...)
```

It should fire after loading the dependencies but before starting the build process. 

i.e. 'before the build' (pre-build). Before the main script of the Development Platform

I suspect then that you would no longer have support requests such as this one, or mine, or the many (many) others who cannot get this to work using the information provided.

Unless I'm missing something of course. But having spent days reading everything that the internet has to offer on the subject, I'm convinced it is as described.

Please feel free to dig a bit further into my post linked to above. Would love to get this resolved 

/DM

ivankravets, 2024-12-13:
Could you try the hook mentioned here https://docs.platformio.org/en/latest/scripting/examples/asking_for_input.html?
```python
if env.IsIntegrationDump():
    # stop the current script execution
    Return()
```

DeeEmm, 2024-12-13:
Will give it a go. This was already part of a solution that I previously tried that did not work, although it was in a sightly different format...

https://community.platformio.org/t/env-addpreaction-help/34715
```
# Make sure 'vscode init' is not the current command
def is_pio_build():
    from SCons.Script import DefaultEnvironment
    env = DefaultEnvironment()
    if "IsCleanTarget" in dir(env) and env.IsCleanTarget(): return False
    return not env.IsIntegrationDump()
```

DeeEmm, 2024-12-13:
And there it is. Now it works

Thank you. 

Please consider updating the documentation for the PreAction with a better example. 👍 

(Still think a pre-build hook is a good idea tho 😉 )

ivankravets, 2024-12-13:
What is the best place where we can put this information? Do you mean https://docs.platformio.org/en/latest/scripting/actions.html ?

DeeEmm, 2024-12-13:
Yes.

The example that I came across most when searching was that people were trying to use the following example verbatim

```
#
# Custom actions when building program/firmware
#

env.AddPreAction("buildprog", callback...)
```

Expecting that this was all that was needed to ensure that the callback executed before the main script of the Development Platform 

However, I believe that the "buildprog" hook is either a nonsense example (placeholder) or does not fire until after the build process.

It would be ideal if the example could be changed to one that COULD be copied verbatim, hence my suggestion of creating a pre-build hook if one does not exist.

Otherwise the best solution is not to use the env.AddPreAction callback, but to do something similar to your given solution, where the code is placed in the file without the callback function, but not that code operation may need to be controlled according to other factors such as testing for IsIntegrationDump()

For example...

**platformio.ini**
```
extra_scripts = 
	pre:user_actions_pre.py
```

**user_actions_pre.py**
```
# stop the current script execution if called from external script
if env.IsIntegrationDump():
    Return()

print("Current CLI targets", COMMAND_LINE_TARGETS)
print("Pre-Build tasks")

# do Pre-Build stuff here
# ...
# ...

```

I know that this might be a very specific use case, but the most confusing part is that the example does not work 'out of the box' and unless you can use one of your build files as the hook, there is no generic hook to use. So the solution is to fall back to just allowing the code in user_actions_pre.py to run uninhibited, but add code to mask out unwanted calls such as 'esp32_exception_decoder' or other similar tasks.

/DM

DeeEmm, 2024-12-13:
Also, the issue is further compounded by the suggestion to use _pio run --list-targets_ to find a target that can be used as the hook.

I would have expected to see a 'build' target, but there is none.

```
esp32dev        Platform  buildfs      Build Filesystem Image
esp32dev        Platform  erase        Erase Flash
esp32dev        Platform  size         Program Size                 Calculate program size
esp32dev        Platform  upload       Upload
esp32dev        Platform  uploadfs     Upload Filesystem Image
esp32dev        Platform  uploadfsota  Upload Filesystem Image OTA
```

DeeEmm, 2024-12-13:
Similarly creating a custom target also fails for the same reason

```
targets = clean, upload
```

So it is difficult to stumble across the solution. I spent some time reading the documentation but never made the association with the _if env.IsIntegrationDump():_ test as it is listed on the page for user input. Python is not my native tongue so I struggle a little with it

Hopefully this gives some insight.

Thanks for your help.

/DM

DeeEmm, 2024-12-13:
Thanks Ivan, you are a star ⭐