platformio/platform-espressif32 #835: pins_arduino.h for LillyGo TTGO t7 1.5 (16MB variant)

AlexFundorin, 2022-06-17:
This board variant is missing from platformio list.
I tried copying and editing 1.4 json configuration but still need to alter pins_arduino.h properly, since pins 16 and 17 access was removed in 1.5 variant and pins 25, 27 were duplicated, instead.
So, I need to alter .h file, but can't figure out how.
Here's the 1.4 pinout:

```
#ifndef Pins_Arduino_h
#define Pins_Arduino_h

#include <stdint.h>

#define EXTERNAL_NUM_INTERRUPTS 16
#define NUM_DIGITAL_PINS        40
#define NUM_ANALOG_INPUTS       16

#define analogInputToDigitalPin(p)  (((p)<20)?(analogChannelToDigitalPin(p)):-1)
#define digitalPinToInterrupt(p)    (((p)<40)?(p):-1)
#define digitalPinHasPWM(p)         (p < 34)

static const uint8_t TX = 1;
static const uint8_t RX = 3;

static const uint8_t LED_BUILTIN = 19;
#define BUILTIN_LED  LED_BUILTIN // backward compatibility

static const uint8_t SDA = 21;
static const uint8_t SCL = 22;

static const uint8_t SS    = 5;
static const uint8_t MOSI  = 23;
static const uint8_t MISO  = 19;
static const uint8_t SCK   = 18;

static const uint8_t A0 = 36;
static const uint8_t A3 = 39;
static const uint8_t A4 = 32;
static const uint8_t A5 = 33;
static const uint8_t A6 = 34;
static const uint8_t A7 = 35;
static const uint8_t A10 = 4;
static const uint8_t A11 = 0;
static const uint8_t A12 = 2;
static const uint8_t A13 = 15;
static const uint8_t A14 = 13;
static const uint8_t A15 = 12;
static const uint8_t A16 = 14;
static const uint8_t A17 = 27;
static const uint8_t A18 = 25;
static const uint8_t A19 = 26;

static const uint8_t T0 = 4;
static const uint8_t T1 = 0;
static const uint8_t T2 = 2;
static const uint8_t T3 = 15;
static const uint8_t T4 = 13;
static const uint8_t T5 = 12;
static const uint8_t T6 = 14;
static const uint8_t T7 = 27;
static const uint8_t T8 = 33;
static const uint8_t T9 = 32;

static const uint8_t DAC1 = 25;
static const uint8_t DAC2 = 26;

#endif /* Pins_Arduino_h */

```

Jason2866, 2022-06-18:
This file is from Arduino Espressif framework not from Platformio.

valeros, 2022-06-21:
Please add your variant to the official repo with the Arduino core for ESP32 [here](https://github.com/espressif/arduino-esp32) and once a new stable release is published, your variant will be available in PlatformIO.

closed 2022-06-21 by valeros