platformio/platform-espressif32 #837: esp32-s3-devkitc-1 issue

fabiencomte, 2022-06-18:
Hello,
Basic hello world doesn't work.

#include "Arduino.h"

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.print("Hello world\n");
}

void loop() {
    delay(1000);
    Serial.print("Hello world\n");
}

This is the monitor output.

Rebooting...
␀�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376c18
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x39c
load:0x403b6000,len:0x9a4
load:0x403ba000,len:0x2868
entry 0x403b61c0

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)

Backtrace:0x403769f6:0x3fceb1800x40379b35:0x3fceb1a0 0x4037f095:0x3fceb1c0 0x42007fc2:0x3fceb2f0 0x40376511:0x3fceb320 0x403bae67:0x3fceb350 0x403bb0ee:0x3fceb380 0x403b6215:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
␀�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376c18
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x39c
load:0x403b6000,len:0x9a4
load:0x403ba000,len:0x2868
entry 0x403b61c0

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)

Backtrace:0x403769f6:0x3fceb1800x40379b35:0x3fceb1a0 0x4037f095:0x3fceb1c0 0x42007fc2:0x3fceb2f0 0x40376511:0x3fceb320 0x403bae67:0x3fceb350 0x403bb0ee:0x3fceb380 0x403b6215:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
␀�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376c18
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x39c
load:0x403b6000,len:0x9a4
load:0x403ba000,len:0x2868
entry 0x403b61c0

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)

Backtrace:0x403769f6:0x3fceb1800x40379b35:0x3fceb1a0 0x4037f095:0x3fceb1c0 0x42007fc2:0x3fceb2f0 0x40376511:0x3fceb320 0x403bae67:0x3fceb350 0x403bb0ee:0x3fceb380 0x403b6215:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
␀�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376c18
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x39c
load:0x403b6000,len:0x9a4
load:0x403ba000,len:0x2868
entry 0x403b61c0
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x10 (DOWNLOAD(USB/UART0))
waiting for download

Jason2866, 2022-06-18:
Looks like used Flash mode does not match with your boards flash type.

fabiencomte, 2022-06-18:
Hello,
How can I change it please ?
If I use platform https://github.com/Jason2866/platform-espressif32 it works
If I just switch to https://github.com/platformio/platform-espressif32 it fails

Jason2866, 2022-06-18:
Since the boards.json for esp32-s3-devkitc-1 are different. 4MB for flash is always wrong (platformio setting) regarding https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html

Jason2866, 2022-06-18:
Easy to solve. Make a folder `boards` copy the orig boards.json and rename it like you want. Adjust the settings to match exactly with your board. Use the board in platformio with the name you gave.

fabiencomte, 2022-06-18:
Thank you for your help.
My model is ESP32-S3-DevKitC-1-N32R8V
I tried with 32 MB but it not works. Do you have another idea please ?

fabiencomte, 2022-06-18:
With the same board json it works on one but not the other. Maybe thez call different versions of somethings.

fabiencomte, 2022-06-18:
When it works, outputs on boot are stranges (see boot anyway):

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x118
load:0x403b6000,len:0xba4
load:0x403ba000,len:0x2878
SHA-256 comparison failed:
Calculated: d16c7a378feb70261621af15434ea6fa968e131c146a139ce3ce721b734f10bd
Expected: b66460354a0aa15d21a5d6c01c56f6d004d78c763067488c6ed77ed56e7c9a87
Attempting to boot anyway...
entry 0x403b61c0

Jason2866, 2022-06-19:
Just saw the official one is missing a mandantory entry. This line https://github.com/Jason2866/platform-espressif32/blob/Tasmota/dev_apple/boards/esp32-s3-devkitc-1.json#L5
This line defines the Flash/PSRAM which the board has. When wrong the board bootloops. For your board it has to be `"memory_type": "opi_opi"`

fabiencomte, 2022-06-19:
Hello Jason2866,
It works better (no more boot loop) but I have a message "[   667][E][esp32-hal-misc.c:252] initArduino(): Failed to initialize NVS! Error: 263" that is strange.
Another point is that if I try to replace 8MB with 32MB and 8388608 with 33554432 => it fails

With this board file:
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
	  "memory_type": "opi_opi"
    },
    "core": "esp32",
    "extra_flags": [
    	"-DBOARD_HAS_PSRAM",
      "-DARDUINO_ESP32S3_DEV",
      "-DARDUINO_USB_MODE=1",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1"
    ],
    "f_cpu": "160000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": [
      "esp-builtin"
    ],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif ESP32-S3-DevKitC-1",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 524288,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html",
  "vendor": "Espressif"
}

I have this output:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
SHA-256 comparison failed:
Calculated: 0bb45b01207a4a9e9fca8880ddfc0f79112bfa4d004bd1e6b2eaa75c5323b19d
Expected: 6eb14ec480aff4f05e39378c41406102ac163573741e63b6cfd73509ef44e3bd
Attempting to boot anyway...
entry 0x403b61d8
[   667][E][esp32-hal-misc.c:252] initArduino(): Failed to initialize NVS! Error: 263

Hello world

Jason2866, 2022-06-19:
This error (NVS) is not a platformio setup problem. I would open a issue in Espressif Arduino github. You have used a 8MB partition scheme?

fabiencomte, 2022-06-19:
Where can I check the partition scheme that I use please ?
Thank you so much for your help.

Jason2866, 2022-06-19:
Platformio has a good docu. See https://docs.platformio.org/en/latest//platforms/espressif32.html?highlight=partitions+csv#partition-tables

closed 2022-08-11 by stale[bot]

chegewara, 2022-09-04:
The problem is with flash type. Some devkits are coming from factory with Octal mode efuse burned and platformio is for some reason not ready to flash those.
I have 2 models of devkit:
- old one, with DIO/QIO flash mode, which works fine with `esp32-s3-devkitc-1`
- new one with OPI efuse burned (ESP32-S3-F32R8), this one works fine with arduino, but fails to boot with pio 
In both cases i am using exactly the same code, the same board model etc, just devkit has different efuse

Im not sure, but i am suspecting there is one bootloader missing or wrong in pio or maybe i am missing one flag to use flash_mode=opi.

Jason2866, 2022-09-04:
@chegewara  There is no efuse set for octal flash. There is a feature [request](https://github.com/espressif/esp-idf/issues/9101) for.
And yes you have to set the correct memory_type. A correct example for OPI / OPI is this [boards.json](https://github.com/platformio/platform-espressif32/blob/develop/boards/esp32s3camlcd.json#L5)
If i mentioned earlier it is a very bad idea from manufacturers to sell under the same name different board memory variants. That ends for the S3 in bootloops.

Jason2866, 2022-09-04:
@valeros please reopen it is still a issue.

chegewara, 2022-09-04:
Thanks @Jason2866 but you are mixing 2 things. This request is about something else (https://github.com/espressif/esp-idf/issues/9101)
I am saying about this efuse for OPI flash:
```
python3 ./espefuse.py -p /dev/<serial_device> --do-not-confirm burn_efuse FLASH_TYPE 1
```
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html#error-handling
and my devkit is with burned this efuse at factory.

I tried to change files, but it did not help:
https://github.com/Jason2866/platform-espressif32/blob/Tasmota/205/builder/main.py#L51-L58
https://github.com/platformio/platform-espressif32/blob/develop/boards/esp32s3camlcd.json#L5

Thanks

Jason2866, 2022-09-04:
ArduinoIDE does not read this fuse. If you look into the flash sequence. It is the same as in Platformio (with this change). Your boards.json is not correct in this case.
My S3 with opi opi does work when flashed with Platformio

chegewara, 2022-09-04:
Im not saying pio or arduino does read it. The bootloader and firmware does.
If your S3 works with PIO then i am doing something wrong, because it still fail to boot.

Thanks anyway, i have few devkits with normal flash mode, so i can wait

Jason2866, 2022-09-04:
Which memory variant do you have opi/opi or opi/qio ? And which flash psram size?
Could you post your boards.json manifest ?

chegewara, 2022-09-13:
Hi @Jason2866, thanks for support.
I am able to flash S3 in OPI mode with your help, eventually. But there is one more problem.

After more digging i even found this change is not required:
https://github.com/Jason2866/platform-espressif32/blob/Tasmota/205/builder/main.py#L51-L58
Here is my `platformio.ini`:
```
[env:esp32-s3-devkitc-2]
board = esp32-s3-devkitc-2
framework = arduino
monitor_speed = 115200
upload_speed = 1000000
upload_port = /dev/ttyUSB0
board_build.flash_mode = dout
```
and `esp32-s3-devkitc-2.json`
```
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_8MB.csv",
      "memory_type": "opi_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_ESP32S3_DEV",
      "-DARDUINO_USB_MODE=0",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dio",
    "hwids": [
      [
        "0X303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": [
      "esp-builtin"
    ],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif ESP32-S3-DevKitC-2-opi",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html",
  "vendor": "Espressif"
}
```
The only additional step i had to do is to rename bootloader:
```
~/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/bin$ mv bootloader_opi_80m.bin bootloader_dout_80m.bin
```
Its because there is no `_dout_` bootloader, and `_opi_` is never used (as far as i know).

Jason2866, 2022-09-14:
@chegewara This part is needed since opi flash needs to be flashed in mode `dout` That is what this line does. It has nothing to do with which bootloader file is flashed. In your example bootloader `dio` ` "flash_mode": "dio",` is used.
See  [Here](https://github.com/espressif/esp-idf/blob/master/components/esptool_py/Kconfig.projbuild) are the flash modes set [Same for IDF 44.](https://github.com/espressif/esp-idf/blob/release/v4.4/components/esptool_py/Kconfig.projbuild)

```
    # Note: we use esptool.py to flash bootloader in
    # dio mode for QIO/QOUT, bootloader then upgrades
    # itself to quad mode during initialisation
    config ESPTOOLPY_FLASHMODE
        string
        default "dio" if ESPTOOLPY_FLASHMODE_QIO
        default "dio" if ESPTOOLPY_FLASHMODE_QOUT
        default "dio" if ESPTOOLPY_FLASHMODE_DIO
        default "dout" if ESPTOOLPY_FLASHMODE_DOUT
        # The 1st and 2nd bootloader doesn't support opi mode,
        # using fastrd instead. For now the ESPTOOL doesn't support
        # fasted (see ESPTOOL-274), using dout instead. In ROM the flash mode
        # information get from efuse, so don't care this dout choice.
        default "dout" if ESPTOOLPY_FLASHMODE_OPI
```

chegewara, 2022-09-14:
Ok @Jason2866 its your opinion.
I tested with few different settings, including yours suggested solution, and i know it is the best working for me.
If `bootloader_opi_80m.bin` is not flashed, then BLE wont work, because for some reason `nvs_flash_init` returns error 0x107, which is strange, because documentation for this function is not saying a word about error 0x107.
Sure, you are platformio expert, and i am not, but my solution works with platformio@5.1.0 and ill stick to it.

> The `opi` bootloader is [there](https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/esp32s3/bin)

i know where is required bootloader, because platformio complained about missing bootloader in path when i set this line
`board_build.flash_mode = dout`
and from there i figured out which bootloader i should rename. If you think i am saying stupid things then just ignore me, i just wanted to thank you, because your suggestions let me to fix the problem.

Jason2866, 2022-09-14:
@chegewara  So there is still a bug for `opi` mode. The change for mode `dout` is correct. BUT as you describe the wrong bootloader seems to be choosen. This needs to be fixed. Your way is not the correct fix, since the bootloader is flashed in a not supported mode (see how IDF does, mentioned above). This may work for you but it is not align with the official way of espressif.

chegewara, 2022-09-14:
> This may work for you but it is not align with the official way of espressif

Again you are incorrect. Platformio and arduino ide are using pre-build bootloaders, and esp-idf is building one, unless you mean "espressif way" in meaning of arduino ide. In that case platformio is not following its way, because in arduino IDE it is working and here its not. Bug is bug, and topic is closed.
Anyway, this start to becoming offtopic. Like i said, i dont care who and what will do with info i provided, i just hope that someone finds it useful not being forced to use tasmota or your fork.

Jason2866, 2022-09-14:
@chegewara  Try this. The entry `"boot": "opi",` was missing.

```
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_8MB.csv",
      "memory_type": "opi_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_ESP32S3_DEV",
      "-DARDUINO_USB_MODE=0",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dio",
    "boot": "opi",
    "hwids": [
      [
        "0X303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": [
      "esp-builtin"
    ],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif ESP32-S3-DevKitC-2-opi",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html",
  "vendor": "Espressif"
}
```

chegewara, 2022-09-14:
My conclusion was correct:
- the wrong bootloader has been flashed i OPI mode

I just found workaround, which maybe is not the best one but its working and you found the correct fix.

Fllamber, 2023-02-01:
This thread helped me, I was having problems with the ' AI Thinker ESP32-CAM ' and just went in `Tools` and changed `Flash Mode` to **DIO** and was able to boot properly. (it was selected QIO) I did not test other Flash Modes.
Navigating some menu options was much easier than changing configurations files. Thanks.

egnispinto, 2023-04-27:
El problema que estas teniendo es que estas mal declarando el puerto Serial para la placa S33. Tu ejemplo funcionaria perfectamente en un ESP normal, pero en un S3 debes hacerlo con "HWCDC PuertoSerial ;" te paso un ejemplo que si corre perfectamente, espero te sirva:
#include "Arduino.h"
HWCDC PuertoSerial ;
void setup(){
    PuertoSerial.begin(9600);
    return;
}

void loop(){
    PuertoSerial.println("Hola mundo");

    delay(1000);
}

commit 4f511507 (2022-09-26): Take into account "memory_type" option to properly select boot mode (#913)

Use correct bootloader `opi` when `"boot": "opi"` is not set for Arduino framework

Resolves #837, enhances #904