platformio/platform-espressif32 #846: CONFIG_FREERTOS_HZ=1000 should be added to sdkconfig.defaults for examples

egnor, 2022-07-05:
If the scheduler HZ is <1000, then small calls to the Arduino delay() function (such as delay(5)) will return immediately. The default HZ is 100. I recommend adding CONFIG_FREERTOS_HZ=1000 to the sdkconfig.defaults for espidf-arduino-blink and espidf-arduino-wifiscan, since people often clone these projects. I'm happy to create a PR for this if you think it makes sense.

(Also, the Arduino core should be better implemented so it works even with chunkier scheduling HZ. I'll file a separate bug about that.)

egnor, 2022-07-07:
Note, https://github.com/espressif/arduino-esp32/pull/6955 (if accepted) will force this issue.

wawan-ikhwan, 2022-07-28:
Yeah I thought so, I expect I get current millisecond from this function`xTaskGetTickCount` , But what I get is centisecond lol. Turns I need set to `CONFIG_FREERTOS_HZ=1000` in sdkconfig,

Jason2866, 2022-08-26:
Please close. Feature is added in espressif32 Arduino github.

closed 2022-08-26 by egnor