platformio/platform-espressif32 #861: -t uploadfsota does not work in with -t nobuild

sebashb, 2022-07-22:
I'm trying to upload the spiffs.bin file that has already been compiled with `pio run -t nobuild -t uploadfsota -e esp32_ota --upload-port 10.20.34.120` but I get the error: **Could not find the file  with partitions table.**.  Removing the flag -t nobuild builds and uploads the FS as expected.

sebashb, 2022-09-09:
This is still an issue.

valeros, 2022-09-09:
Hi @sebashb ! Indeed, the partition table is set dynamically, so if you specify the `nobuild` target then the framework build scripts are not executed and hence the partition file is missing. As a workaround you can set a partition table by yourself in an extra script, for example:
```ini
[env:lolin32]
platform = espressif32
framework = arduino
board = lolin32
extra_scripts = 
    pre:set_partition_table.py
```
where `set_partition_table.py` is the following:
```python
import os

from SCons.Script import COMMAND_LINE_TARGETS

Import("env")

board_config = env.BoardConfig()
framework_dir = env.PioPlatform().get_package_dir("framework-arduinoespressif32")
assert os.path.isdir(framework_dir)

if "nobuild" in COMMAND_LINE_TARGETS:
    env.Replace(
        PARTITIONS_TABLE_CSV=os.path.join(
            framework_dir,
            "tools",
            "partitions",
            board_config.get("build.partitions", "default.csv"),
        )
    )

```

closed 2022-10-19 by stale[bot]