platformio/platform-espressif32 #862: ESP-C3-32S-KIT-2M。After burning the program, it cannot run normally, but uploading the program in arduinoide is available.

xiezhoubin, 2022-07-25:


ivankravets, 2022-07-25:
Do you use the same version of Arduino Core for ESP32? See related issue https://community.platformio.org/t/why-the-ble-client-example-works-on-arduino-ide-but-dont-work-on-the-vscode-platformio/28857

xiezhoubin, 2022-07-26:
Still not.
[image]
[image]
[image]

xiezhoubin, 2022-07-26:
Arduino for esp32 version is 2.0.4.
[image]

maxgerhardt, 2022-07-26:
Seems to work when creating a board JSON file that has 2MByte set as `flash_size`. Setting it via the platformio.ini still somehow results in PIO using the original board JSON file's 4MByte value, and this crashes the boot process. See [here](https://community.platformio.org/t/esp-c3-32s-kit-2m-after-burning-the-program-it-cannot-run-normally-but-uploading-the-program-in-arduinoide-is-available/28910/8?u=maxgerhardt).

valeros, 2022-07-26:
Shouldn't it be `board_upload.flash_size = 2MB` instead of `board_build.flash_size = 2MB`?

xiezhoubin, 2022-07-26:
> board_upload.flash_size = 2MB

you are right.
[image]

xiezhoubin, 2022-07-26:
> Shouldn't it be `board_upload.flash_size = 2MB` instead of `board_build.flash_size = 2MB`?
So what is the function of "board_build.flash_size"?

valeros, 2022-07-26:
> So what is the function of "board_build.flash_size"?

It's a typo, there is no `flash_size` option in the `build` section of the board manifest.

xiezhoubin, 2022-07-26:
> Ohh indeed. Technically that should also be `board_upload.maximum_size`, but this is dynamically recalculated by PIO anyways.

Do you need to set "board_build.maximum_size"? Or will it be generated automatically?

xiezhoubin, 2022-07-26:
In addition, where can I find relevant instructions about configuration parameters such as "board_upload" or "board_build"?

valeros, 2022-07-26:
> Do you need to set "board_build.maximum_size"? Or will it be generated automatically?

It might be a good idea to be consistent and update `maximum_size` as well. 

> In addition, where can I find relevant instructions about configuration parameters such as "board_upload" or "board_build"?

More info available here https://docs.platformio.org/en/latest/platforms/espressif32.html#configuration

closed 2022-07-26 by valeros

xiezhoubin, 2022-07-26:
ok,thanks very much.