platformio/platform-espressif32 #863: cmake fails with a KeyError: 'nameOnDisk' in espidf.py

Captainohair, 2022-07-25:
I’m attempting to include an external cmake project into my platformio wesp32 project being built in VS Code’s platformio system. There are currently no errors without attempting to include the external cmake project. However, I get this error when I build with the following configuration. I have already deleted my entire .platformio directory under my user directory and let platformio rebuild everything there and done multiple cleans and rebuilds to no effect. I am struggling to find any help online about this particular issue already hence here I am. The log output isn’t terribly helpful to me though I am not familiar with the files in the call stack.

platformio.ini:
```
[platformio]
default_envs = esp-wesp32

[env:esp-wesp32]
platform = espressif32
board = wesp32
framework = espidf
build_flags = -DBOARD_WESP32
;upload_port = /dev/cu.usbserial-20
;monitor_port = /dev/cu.usbserial-20
monitor_speed = 115200

[env:esp-espressif-dev]
platform = espressif32
board = esp-wrover-kit
framework = espidf
build_flags = -DBOARD_ESPRESSIF_DEV
;upload_port = /dev/cu.usbserial-201
;monitor_port = /dev/cu.usbserial-201
monitor_speed = 115200
```
CMakeLists.txt at root of project dir:
```
cmake_minimum_required(VERSION 3.16.0)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
include(FetchContent)
project(leigh-esp32-cube)

set(ETCPAL_OS_TARGET freertos CACHE STRING "The OS target for EtcPal")
set(ETCPAL_NET_TARGET lwip CACHE STRING "The network target for EtcPal")

add_library(FreeRTOS INTERFACE)
target_include_directories(FreeRTOS INTERFACE [freertos])
add_library(lwIP INTERFACE)
target_include_directories(lwIP INTERFACE [lwip])

FetchContent_Declare(
        sACN
        GIT_REPOSITORY "https://github.com/ETCLabs/sACN"
        GIT_TAG "v2.0.1"
)
FetchContent_MakeAvailable(sACN)
```
Build log output:
```
Processing esp-wesp32 (platform: espressif32; board: wesp32; framework: espidf)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wesp32.html
PLATFORM: Espressif 32 (5.0.0) > Silicognition wESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.40401.0 (4.4.1) 
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
KeyError: 'nameOnDisk':
  File "C:\Users\CaptainOHair\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 188:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\CaptainOHair\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\CaptainOHair\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\CaptainOHair\.platformio\platforms\espressif32\builder\main.py", line 283:
    target_elf = env.BuildProgram()
  File "C:\Users\CaptainOHair\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\CaptainOHair\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProgramDeps()
  File "C:\Users\CaptainOHair\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\CaptainOHair\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 122:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\CaptainOHair\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\CaptainOHair\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 328:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\CaptainOHair\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 660:
    return method(*args, **kw)
  File "C:\Users\CaptainOHair\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\CaptainOHair\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\CaptainOHair\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 1234:
    build_components(env, framework_components_map, PROJECT_DIR)
  File "C:\Users\CaptainOHair\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 827:
    components_map[k]["lib"] = build_library(
  File "C:\Users\CaptainOHair\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 269:
    lib_name = lib_config["nameOnDisk"]
```

valeros, 2022-07-25:
Hi @Captainohair ! Thanks for reporting. To put it simply, the current integration of the IDF framework is not intended to work with arbitrary external CMake projects (not related to the IDF software ecosystem). Anyway, I've pushed a small fix to handle the issue, so it should help you move forward. Finally, I strongly recommend you to make sure your project is compilable within the native IDF build system before porting it to PlatformIO as it seems the `sACN` project will require some additional CMake tweaks to compile properly.

closed 2022-07-25 by valeros

Captainohair, 2022-07-25:
Grand, I appreciate that.

And noted. Platformio docs seemed to indicate I could use external cmake projects so I moved forward under that assumption. I'll give this another go with your fix and see where I get to now.

I'm not familiar with the release cycle here. Is your pushed fix an immediate release or will that take a bit to show up in whatever next version makes its way down the pipe? I'm fairly new to IDF and Platformio.

valeros, 2022-07-26:
> I'm not familiar with the release cycle here. Is your pushed fix an immediate release or will that take a bit to show up in whatever next version makes its way down the pipe? I'm fairly new to IDF and Platformio.

Usually, we publish a new stable release once we have a considerable amount of improvements or there are critical bugfixes. But keep in mind that you don't need to wait for a new release, you can always switch to the upstream version of the platform from the development branch:
```ini
[env:esp-wesp32]
platform = https://github.com/platformio/platform-espressif32.git
board = wesp32
framework = espidf
...
```

Captainohair, 2022-07-26:
Grand, thank you for explaining.