platformio/platform-espressif32 #866: Arduino Serial Monitor Empty

sharonf88, 2022-07-31:
Hi guys!

I'm trying to operate a simple example using arduino to display some value on the esp32c3.
I manage to connect but i'm left with :
Writing at 0x000b3a12... (100 %)
Wrote 685920 bytes (419292 compressed) at 0x00010000 in 9.5 seconds (effective 577.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

And an empty serial monitor screen.
Help?

valeros, 2022-08-01:
Hi @sharonf88 ! What happens when you press the reset button on your board?

sharonf88, 2022-08-02:
> 

Hi! I hear the USB connection windows sound but still nothing shows up.
I tried the discord for help with no results.

valeros, 2022-08-02:
Could you please share your `platformio.ini` here?

Jason2866, 2022-08-03:
Which board do you use? If the inbuilt USB modem (=CDC) is used you have to use a boards.json which have the needed settings for. If you use a standard c3 boards.json you will see nothing on the serial console!

closed 2022-09-20 by stale[bot]