platformio/platform-espressif32 #868: TinyUSB CDC example under PlatformIO

LarsMygh, 2022-08-02:
Hello,

I have been trying to get the USB facility on the ESP32-S2 for some time. None of the examples will compile out of the box under PlatformIO.

Finally, I learned that whereas there seems to be many ‘versions’ of the TinyUSB, but I found a supposedly easy to adapt example:

I followed the instructions on:
https://registry.platformio.org/libraries/adafruit/Adafruit%20TinyUSB%20Library/installation

Adding the library went fine.

However when I followed the Examples link on the page above, I went and copied/adapted the cdc_multi.ino example:
https://registry.platformio.org/libraries/adafruit/Adafruit%20TinyUSB%20Library/examples/CDC/cdc_multi/cdc_multi.ino

Among many compile errors is the representative first one:
'Adafruit_USBD_CDC' does not name a type

Apparently the TUSB_OPT_DEVICE_ENABLED in addition with other symbols, used in Adafruit_TinyUSB.h need to be defined?

Googl'ing the issue, I came across many explainations, like the order in which libraries are compiled, supposedly solved by renaming library files using python scripts.

Long story short:
Is it missing a description in your page or am I missing something? Is there a stocked away example somewhere which will compile a TinyUSB CDC example under PlatformIO?

valeros, 2022-08-02:
Hi @LarsMygh , thanks for reaching out. Is there any chance that the `cdc_multi` example is not meant to be used with the ESP32 platform? I'm asking because the `cdc_multi` example in the  [TinyUSB repository](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/CDC/cdc_multi) contains special files (e.g. `.metroesp32s2.test.skip`) that exclude ESP32 targets from running in CI for some reason. Also, could you please try to compile any example in the Arduino IDE? Does it work there?

LarsMygh, 2022-08-02:
Having installed the AdaFruit TinyUSB library. I've a tried to compile the cdc_multi.ino under the Arduino IDE. Same error.
Inspired by your comment, I also noticed the .metroesp32s2.test.skip file,
When looking at the one of the other examples, HID, I can it git see a 10 month ago update comment:
[update hid examples to work esp32s2]

So I guess, that they've never gone around to updating the CDC examples as well. I'll guess the ball is in AdaFruits court yard. I'll contact them.

Should I subsequently update with a comment here, if I get a positive reponse? (Sorry, thought I better ask: I am not an expert in how these Fora normlly works)

valeros, 2022-08-02:
At this stage, I'd prefer to close the issue as it doesn't seem that it has something to do with PlatformIO. You can reopen the issue if you bump into a TinyUSB example that compiles just fine in the Arduino IDE but doesn't work under PlatformIO.

closed 2022-08-16 by valeros