platformio/platform-espressif32 #872: Error programming with ESP-PROG on 5.1.0

FireDeveloper, 2022-08-08:
Hello,

Uploading with the same settings was working fine with previous version.
(10 mins before it was working, no cable changed/touched, nor settings)
After the upgrade I get the following error:
(Same with 5000khz speed)

```
Building in release mode
Retrieving maximum program size .pio\build\env1\firmware.elf
Checking size .pio\build\env1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.4% (used 50364 bytes from 327680 bytes)
Flash: [========  ]  79.4% (used 1040622 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esp-prog
Uploading .pio\build\env1\firmware_merged.bin
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 20000 kHz

DEPRECATED! use 'adapter speed' not 'adapter_khz'
adapter speed: 20000 kHz

** Programming Started **
Error: Invalid offset!
Error: failed erasing sectors 0 to 270
embedded:startup.tcl:1184: Error: ** Programming Failed **
in procedure 'program_esp' 
in procedure 'program_error' called at file "C:/Users/Admin/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 171
at file "embedded:startup.tcl", line 1184
*** [upload] Error 1
```

Platform IO settings:
```
[platformio]
default_envs = env1

[env:env1]
platform = espressif32@5.1.0
framework = arduino
board = esp32-s2-saola-1
debug_tool = esp-prog
debug_speed = 20000
upload_protocol = esp-prog
; upload_port = COM6
; debug_tool = jlink
build_flags = 
	-I include
lib_deps = 
	rlogiacco/CircularBuffer@1.3.3
	milesburton/DallasTemperature@^3.11.0
	paulstoffregen/OneWire@^2.3.7
	bblanchon/ArduinoJson@^6.19.4
	ottowinter/ESPAsyncWebServer-esphome@^2.1.0
	ayushsharma82/AsyncElegantOTA@^2.2.7
	https://github.com/tzapu/WiFiManager/archive/refs/tags/v2.0.11-beta.zip
	arduino-libraries/NTPClient@^3.2.1
```

valeros, 2022-08-09:
Hi @FireDeveloper ! Thanks for reporting. Could you please upload your firmware in verbose mode? Just run `pio run -e env1 -t upload -v` in the IDE terminal.

FireDeveloper, 2022-08-09:
```
.xt.lit._ZN11WiFiManager8DEBUG_WMIPKc6StringEEvNS_15wm_debuglevel_tET_T0_                                                                                                                                               0        
    0

.xt.lit._ZN11WiFiManager8DEBUG_WMIPK19__FlashStringHelperhEEvNS_15wm_debuglevel_tET_T0_                                                                                                                                 0        
    0

.xt.lit._ZN11WiFiManager8DEBUG_WMIPKcS2_EEvNS_15wm_debuglevel_tET_T0_                                                                                                                                                   0        
    0

.xt.lit._ZN11WiFiManager8DEBUG_WMIPKcEEvNS_15wm_debuglevel_tET_                                                                                                                                                         0        
    0

.xt.lit._ZNSt6vectorIPKcSaIS1_EEC5ESt16initializer_listIS1_ERKS2_                                                                                                                                                       8        
    0

.xt.lit._ZNSt12_Vector_baseIhSaIhEED5Ev                                                                                                                                                                                 0        
    0

.xt.lit._ZN11WiFiManager8DEBUG_WMI9IPAddressPKcEEvNS_15wm_debuglevel_tET_T0_                                                                                                                                            0        
    0

.xt.lit._ZN11WiFiManager8DEBUG_WMIiPK19__FlashStringHelperEEvNS_15wm_debuglevel_tET_T0_                                                                                                                                 0        
    0

.xt.lit._ZN11WiFiManager8DEBUG_WMI6StringPKcEEvNS_15wm_debuglevel_tET_T0_                                                                                                                                               0        
    0

.xt.lit._ZN11WiFiManager8DEBUG_WMI15StringSumHelperPKcEEvNS_15wm_debuglevel_tET_T0_                                                                                                                                     0        
    0

.xt.lit._ZNSt6vectorIhSaIhEE12emplace_backIJhEEEvDpOT_                                                                                                                                                                  8        
    0

.xt.lit._ZN11WiFiManager8DEBUG_WMIPK19__FlashStringHelperPcEEvNS_15wm_debuglevel_tET_T0_                                                                                                                                0        
    0

.xt.lit._ZN11WiFiManager8DEBUG_WMIPK19__FlashStringHelper16wifi_auth_mode_tEEvNS_15wm_debuglevel_tET_T0_                                                                                                                0        
    0

.xt.lit._ZNSt8functionIFvvEEaSERKS1_                                                                                                                                                                                    0        
    0

.xt.prop._ZN3UriD5Ev                                                                                                                                                                                                   12        
    0

.xt.prop._ZNK3Uri5cloneEv                                                                                                                                                                                              84        
    0

.xt.prop._ZN3Uri9canHandleERK6StringRSt6vectorIS0_SaIS0_EE                                                                                                                                                             36        
    0

.xt.prop._ZNSt14_Function_base13_Base_managerISt5_BindIFM11WiFiManagerFvvEPS2_EEE10_M_managerERSt9_Any_dataRKS9_St18_Manager_operation                                                                                 96        
    0

.xt.prop._ZNSt14_Function_base13_Base_managerISt5_BindIFM11WiFiManagerFvbEPS2_bEEE10_M_managerERSt9_Any_dataRKS9_St18_Manager_operation                                                                                96        
    0

.xt.prop._ZNSt14_Function_base13_Base_managerISt5_BindIFM11WiFiManagerFv18arduino_event_id_t20arduino_event_info_tEPS2_St12_PlaceholderILi1EES8_ILi2EEEEE10_M_managerERSt9_Any_dataRKSE_St18_Manager_operation         96        
    0

.xt.prop._ZNSt17_Function_handlerIFv18arduino_event_id_t20arduino_event_info_tESt5_BindIFM11WiFiManagerFvS0_S1_EPS4_St12_PlaceholderILi1EES8_ILi2EEEEE9_M_invokeERKSt9_Any_dataOS0_OS1_                                48        
    0

.xt.prop._ZNK6StringcvMS_KFvvEEv                                                                                                                                                                                       72        
    0

.xt.prop._ZNK6StringneEPKc                                                                                                                                                                                             36        
    0

[... 916 lines cut ...]
Info : [esp32s2] requesting target halt and executing a soft reset
[... cut ...]
in procedure 'program_esp'
in procedure 'program_error' called at file "C:/Users/Admin/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 171
at file "embedded:startup.tcl", line 1184
*** [upload] Error 1
================================================================================================== [FAILED] Took 8.40 seconds ================================================================================================== 
PS E:\RnD\T-DAQ\B.30_Firmware\t-daq>    
```

valeros, 2022-08-09:
Seems like a bug to me, please temporarily roll back to the previous version of the platform until it's fixed:
```ini
[env:env1]
platform = espressif32@~5.0.0
framework = arduino
board = esp32-s2-saola-1
...
```

closed 2022-08-16 by valeros

valeros, 2022-08-16:
Hi @FireDeveloper ! Could you please retest your project using the upstream version of the platform? 
```ini
[env:env1]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
board = esp32-s2-saola-1
```

FireDeveloper, 2022-09-17:
Hello @valeros, I didn't tested it until now because I moved to bare IDF and eclipse.
It flashed it but application doesn't boot. (With serial flashing it works as expected)
This message keeps printing on serial 115200
`ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x7 (TG0WDT_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40010d75
SPIWP:0xee
mode:QIO, clock div:2
load:0x3ffe6100,len:0x524
ets_loader.c 65 `

Jason2866, 2022-10-02:
@valeros Tried ESP-PROG with actual version. Does not work fully. `Upload and debug` fails. When flashing the sketch before and doing `Debug without upload` it does work.

Please reopen.

valeros, 2022-10-04:
@Jason2866 If it doesn't work with built-in usb/jtag please open a separate issue with comprehensive description of the observed behavior.

Jason2866, 2022-10-04:
@valeros  I found the reason for. Since our main app does **not** run at address 0x10000 the actual implementation does not work. I found a solution for this for us.

commit cdd0867d (2022-08-16): Update bootloader image headers before debugging or uploading via debug tools

This approach is less intrusive than merging the entire application into one binary
implemented in #006d64e8b268e479703a0aac7eed8bef1ebea587

In this implementation we safely copy the required bootloader binary to the
build directory, adjust the headers via esptoolpy and the "merge_bin" command
according to --flash-size and --flash-mode arguments.

Resolves #872