platformio/platform-espressif32 #874: Add usb_reset support to platform.ini

msxconsulting, 2022-08-09:
Currently, the reset method for `esptool` before flashing is hard-coded to `--before=default_reset` in at least one location in main.py (found under: `C:\Users\<user>\.platformio\platforms\espressif32\builder\main.py`):

https://github.com/platformio/platform-espressif32/blob/develop/builder/main.py#L405-L412

We are using an ESP32-C3 chip.  When using the ESP32-C3-DevKitM-1 development kit, the board programs and resets correctly using the USB serial programmer.  Our custom PCBA with the ESP32-C3 does not reset because the custom board uses the `default_reset` hard-coded above.

This is resolved by forcing `usb_reset` either by:

1) Using a custom `upload_command` in platform.ini OR
2) Modifying the `main.py` file above

It would be great if the `--before` option was not hard-coded and could be modified in platform.ini.  This issue request is for adding that feature.

More information can be found in my original post on the [Platform IO forums here](https://community.platformio.org/t/enabling-usb-serial-on-esp32-c3/29105).

valeros, 2022-08-09:
Hi @msxconsulting ! Thanks for reaching out. I've just pushed a small fix to allow this functionality, please retest with the upstream version of the platform:
```ini
platform = https://github.com/platformio/platform-espressif32.git
```

closed 2022-08-09 by valeros

ferm10n, 2022-08-30:
I found this looking for a way to download a program to an esp32 without resetting it (I'm manually putting it into download mode and using another esp as a programmer by joining the rx and tx pins between them). Similarly I've been crafting a custom command for esptool.

I can see that I'm using 5.1.1 for `platform = espressif32` but I'm getting this warning and it fails to program: `Warning! Ignore unknown configuration option 'upload.before_reset'`

Is there something else I have to update to get the benefits of these changes?

commit 358e8483 (2022-08-09): Allow overriding reset operation via project file

Resolve #874