platformio/platform-espressif32 #875: How to instruct PlatformIO to store the platform with versioning suffix?

lexologe, 2022-08-09:
Hello,

I am searching for a solution to have versioned platforms in my ~/.platformio/platforms folder.

If I have in platformio.ini

`platform = espressif32`

or if I use my own platform in platformio.ini

`platform = https://example.com/my-own-platform-espressif32.tar.gz`

In both cases the folder in ~/.platformio/platforms is called `espressif32`

Would it be possible to add any versioning information to the name of the folder in order to differentiate between both?

valeros, 2022-08-09:
Hi @lexologe ! What problem are you trying to solve? In a nutshell, there is no easy way to do what you want mainly because PlatformIO handles package names by its own rules.

lexologe, 2022-08-09:
Hi @valeros, the problem to solve is that the two platforms are based on different ESP IDF versions. If someone has already installed the `platform = espressif32` variant and opens my project, the platform in ~/.platformio/packages/espressif32 will not be updated to my custom one. In that case the project will not build correctly and I have to ask the developer to delete the folder before building my project.

If a Git repo is used, I can use a branch or a tag as a suffix to the URL:
e.g. `https://github.com/platformio/platform-espressif32.git@1.2.0`

If I use a ZIP or tar.gz package, I don't know how to add versioning information. I dont think that this will work:
e.g. `https://example.com/my-own-platform-espressif32-v1.2.3.tar.gz@1.2.3`
Or may be better or easier if a hash of the tar.gz file is used as a suffix to the `~/.platformio/packages/espressif32@<tar-gz-hash>`

valeros, 2022-08-09:
Are you in control of the `platformio.ini` files in your project examples? Why don't you purposely specify `platform = https://example.com/my-own-platform-espressif32-v1.2.3.tar.gz` in all of them? Alternatively, you can add [an extra script](https://docs.platformio.org/en/latest/scripting/index.html) that will verify whether the right platform is used (e.g. by checking the `platform.json` file) and simply raise an error if the platform is not the right one.

xsrf, 2022-08-22:
If i understand it correctly `~/.platformio/packages/espressif32` is always the latest / default one that PIO uses when you just specify `platform = espressif32`. However, you can specify a specific version `platform = espressif32 @ 5.1.0` in your `platformio.ini` and PIO will use it or download it for you.

closed 2022-10-09 by stale[bot]