platformio/platform-espressif32 #876: Missing platformio/toolchain-esp32ulp on Raspberry Pi 64 bit (linux_aarch64)

lexologe, 2022-08-09:
Hi,

I tried to compile on Raspberry Pi 64 bit lite and came across following problem:

`Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.22851.0
Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'`

Should I switch to 32 bit Raspberry OS? - or can I somehow allow the package manager to ignore the missing package, because I do not use the ULP in my project.

valeros, 2022-08-09:
Hi @lexologe ! Unfortunately, Espressif doesn't provide [packages](https://github.com/espressif/binutils-esp32ulp/releases/tag/v2.28.51-esp-20191205) for AARCH64. But there is an issue on this matter in their repository https://github.com/espressif/binutils-esp32ulp/issues/23.

closed 2022-08-09 by valeros

SerialVelocity, 2022-09-07:
Hey @valeros, Espressif has replied now on that ticket and a release has been cut in espressif/binutils-gdb. They mention adding it to IDF, is that something that means it will automatically be pulled in by platformio or will platformio have to switch to that new release independently?

webash, 2022-10-05:
Seems that's changed recently: https://github.com/espressif/esp-idf/commit/b1387fb7cb1761fdebd692f5d63f0be9fc2ccbe0

webash, 2022-10-06:
I'm using platformio under Esphome, so once I work out how to use pio PRs under that - I shall 😅