platformio/platform-espressif32 #877: v5.1.0 broked something related to LittleFS and SPIFFS with TinyPICO board... works in Arduino IDE.

sblantipodi, 2022-08-10:
This is a small code sample that read and write to Little FS on ESP32.

```
#include "WiFi.h"
#include <FS.h>
#include <LittleFS.h>

void setup() {
  Serial.begin(115200);
  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
  Serial.println("Setup done");
  if (!LittleFS.begin(true)) {
    Serial.println("Formatted");
  } else {
    Serial.println("Formatteddd");
    LittleFS.format();
  }
  File file = LittleFS.open("/prova.txt", "w");
  //Write to the file
  file.print("dsadasdsa");
  delay(1);
  //Close the file
  file.close();
  Serial.println("wrote");
  File file2 = LittleFS.open("/prova.txt", "r");
  if (!file2) {
    Serial.println("No Saved Data!");
  }
  Serial.println(file2.read());
}

void loop() {
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0) {
    Serial.println("no networks found");
  } else {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i) {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN) ? " " : "*");
      delay(10);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(5000);
}
```

if I use a TinyPICO board, this small snippet fails reading writing LittleFS when using 
`espressif32@5.1.0`
it works well when using
`espressif32@5.0.0`

I noticed that the only way to make it work with 
`espressif32@5.1.0`
is to use
`board_build.flash_mode = qio`
instead of
`board_build.flash_mode = dio`

but this does not have sense to me.

when using `espressif32@5.0.0`
that snippet works with both `QIO` and `DIO`.

the problem happen on TinyPICO board only, it does not happen on LolinD32 or other ESP32 boards I have.

Same snipped always works well with Arduino IDE when using Core 2.0.4.

Jason2866, 2022-08-10:
Same as https://github.com/platformio/platform-espressif32/issues/867
@valeros  I still think it is a good idea to change the default flash mode to `qio` to all newer ESP32x MCUs.

@sblantipodi There was a change/fix in the way how the Arduino Bootloaders are compiled in IDF for Arduino for core 2.0.4.

sblantipodi, 2022-08-10:
> @sblantipodi There was a change/fix in the way how the Arduino Bootloaders are compiled in IDF for Arduino for core 2.0.4.

thanks for the answer and for joining in @Jason2866 
you are referring to a "change/fix", is there a way I can fix this problem?

Jason2866, 2022-08-10:
@sblantipodi  First try flash mode `qio` for all boards you have. This should fix when flashing via Platformio. For generating a valid factory firmware you have to do changes.
Discussed [here](https://github.com/esphome/esp-web-tools/issues/278#issuecomment-1210669894)

sblantipodi, 2022-08-10:
> @sblantipodi First try flash mode `qio` for all boards you have. This should fix when flashing via Platformio. For generating a valid factory firmware you have to do changes. Discussed [here](https://github.com/esphome/esp-web-tools/issues/278#issuecomment-1210669894)

I noticed it, QIO is the flash mode needed to work well on all boards now.
Thanks!!!

Jason2866, 2022-08-15:
@valeros  Looking at this https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build-esp32.py#L306-L313 there is always the qio lib included since the variable there is not set from anyware. Or do i miss here something?

TD-er, 2022-08-15:
So then we could also set `build.arduino.memory_type` to `dio_qspi` in the board definitions?

Jason2866, 2022-08-15:
yep.

TD-er, 2022-08-15:
OK, this really has become my new nightmare....

I tested myself to make a build where I set the `memory_type` to `dio_qspi` along with the `flash_mode` to `dio` as shown above.

- OTA update on board previously flashed with `flash_mode` set to `DOUT` => Boots fine, but in DOUT mode, while build was set to use DIO
- OTA update on board previously flashed with `flash_mode` set to `DIO` => Boots fine, in DIO mode as expected.
- Serial update on same board I just OTA updated in previous step => Boots in DIO mode, **but LittleFS filesystem is gone!!!**

N.B. the old builds apparently must have had `qio_qspi` set as `memory_type` as these boards all have been flashed in the past few weeks during development.

cyberman54, 2022-08-16:
This issue also seems to affect Arduino `Preferences` class?
It does not work any more after ugrading v5.0.0 to v5.1.0

valeros, 2022-08-16:
@Jason2866 
> Looking at this https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build-esp32.py#L306-L313 there is always the qio lib included since the variable there is not set from anyware. Or do i miss here something?

TBH, I'm a bit unaware why the Espressif team decided to use the `qio_qspi` by default. What is correct behavior here? Should the build script dynamically configure the `memory_type` field according to `flash_mode`?

TD-er, 2022-08-16:
@valeros See also my findings here: https://github.com/letscontrolit/ESPEasy/pull/4200

TL;DR
Changing the `qio_qspi` to `dio_qspi` does still seem to work fine when I OTA update a board with LittleFS.
But flashing the same board via serial and thus updating the bootloader too, resulted in an unmountable file system.

Jason2866, 2022-08-16:
@valeros  I did tests this morning regarding to. When i changed the setup to `memory_type` the same as `flash_type` i had no boot loops nor a defect LittleFS.
When the are not align it did not always work.
BUT until now i **never** managed to get the device in flash mode `qio`.
Tbh i have no idea why the second stage bootloader always enables only `dio`
or when used `qout` it always ends in running mode `dout`.
Btw. opened #880 small changes/fixes for flash modes. I think not all changes are there in your latest commit

TD-er, 2022-08-16:
> Tbh i have no idea why the second stage bootloader always enables only `dio`

Not sure if that's true.
I have no idea  how to check the actual flash mode being used. It looks like the function to get the mode is reading a header. But I am not sure if that's a guarantee to be the same as the runtime mode used on the flash chip.

Jason2866, 2022-08-16:
@TD-er True! Good point.

Jason2866, 2022-08-16:
Opened issue https://github.com/espressif/arduino-esp32/issues/7140

TD-er, 2022-08-16:
Ah great :)
Yep good teamwork 👍 

I'm curious whether my unit here may indeed not support QIO but only QOUT.
And also on what basis the 2nd stage bootloader decides to switch on QIO or perhaps QOUT.

Jason2866, 2022-08-17:
> This issue also seems to affect Arduino `Preferences` class? It does not work any more after ugrading v5.0.0 to v5.1.0

@cyberman54 yes, since including the wrong SPI lib driver results in damaging flash content. This is hitting LittleFS, SPIFFS, NVS and all code stuff writing to flash.

valeros, 2022-08-17:
I've updated the Arduino package for ESP32 with the patch by @Jason2866 submitted in https://github.com/espressif/esp32-arduino-lib-builder/pull/80

valeros, 2022-08-17:
@Jason2866 I just glanced at https://github.com/espressif/esp32-arduino-lib-builder/pull/80 and didn't see any branches special for S3. Anyway, fixed in the latest package.

cyberman54, 2022-08-17:
How can i use the current version with the fix with platformio?

valeros, 2022-08-17:
@cyberman54 Just run `pio platform update espressif32` in the IDE terminal.

Jason2866, 2022-08-17:
@valeros Perfect! Thx, for your great support :-)

valeros, 2022-08-18:
@Jason2866 Do you mean a PR to https://github.com/espressif/esp32-arduino-lib-builder ? I wasn't planning on submitting a new PR, let's see what the maintainers say about yours.

Jason2866, 2022-08-18:
Fine, the PR does work. But it is not easy to understand at first sight was it does ;-)

sblantipodi, 2022-08-26:
@Jason2866 Hi, I'm from mobile now, I'll try it next week as soon as I will be back home. Thanks for quoting.

sblantipodi, 2022-08-28:
I confirm it works awesome, congrats guys!
Thank you!

closed 2022-08-28 by sblantipodi