platformio/platform-espressif32 #879: ESP32 Platform Framework Version 3.20004.0 leads to crash on specific Module

Rai-Sp, 2022-08-12:
ESP32 Platform Framework Version 3.20004.0 leads to crash on specific Module.

Simple Test App:
-----------------------------------------------------------
#include <Arduino.h>

unsigned lcount;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  ++lcount;
  Serial.printf("Nr: %u %s\n", lcount, ESP.getSdkVersion());
  delay(1000);
}
------------------------------------------------------------

Running on TTGO Module labelled "T8_V1.7.1 20190702" with Chip Revision 1:
.............................
Nr: 13 v4.4.1-472-gc9140caf8c
Nr: 14 v4.4.1-472-gc9140caf8c
.............................
works as expected.

Running on TTGO T-Display Module labelled "19-6-28 V1.1 XY-CP" with Chip Revision 3:
idf_monitor.py gives the following output:
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
0x40080400: _init at ??:?

entry 0x400805e4

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)

Backtrace:0x400833d1:0x3ffe3aa00x40087cfd:0x3ffe3ac0 0x4008ca49:0x3ffe3ae0 0x400d8eba:0x3ffe3c10 0x40082aed:0x3ffe3c40 0x400792ba:0x3ffe3c90  |<-CORRUPTED
0x400833d1: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402

0x40087cfd: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:128

0x4008ca49: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85

0x400d8eba: do_core_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:290
 (inlined by) start_cpu0_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:401

0x40082aed: call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/cpu_start.c:648

My first guess: The problem occurs on (some?) Modules with ESP32 Chip Revision 3.

Rai-Sp, 2022-08-12:
By the way, the problem did not exist with Framework Version 3.20003.220626

valeros, 2022-08-12:
Hi @Rai-Sp ! Does it work correctly in the Arduino IDE?

Rai-Sp, 2022-08-12:
Hello valeros,
I am not sure, that I got your question corretly.
I have installed Ardino IDE 1.8.19, using https://dl.espressif.com/dl/package_esp32_index.json (1.0.6)
The program runs on T-Display Module , but outputs "Nr: xxx v3.3.5-1-g85c43024c", meaning that a very different SDK Version is used.

valeros, 2022-08-12:
You installed an outdated package from the old index file, please uninstall it completely. You need the following index file https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json and the version 2.0.4

Rai-Sp, 2022-08-12:
Hello again,
I have changed ESP32 Package (now 2.0.4). The program runs on T-Display Module and outputs: "Nr: xxxx v4.4.1-472-gc9140caf8c", meaning that the correct SDK Version is used.

valeros, 2022-08-12:
Does it work with the ESP32 Chip Revision 3?

Rai-Sp, 2022-08-12:
By the way, I have tested a second T-Display Module with PlatformIO to make sure it's not a hardware problem. But the problem persists, so two samples ot the T-Display Module with Chip Revision 3 have the same problem with PlatformIO.

Rai-Sp, 2022-08-12:
Exactly the same piece of harware works with Arduino IDE.

valeros, 2022-08-12:
Please provide here your `platformio.ini` file.

Rai-Sp, 2022-08-12:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino

monitor_speed = 115200

valeros, 2022-08-12:
Please attach here also a screenshot of your board configuration in the Arduino IDE.

Rai-Sp, 2022-08-12:
One more piece of information:
I am using the T-Display Module in a quite complex application, built with ESP-IDF and Arduino Libraries.
The Version are exactly the ones labelled "v4.4.2" and "2.04", and evrything works as expected.

Rai-Sp, 2022-08-12:
[image]

Jason2866, 2022-08-12:
Please try
```
[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
board_build.flash_mode = qio
framework = arduino

monitor_speed = 115200
```

Rai-Sp, 2022-08-13:
It works with board_build.flash_mode = qio.
Thanks for your assistence.

Anyway, board_build.flash_mode = qio was not necessary in previous versions.

By the way, I know that TTGO T-Display Module is not kind of
"esp-wrover-kit" (has no PSRAM), but I left it as I am using
the original espressif Wrover-Kit quite often (for debugging).

Another strange experience:
When everything was fine, "just for fun", I removed .platformio folder
completely and let PlatformIO Installer do its work (took a very long time).
After that, it also worked without board_build.flash_mode = qio.
Before, I had just upgraded from 5.0.0 to 5.1.0; maybe something
went wrong during this step?

So, I close the issue now, and many thanks again.

closed 2022-08-13 by Rai-Sp