platformio/platform-espressif32 #881: Blufi library support, unable to ld

chuhangRadarDev, 2022-08-15:
```
c:/users/fangyu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: .pio/build/esp32c3-4M-NoBluetooth/src/blufi.cpp.o: in function `.L0 ':
C:\Users\fangyu\Documents\PlatformIO\Projects\wifiVSRadar/src/blufi.cpp:213: undefined reference to `esp_blufi_adv_stop()'
c:/users/fangyu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: .pio/build/esp32c3-4M-NoBluetooth/src/blufi.cpp.o: in function `example_event_callback':
C:\Users\fangyu\Documents\PlatformIO\Projects\wifiVSRadar/src/blufi.cpp:219: undefined reference to `esp_blufi_adv_start()'
c:/users/fangyu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\fangyu\Documents\PlatformIO\Projects\wifiVSRadar/src/blufi.cpp:253: undefined reference to `esp_blufi_disconnect()'
c:/users/fangyu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: .pio/build/esp32c3-4M-NoBluetooth/src/blufi_init.cpp.o: in function `esp_blufi_gap_register_callback()':
C:\Users\fangyu\Documents\PlatformIO\Projects\wifiVSRadar/src/blufi_init.cpp:37: undefined reference to `esp_blufi_gap_event_handler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*)'
c:/users/fangyu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\fangyu\Documents\PlatformIO\Projects\wifiVSRadar/src/blufi_init.cpp:43: undefined reference to `esp_blufi_gap_event_handler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*)'
c:/users/fangyu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: .pio/build/esp32c3-4M-NoBluetooth/src/blufi_security.cpp.o: in function `blufi_dh_negotiate_data_handler(unsigned char*, int, unsigned char**, int*, bool*)':
C:\Users\fangyu\Documents\PlatformIO\Projects\wifiVSRadar/src/blufi_security.cpp:113: undefined reference to `btc_blufi_report_error(esp_blufi_error_state_t)'
c:/users/fangyu/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\fangyu\Documents\PlatformIO\Projects\wifiVSRadar/src/blufi_security.cpp:89: undefined reference to `btc_blufi_report_error(esp_blufi_error_state_t)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32c3-4M-NoBluetooth\firmware.elf] Error 1
========================================================================= [FAILED] Took 109.62 seconds =========================================================================
```
My configuration()：
```ini
[env:esp32c3-4M-NoBluetooth]
platform = espressif32
framework = arduino
board_build.filesystem = littlefs
board = esp32-c3-devkitm-1
; upload_port=COM4
upload_speed = 921600
monitor_speed = 115200
; board_build.partitions = partition.csv
board_build.partitions = partitionNoBluetooth.csv
lib_deps =
  knolleary/PubSubClient @ ^2.8
  bblanchon/ArduinoJson @ ^6.19.4
  ottowinter/ESPAsyncWebServer-esphome @ ^2.1.0
```
What is the solution? thanks!
[image]

Jason2866, 2022-08-15:
Please provide a MCVE. The Platformio setup only is not enough.

valeros, 2022-08-16:
@chuhangRadarDev You sure you code is compatible with the latest Arduino core v2.0.4? Does it work in the Arduino IDE?

chuhangRadarDev, 2022-08-17:
> @chuhangRadarDev You sure you code is compatible with the latest Arduino core v2.0.4? Does it work in the Arduino IDE?

The project uses the Arduino framework. This part calls the Blufi distribution network function at the bottom of esp32 IDF. However, the header file can be found and compiled, but it cannot be linked. I haven't experimented in Arduino IDE yet.

chuhangRadarDev, 2022-08-17:
> Please provide a MCVE. The Platformio setup only is not enough.

I will creat one..

chuhangRadarDev, 2022-08-17:
@valeros I use latest Espressif 32 SDK(version 5.1.0) in platformio
[image]
Maybe some code blocked in  [libbt.a](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32c3/lib/libbt.a)
as BLUFI_INCLUDED = FALSE in precompiled code?
[image]

valeros, 2022-08-17:
> Maybe some code blocked in libbt.a

Apparently, if the the underlying IDF framework was compiled without BluFi then you won't be able to compile projects that require this functionality. But again, please check whether your project works in the Arduino IDE.

Jason2866, 2022-08-17:
Since libbt.a is a closed source lib which is "just" linked no compile setting in IDF will or can change this lib.
So if a needed reference in the lib is not there anymore -> Game Over.

Jason2866, 2022-08-17:
[BluFI](https://github.com/espressif/esp32-arduino-lib-builder/blob/master/configs/defconfig.esp32c3) support is enabled

chuhangRadarDev, 2022-08-18:
@Jason2866 @valeros My Blufi [MCVE](https://github.com/chuhangRadarDev/BlufiMCVE),@valeros I haven't used Arduino IDE, so I'm not familiar with compiling with Arduino IDE

chuhangRadarDev, 2022-08-18:
[Github Actions](https://github.com/chuhangRadarDev/BlufiMCVE/runs/7892518426?check_suite_focus=true#step:5:114)

chuhangRadarDev, 2022-08-18:
Solved， need extern "C".Thanks All!

closed 2022-08-18 by chuhangRadarDev