platformio/platform-espressif32 #882: undefined reference with esp-idf dependency

elestedt, 2022-08-15:
I have `platformio.ini` as

```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = espidf
lib_deps = 
    https://github.com/tonyp7/esp32-wifi-manager.git#v3.3.1
```

And the I've added `Kconfig` from that dep to my `src/Kconfig` as required right now. But when I try to link (using that projects simple example `(main.c)[https://github.com/tonyp7/esp32-wifi-manager/blob/master/examples/default_demo/main/user_main.c]`) I'm getting some undefined reference to things the library is trying to embed, but failing when running via PlatformIO

```
${HOME}/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/libceb/libesp32-wifi-manager.a(http_app.o):(.literal.http_server_get_handler+0x1c): undefined reference to `_binary_index_html_end'
${HOME}/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/libceb/libesp32-wifi-manager.a(http_app.o):(.literal.http_server_get_handler+0x20): undefined reference to `_binary_index_html_start'
${HOME}/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/libceb/libesp32-wifi-manager.a(http_app.o):(.literal.http_server_get_handler+0x2c): undefined reference to `_binary_code_js_end'
${HOME}/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/libceb/libesp32-wifi-manager.a(http_app.o):(.literal.http_server_get_handler+0x30): undefined reference to `_binary_code_js_start'
${HOME}/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/libceb/libesp32-wifi-manager.a(http_app.o):(.literal.http_server_get_handler+0x40): undefined reference to `_binary_style_css_end'
${HOME}/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/esp32-s3-devkitc-1/libceb/libesp32-wifi-manager.a(http_app.o):(.literal.http_server_get_handler+0x44): undefined reference to `_binary_style_css_start'
```

Running all the latest stable versions via Visual Studio Code at time of writing.

valeros, 2022-08-16:
Hi @elestedt ! You need to manually add extra files as described here https://docs.platformio.org/en/latest/platforms/espressif32.html#embedding-binary-data

closed 2022-08-16 by valeros

elestedt, 2022-08-16:
@valeros That seems to indicate how to include things from my own code. But this is things from a dependency that is not being included properly.
The page you linked doesn't deal with esp dependencies that includes things.

valeros, 2022-08-17:
Unfortunately, this is a limitation of the current ESP-IDF integration. The code model generated by the IDF build system doesn't contain all necessary information to avoid adding those third-party things manually to the `platformio.ini` file.

elestedt, 2022-08-17:
Any idea of when that will be fixed? That and the thing with menuconfig things not showing up without manually copying the KConfig is currently preventing me from using PlatformIO for this project - which I really like otherwise.

valeros, 2022-08-17:
Unfortunately there is no ETA, mainly because the CMake's [File-based API](https://cmake.org/cmake/help/latest/manual/cmake-file-api.7.html) doesn't export a fully comprehensive code model to reproduce the entire build process in PlatformIO. 

> That and the thing with menuconfig things not showing up without manually copying the KConfig

Not sure what you mean, what's the error message in this case?

elestedt, 2022-08-17:
For menuconfig items to be shown from esp-idf component I have to copy the content of the components Kconfig to my src/Kconfig

valeros, 2022-08-17:
Where is that  esp-idf component is located? Is it a part of the IDF framework?

elestedt, 2022-08-17:
Here is an exanple file https://github.com/tonyp7/esp32-wifi-manager/blob/master/Kconfig
To get that into my project, with the same platformiu file as above, I have to copy it to src/Kconfig.

valeros, 2022-08-17:
It seems that `esp32-wifi-manager` is a full-featured IDF component and in your configuration above you're importing it as a generic library. This way the component is not processed by the underlying IDF build system and hence is not visible in menuconfig. More detailed info on how to properly add IDF components to PlatformIO is described here https://docs.platformio.org/en/latest/frameworks/espidf.html#esp-idf-components

elestedt, 2022-08-17:
Thanks. It's not really clear when it can't just be a "library dependency"

elestedt, 2022-08-17:
But I'll look at this and hopefully it resolves both issues

valeros, 2022-08-17:
It won't resolve the first issue with embedding extra files to the final firmware, but should help with the menuconfig problem.