platformio/platform-espressif32 #889: Documentation for how to update IDF version

johnboiles, 2022-08-22:
Forgive me if this is documented anywhere, I haven't been able to find it. I frequently new IDF features as they're released and am trying to figure out how to build a custom `platform-espressif32` using a specific SHA from IDF. I'm happy to help document and possibly automate this once I understand how to do it.

I see that there's a [`builder/main.py`](https://github.com/platformio/platform-espressif32/blob/develop/builder/main.py) script and a [builder/frameworks/espidf.py](https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/espidf.py). My understanding is that those get invoked to build `platform-espressif32` and `framework-espidf` respectively, but I don't see anywhere that the actual IDF version, SHA, or local path gets set. 

Looking at the [`framework-espidf` package `tar.gz`](https://api.registry.platformio.org/v3/packages/platformio/tool/framework-espidf) files, it looks like it has pretty much all of IDF in there. I see [`IDF_PATH` is set from `FRAMEWORK_DIR`[(https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/espidf.py#L218),  but I don't see where the IDF files get into `FRAMEWORK_DIR` to begin with.

Help here would be much appreciated. And I'm happy to document this (maybe in the README) once we have a procedure worked out. @Jason2866 I [know](https://github.com/platformio/platform-espressif32/issues/619#issuecomment-1002306803) you've done this before (and thank you so much for doing it!!) do you have the process written down anywhere?

Jason2866, 2022-08-22:
@johnboiles Currently it is not possible to update the IDF 4.4.x version in PlatformIO since the (c)make part in actual IDF4.4.x is updated and not anymore compatible to the older version and build process in PlatformIO.
To get the actual IDF 4.4.x version running i did some reverts in IDF of this part. Dirty.... For a real solution Platformio crew have to update this part. I do not have the knowledge for.
If this is done it is pretty straightforward, clone IDF repo. Activate the github actions in your fork. Tag the commit you want with a release tag. Github actions will do a complete IDF package. You just have to add to this zip a valid package.json (just take a look in a current used in Platformio). If the json is in the package you can use the zip in Platformio via `platform_packages`

johnboiles, 2022-08-24:
Thank you @Jason2866!! I'll dig into your `tasmota/esp-idf` repo. Are the cmake reverts in your git or do you make them manually in the zip file after releasing the `org` version of the zip?

So then I can point `platform_packages` to the manually patched version of the zip right? Like this:
```
platform_packages = framework-espidf @ https://github.com/tasmota/esp-idf/releases/download/v4.4.2/esp-idf-v4.4.2.zip
```

Jason2866, 2022-08-24:
Yes, manually changed, since it is a dirty hack. It is not in the repo. Idk if the hack will work with newer IDF commits....
Your assumption how to use the framework is correct.

johnboiles, 2022-08-24:
Ok! I'll diff the contents of `esp-idf-v4.4.2.zip` and `esp-idf-v4.4.2-org.zip` to find out your hacks!

valeros, 2022-08-31:
Hi @johnboiles ! Currently, PlatformIO supports only stable versions of frameworks and IDF is no exception. Once the stable v5.0 is released it will be updated here as well.

johnboiles, 2022-09-01:
IDF 4.4.2 is stable!

valeros, 2022-09-01:
> IDF 4.4.2 is stable!

Sorry, my bad, I misread the issue. There is no precise ETA, but support for v4.4.2 will be added within next weeks.

Jason2866, 2022-09-02:
@johnboiles  I found the reason for failing. IDF Components Manager is now enabled by default.
If you change https://github.com/tasmota/esp-idf/commit/ee6844367a2816e4182ffb8db1909ff4fad344c9so 
it will work again with PlatformIO. The change does disable the Component Manager by default.

Jason2866, 2022-09-02:
@johnboiles  I have changed in my espressif IDF fork (branch release/v4.4) the github actions workflow. A release Tag now generates a small release which can be used directly with platformio. No more manually work :-)

johnboiles, 2022-09-02:
You are awesome @Jason2866!!!! I'm excited to try this

commit 0abe8d30 (2022-09-13): Update ESP-IDF to v4.4.2

Resolve #906, resolve #889
closed 2022-09-13 by valeros