platformio/platform-espressif32 #893: Please add toolchain 11.2.0 to the pio package registry

X-Ryl669, 2022-08-23:
These [toolchains](https://github.com/espressif/crosstool-NG/releases/tag/esp-2022r1) aren't in the PIO registry, so it's impossible to use them, thus it's impossible to use a framework-espidf based on master (for development) since cmake refuse to build without the updated toolchain.

Jason2866, 2022-08-24:
toolchains from and for espressif in Platformio Registry are maintained since a while from espressif.

X-Ryl669, 2022-08-25:
Thanks, I've also asked [there](https://github.com/espressif/crosstool-NG/issues/20#issuecomment-1224940278), but so far, no reaction. Maybe I should raise a new issue instead of hitchhiking a closed one...

Jason2866, 2023-04-25:
Issue can be closed.