platformio/platform-espressif32 #896: On-board LED Fade not working

mohitreact, 2022-08-28:
Attempting to upload the following code to ESP32 from Platform IO (Core 6.1.4, Home 3.4.3) vs Arduino IDE 1.8.19. The on-board LED on the ESP32 is on GPIO pin 2. This works (LED fade is visible) when flashed from the Arduino IDE, and does not work when flashed from Platform IO (LED fade is not visible). 

Verified that the correct board "Espressif ESP32 Dev Module" (or esp32dev) is selected on both. Also verified that both IDEs "see" pin 2 as the LED (Blink with digitalWrite(2, HIGH/LOW) works on both IDEs). Following is the code used, succeeded by the "platformio.ini" file text (attempted with minimal settings, to no avail). Does Platform IO have implementation issues with PWM on ESP32?

`#include <esp32-hal.h>
#include <esp32-hal-ledc.h>
#include <Arduino.h>
int frequency = 200000;
int duty = 10;
int pin = 2;
void setup()
{
  ledcSetup(2, frequency,duty);
  ledcAttachPin(pin, 2);
  ledcWrite(2,1023);
}

void loop()
{
  for (int i = 0 ; i < 1023; i++)
  {
    ledcWrite(2,i);
    delay(2);
  }
}`

`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_archive = false
board_build.partitions = min_spiffs.csv
upload_speed = 921600
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
build_flags = 
	-DCONFIG_SPIRAM_CACHE_WORKAROUND
	-DCORE_DEBUG_LEVEL=0
lib_deps = 
	contrem/arduino-timer@^2.3.1
	olkal/HX711_ADC@^1.2.12`

valeros, 2022-08-31:
Hi @mohitreact ! You sure you're compiling the same code in the Arduino IDE and PlatformIO? I just tried the example you provided above and it doesn't work in the Arduino IDE as well. It fails with the following error:
```
E (13) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=100
``` 

If I adjust the frequency to `20000` then it works in both PlatformIO and Arduino IDE. Am I missing something?

mohitreact, 2022-09-01:
Hi @valeros - I'm compiling the exact same code across both, and it works on the Arduino IDE, and not on PlatformIO.

However, there are two things I discovered: 
- A poor practice bug - the frequency is declared as an int, and is assigned as 200,000. This is probably an overflow that the Arduino IDE is allowing (in Arduino 1.8.5 atleast) while PlatformIO + VSCode don't allow this. This might be one source of the difference in results.
- Correcting for this to an unsigned long declaration lets me declare a maximum frequency of 78300L (LED fades till this frequency, but doesn't light up at all beyond this frequency). 

My production system has a PWM frequency of 200,000 in it. Let me write back with more findings from the actual system

valeros, 2022-09-01:
> This is probably an overflow that the Arduino IDE is allowing (in Arduino 1.8.5 atleast) while PlatformIO + VSCode don't allow this.

I doubt it's an issue here. The integer can hold values up to `2147483647`.

> I'm compiling the exact same code across both, and it works on the Arduino IDE, and not on PlatformIO.

Could you please check what's printed to the serial monitor after you flash the board within PlatformIO?

mohitreact, 2022-09-07:
This is probably the resolution for the problem - the PWM frequency is limited by the duty resolution demanded. This thread on ESP32 maximum resolution [here](https://forum.micropython.org/viewtopic.php?f=18&t=3717#p24383) provides a clue as to why PWM frequency is being limited to ~78 kHz. The relevant section in the Espressif documentation is [here](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#ledpwm). 

When probed with an oscilloscope, code uploaded by both Platform IO (with firmware frequency = 78.3 kHz) and by Arduino IDE (with firmware frequency = 200kHz) produces a real world frequency of 78.3 kHz. The difference in behaviour seems to be in how Arduino IDE handles maximum frequency of PWM (limits it to the realistic maximum value itself), and how Platform IO handles maximum frequency (seems to treat anything > maximum theoretical as zero).

mohitreact, 2022-10-10:
Closing this issue based on my last comment - maximum frequency that Arduino PWM can put out is 78.3kHz. Arduino IDE caps any value above this automatically to 78.3kHz while PlatformIO seems to treat any value above this as zero.

closed 2022-10-10 by mohitreact