platformio/platform-espressif32 #897: Apple Silicon support without Rosetta

mailinger-mate, 2022-08-31:
Hello,

ESP-IDF v5.0 release should now have native support of macos-arm64 for Apple Silicon:
https://github.com/espressif/esp-idf/releases/tag/v5.0-beta1

Until recently we still needed Rosetta to build a project on M1:
https://github.com/platformio/platform-espressif32/issues/724

Would it possible to roll out native support *without* Rosetta on PlatformIO too?

Thank you!

valeros, 2022-08-31:
Hi @mailinger-mate ! Will be added once a stable IDF v5.0 is available.

Jason2866, 2022-08-31:
@mailinger-mate  If you do not want to wait you can use my fork. I have compiled the espressif toolchains ([8.4.0-2021r2-patch3) for Apple ARM. The are used for Arduino and IDF 4.4. Performance gain between 15 and 25%.
You can use with:
```
platform  = https://github.com/jason2866/platform-espressif32.git#Tasmota/205
```

Jason2866, 2022-08-31:
```
Compiling .pio/build/esp32-s3/bt/common/btc/profile/esp/blufi/blufi_protocol.o
Compiling .pio/build/esp32-s3/bt/common/osi/alarm.o
Compiling .pio/build/esp32-s3/bt/common/osi/allocator.o
Compiling .pio/build/esp32-s3/bt/common/osi/buffer.o
Compiling .pio/build/esp32-s3/bt/common/osi/config.o
Compiling .pio/build/esp32-s3/bt/common/osi/fixed_queue.o
Compiling .pio/build/esp32-s3/bt/common/osi/pkt_queue.o
Compiling .pio/build/esp32-s3/bt/common/osi/fixed_pkt_queue.o
Compiling .pio/build/esp32-s3/bt/common/osi/future.o
Compiling .pio/build/esp32-s3/bt/common/osi/hash_functions.o
Compiling .pio/build/esp32-s3/bt/common/osi/hash_map.o
Compiling .pio/build/esp32-s3/bt/common/osi/list.o
Compiling .pio/build/esp32-s3/bt/common/osi/mutex.o
Compiling .pio/build/esp32-s3/bt/common/osi/thread.o
Compiling .pio/build/esp32-s3/bt/common/osi/osi.o
Compiling .pio/build/esp32-s3/bt/common/osi/semaphore.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_a2dp_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_avrc_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_bt_device.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_bt_main.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_gap_ble_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_gap_bt_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_gatt_common_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_gattc_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_gatts_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_hidd_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_hf_ag_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_hidh_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_hf_client_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/api/esp_spp_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/ar/bta_ar.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/av/bta_av_aact.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/av/bta_av_act.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/av/bta_av_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/av/bta_av_cfg.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/av/bta_av_ci.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/av/bta_av_main.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/av/bta_av_sbc.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/av/bta_av_ssm.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/dm/bta_dm_act.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/dm/bta_dm_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/dm/bta_dm_cfg.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/dm/bta_dm_ci.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/dm/bta_dm_co.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/dm/bta_dm_main.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/dm/bta_dm_pm.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/dm/bta_dm_sco.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/dm/bta_dm_qos.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gatt_common.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gattc_act.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gattc_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gattc_cache.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gattc_ci.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gattc_co.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gattc_main.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gattc_utils.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gatts_act.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gatts_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gatts_co.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gatts_main.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/gatt/bta_gatts_utils.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/hd/bta_hd_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/hd/bta_hd_act.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/hd/bta_hd_main.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/hh/bta_hh_act.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/hh/bta_hh_api.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/hh/bta_hh_cfg.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/hh/bta_hh_le.o
Compiling .pio/build/esp32-s3/bt/host/bluedroid/bta/hh/bta_hh_main.o
[... 56 lines cut ...]
Indexing .pio/build/esp32-s3/bootloader/esp-idf/micro-ecc/libmicro-ecc.a
Linking .pio/build/esp32-s3/bootloader.elf
Building .pio/build/esp32-s3/bootloader.bin
esptool.py v4.2.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Linking .pio/build/esp32-s3/firmware.elf
Retrieving maximum program size .pio/build/esp32-s3/firmware.elf
Checking size .pio/build/esp32-s3/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.4% (used 11260 bytes from 327680 bytes)
Flash: [=         ]   5.1% (used 155661 bytes from 3080192 bytes)
Building .pio/build/esp32-s3/firmware.bin
esptool.py v4.2.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
====================================== [SUCCESS] Took 50.50 seconds ======================================

Environment    Status    Duration
-------------  --------  ------------
esp32-s3       SUCCESS   00:00:50.501
======================================= 1 succeeded in 00:00:50.501 =======================================
```

Jason2866, 2022-09-13:
Good news, espressif released new 8.4.0 toolchains too. Now with native Apple Silicon support :-)

Jason2866, 2023-01-17:
Rosetta is nowhere used anymore.

mickeyl, 2023-09-06:
A fresh install still leads to the following output:

```
UserSideException: Processing esp32-c3-wican (platform: espressif32; board: esp32-c3-devkitm-1; framework: espidf)
--------------------------------------------------------------------------------
/Users/mickey/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2.0 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-builtin) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50101.230828 (5.1.1) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.9.0 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 12.2.0+20230208
Reading CMake configuration...
OSError: [Errno 86] Bad CPU type in executable: '/Users/mickey/.platformio/packages/tool-cmake/bin/cmake':
  File "/Users/mickey/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/mickey/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/mickey/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/mickey/.platformio/platforms/espressif32/builder/main.py", line 346:
    target_elf = env.BuildProgram()
  File "/Users/mickey/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/Users/mickey/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "/Users/mickey/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/Users/mickey/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/Users/mickey/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/Users/mickey/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/piobuild.py", line 342:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/Users/mickey/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 662:
    return method(*args, **kw)
  File "/Users/mickey/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/mickey/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/mickey/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1341:
    project_codemodel = get_cmake_code_model(
  File "/Users/mickey/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 222:
    run_cmake(src_dir, build_dir, extra_args)
  File "/Users/mickey/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 757:
    run_tool(cmd)
  File "/Users/mickey/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 712:
    result = exec_command(cmd, env=idf_env)
  File "/Users/mickey/.platformio/penv/lib/python3.9/site-packages/platformio/proc.py", line 114:
    with subprocess.Popen(*args, **kwargs) as p:
  File "/Applications/Xcode-15.0.0-Beta.6.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/subprocess.py", line 951:
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Applications/Xcode-15.0.0-Beta.6.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/subprocess.py", line 1821:
    raise child_exception_type(errno_num, err_msg, err_filename)
========================== [FAILED] Took 0.54 seconds ==========================
```

mickeyl, 2023-09-06:
So, for some reason, the wrong `cmake` seems to be installed.

astorun, 2024-03-07:
I am getting the same error on a brand new MacBook Pro M3 Pro CPU with new fresh install.
Does it still an issue?

mickeyl, 2024-03-07:
I didn't pursue it any further, since I moved away from `platformio` towards _pure_ ESP-IDF. If you're using that, there are no dependencies on Rosetta.

kquinsland, 2024-11-17:
I read through this thread twice; please excuse me if I missed it.
What's keeping this open?

I can still reproduce - without `rosetta2` installed - the issue so it's not _fixed_ but I am using relatively recent versions of everything so it seems like it's [not? enough to have ESP-IDF 5.0+](https://github.com/platformio/platform-espressif32/issues/897#issuecomment-1233024842):

```shell
❯ esphome run ./projects/my_project.yaml
INFO ESPHome 2024.10.3
<...>
Processing my-project (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
<...>
Platform Manager: espressif32@5.4.0 has been installed!
Tool Manager: toolchain-riscv32-esp@8.4.0+2021r2-patch5 has been installed!
Tool Manager: framework-arduinoespressif32@3.20005.220925 has been installed!
Tool Manager: tool-esptoolpy@1.40400.0 has been installed!
Tool Manager: tool-scons@4.40700.0 has been installed!
Library Manager: AsyncTCP-esphome@2.1.4 has been installed!
Library Manager: ESPAsyncWebServer-esphome@3.2.2 has been installed!
Library Manager: ArduinoJson@6.18.5 has been installed!
<....>
sh: /Users/karl/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-g++: Bad CPU type in executable
```

carry0987, 2025-03-02:
> I read through this thread twice; please excuse me if I missed it. What's keeping this open?
> 
> I can still reproduce - without `rosetta2` installed - the issue so it's not _fixed_ but I am using relatively recent versions of everything so it seems like it's [not? enough to have ESP-IDF 5.0+](https://github.com/platformio/platform-espressif32/issues/897#issuecomment-1233024842):
> 
> ❯ esphome run ./projects/my_project.yaml
> INFO ESPHome 2024.10.3
> <...>
> Processing my-project (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
> <...>
> Platform Manager: espressif32@5.4.0 has been installed!
> Tool Manager: toolchain-riscv32-esp@8.4.0+2021r2-patch5 has been installed!
> Tool Manager: framework-arduinoespressif32@3.20005.220925 has been installed!
> Tool Manager: tool-esptoolpy@1.40400.0 has been installed!
> Tool Manager: tool-scons@4.40700.0 has been installed!
> Library Manager: AsyncTCP-esphome@2.1.4 has been installed!
> Library Manager: ESPAsyncWebServer-esphome@3.2.2 has been installed!
> Library Manager: ArduinoJson@6.18.5 has been installed!
> <....>
> sh: /Users/karl/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-g++: Bad CPU type in executable

Is here any update?

JeanBlignaut, 2025-07-07:
I'm still getting this with latest installs (no rosetta) in 2025 while Arduino die does not have this issue - what is the hold up?

Jason2866, 2025-07-08:
I do not see the use of Rosetta.

washley, 2025-07-09:
Still failing on apple silicon macs, in my case it's hitting an x86_64 ninja binary. This is via `esphome` installed with homebrew.
```
$ esphome run test.yaml 
INFO ESPHome 2025.6.3
INFO Reading configuration test.yaml...
INFO Generating C++ source...
INFO Core config, version or integrations changed, cleaning build files...
INFO Cloning https://github.com/espressif/esp-protocols.git@mdns-v1.8.2
INFO Fetching mdns-v1.8.2
INFO Compiling app...
Processing esp32c6 (board: esp32-c6-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32C6 160MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.9.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 13.2.0+20240530
Reading CMake configuration...
-- Found Git: /opt/homebrew/bin/git (found version "2.50.1")
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /Users/washley/.platformio/packages/framework-espidf/tools/cmake/project.cmake:564 (__project):
  Running

   '/Users/washley/.platformio/packages/tool-ninja/ninja' '--version'

  failed with:

   Unknown system error -86
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

======================================================== [FAILED] Took 0.48 seconds ========================================================
```

Jason2866, 2025-07-13:
@washley I could update the build environment for mklittlefs. Now in fork pioarduino the used mklittlefs tools has native Apple ARM versions.
Ninja is updated too and uses now a universal Binary