platformio/platform-espressif32 #899: When will menuconfig be updated

versamodule, 2022-09-02:
When it comes to the ESP32 platform, menuconfig is well…To DOS like. When others have made it far more GUI like. When is PlatformIO going to put more effort into doing the same ?
Espressif’s IDF extension uses VSCode, so I can see no reason why PlatformIO cannot do the same when it comes to menuconfig. Even Eclipse using ESP-IDF has the same GUI interface.

Here is PlatformIO’s menuconfig.

[image]

This is Espressif’s IDF extension, and Eclipse is similar. 

[image]

Jason2866, 2022-09-03:
Yes, the Espressif VSC extension does look nice. Imho it is cluttered. The old style menuconfig is not looking so nice but is structured.

versamodule, 2022-09-03:
"Imho it is cluttered" you really thing so? 
I am shocked that someone would think that. It's all well organized, and it's also structured.  If you are saying that because of the photo I would suggest trying it. Can't see how anyone would want to use the current method compared to the GUI way.  Not to mention you have to use the J and K keys to move up and down.