platformio/platform-espressif32 #903: Please add Wemos Lolin S3

davy39, 2022-09-03:
Hello,
I've just purchased a Lolin S3 which has been [added on arduino-esp32](https://github.com/espressif/arduino-esp32/pull/7094).
Could someone add this board to platformio please ?
Thanks by advance !

davy39, 2022-09-04:
I don't know if it's optimal since I'm a newbie, but I've created the following json that seems pretty functional.
Could someone tell me if I'm missing something ?
```title=lolin_s3.json json
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_LOLIN_S3",
      "-DARDUINO_USB_MODE=1",
      "-DARDUINO_USB_CDC_ON_BOOT=1",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "lolin_s3"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "LOLIN S3",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 1310720,    
    "use_1200bps_touch": true,
    "wait_for_upload_port": true,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://www.wemos.cc/en/latest/s3/index.html",
  "vendor": "Wemos"
}
```

Jason2866, 2022-09-04:
`maximum_size` is wrong. For 16MB it is `16777216`

davy39, 2022-09-04:
I didn't understand neither... But I took same parameter than in Wemos PR [here](https://github.com/espressif/arduino-esp32/pull/7094/files ).
Do you think this is a mistake ?

Jason2866, 2022-09-04:
Yes, it is wrong in the Arduino PR from Wemos.

davy39, 2022-09-04:
Thanks for your replies, I've modified my json. Shall I open a PR ?

Jason2866, 2022-09-04:
Good idea

closed 2022-09-26 by valeros