platformio/platform-espressif32 #906: how to use ESP-IDF v4.4.2

looping84, 2022-09-05:
https://github.com/espressif/esp-idf/releases

espidf v4.4.2 fixed lots of bugs。 it's a stable version

platformio only has v4.4.0 version and v5.0 beta

[image]

Jason2866, 2022-09-05:
In short, no easy way. Either Platformio or IDF needs a (small) change.
You have to wait for a official release or you can try my build in this [fork](https://github.com/Jason2866/platform-espressif32/tree/Tasmota/205)
Use with:
```
platform = https://github.com/Jason2866/platform-espressif32.git#Tasmota/205
```

RoscoeTheDog, 2022-09-09:
> In short, no easy way. Either Platformio or IDF needs a (small) change. You have to wait for a [official [...]

Could you kindly add board definitions for `esp32-s2-saola-1` on your branch? I'd like to use it as well but that is the board variant I am working with.

Jason2866, 2022-09-10:
Board esp32-s2-saola-1 is there!

RoscoeTheDog, 2022-09-10:
> Board esp32-s2-saola-1 is there!

Graci. I tried last night and it wouldn't build. Funny it should work now unless you added it overnight upon request. Regardless, everythings working as expected. Thanks!

Jason2866, 2022-09-10:
@RoscoeTheDog  I did not change anything.

commit 0abe8d30 (2022-09-13): Update ESP-IDF to v4.4.2

Resolve #906, resolve #889
closed 2022-09-13 by valeros