platformio/platform-espressif32 #912: 5.1.1  esp32s3box  sdkconfig.h  no such file

txp666, 2022-09-14:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results
[image]

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32s3box]
platform = espressif32
board = esp32s3box
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Additional info

txp666, 2022-09-14:
it's not worked

maxgerhardt, 2022-09-14:
Compiles just fine with your platformio.ini + code. 

```
Linking .pio\build\esp32s3box\firmware.elf
Retrieving maximum program size .pio\build\esp32s3box\firmware.elf
Checking size .pio\build\esp32s3box\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.9% (used 12924 bytes from 327680 bytes)
Flash: [==        ]  15.5% (used 203477 bytes from 1310720 bytes)
```

If it still fails, your files still have a problem. Delete the folders `C:\Users\<user>\.platformio\.cache`,
`C:\Users\<user>\.platformio\packages`, and `C:\Users\<user>\.platformio\platforms` folders and build the project again to get a clean download.

txp666, 2022-09-14:
[image]
the version is 5.1.1   I build the projet failed on two computers

maxgerhardt, 2022-09-14:
Oh, you're actually right. On 5.1.1 it's broken. Use 

```
platform = espressif32@5.0.0
```
in the `platformio.ini` for a workaround.

txp666, 2022-09-14:
ok thank u  
i foud a way it's also worked 
[image]

Jason2866, 2022-09-14:
@valeros ALL S3 entrys for `"memory_type"` needs to be checked since the naming changed from core 2.0.3 to actual core 2.0.4

PR #914 (2022-09-26): fix "memory_type" for esp32s3box
since the naming changed. Fixes #912
Remove not needed entrys
closed 2022-09-26 by valeros